stm32duino/Arduino_Core_STM32 #686: UART baudrate calculation error at high clock

resetcoder, 2019-10-08:
Problem: STM32F407 can not calculate correct baud rate if MCU is overclocked over 171MHz.

UART Baudrate calculation happens in (i.e.) `stm32f4xx_hal_uart.c` in function `UART_SetConfig(UART_HandleTypeDef *huart)`.

Inside this function the following line calculates the `DIV_Mantissa` and `DIV_Fraction` values for the UART `BRR` register, which sets up a clock divider for the serial port to be as close as possible to the desired baud rate clock:

`huart->Instance->BRR = UART_BRR_SAMPLING8(HAL_RCC_GetPCLK2Freq(), huart->Init.BaudRate);`

(There is 16 bit caltulation also, but it is irrelevant for now.)

The function that returns the register value as result is a macro (in `stm32f4xx_hal_uart.h`).
Inside this macro the return value of `HAL_RCC_GetPCLK2Freq()` is multiplied by (uint)25, (uint)100, divided by (uint)100, etc.

The return value of `HAL_RCC_GetPCLK2Freq()` is the system clock frequency in Hz. In a 168MHz environment this is a relatively huge number for the (uint32_t) return type, because this number is later multiplied, divided by the (uint) numbers inside the macro.

My observation is if the system clock frequency is the following:

`UART_BRR_SAMPLING8((uint32_t)171798691, (uint32_t)115200)` returns `2979`,
`UART_BRR_SAMPLING8((uint32_t)171798692, (uint32_t)115200)` returns `0`

This is clearly an overflow, and system clock above 171.8MHz can not calculate the uart baudrate divider.

`UART_BRR_SAMPLING8((uint64_t)171798692, (uint32_t)115200)` returns `2979` again.

Changing the type from 32 bit uint to 64 bit uint of the first parameter fixes this issue, but getting the clock is coded inside the library and `HAL_RCC_GetPCLK2Freq()` return uint32_t.

This might be fixed systemwide by changing the return type of `HAL_RCC_GetPCLK2Freq()` to `uint64_t` or by changing the macro to force type casting for the first input parameter to 64 bit uint (but then this will fix only the uart related problems).

For now I fix this issue by changing the return type of `HAL_RCC_GetPCLK2Freq()` to 64 bit, but later I'd like to use the library updates, so please fix this issue if possible.

Thank You!

fpistm, 2019-10-08:
Hi @resetcoder 
Thanks for pointing this.
This is an issue in the HAL driver from the cube firmware.
This should be raised here:
https://github.com/STMicroelectronics/STM32CubeF4/issues
This repo only use it.

resetcoder, 2019-10-08:
Thank You for the answer.

I know it is the STM HAL driver, but I have no idea why did they use 32 bit integer, when they have MCU's faster thatn 172MHz. I'm not sure what is the reason behind it, and not sure ST would change it. I'm developing in the platformio environment and that uses this stm32duino library, this is why I posted it here. I see a little chance ST would change the driver, but if they won't, at least it should be fixed somehow in tha stm32duino lib.

I'll try it anyways.

fpistm, 2019-10-08:
No worry, I understand why you posted here.
I avoid to change Cube firmware source code because it will be to hard to track all change.
When an issue is raised, I submitted internally a ticket to solve it. Now, it is easier for community to submit issue.
I could not tell if there is a specific reason anyway they should be able to answer you.
As you can see, there are already some issues answered.

resetcoder, 2019-10-08:
Ok, I submitted the same issue there, waiting for reply. Thanko You for your help!

fpistm, 2019-10-08:
Thanks.
I close this one as this is not related to core and tracked here:
 https://github.com/STMicroelectronics/STM32CubeF4/issues/4

closed 2019-10-08 by fpistm