stm32duino/Arduino_Core_STM32 #687: [USB] Features request list

fpistm, 2019-10-08:
This issue aims to list all USB requests:

* RNDIS (#506): networking over USB
* Library (#559): Extend USB class support using Pluggable USB from Arduino and providind more classe support. Example in [Mbed](https://os.mbed.com/docs/mbed-os/v5.13/apis/usb.html):
[image]
* [USB Composite](https://github.com/arpruss/USBComposite_stm32f1)(#668): request to port it on this core. Not based on STM32 USB Device Library. Currently, target STM32F1 using [Arduino_STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32)

senceryazici, 2019-12-04:
Like i've mentioned in #817 i'm looking foward for USB Host implementation, spesifically ACM VCP implementation. I basically need to create virtual serial port and talk to device through that. 

Also, when the library is implemented I'll be happy to share my project as an example for the library maybe ? 
Thank you

rtek1000, 2020-03-02:
Done.

https://www.stm32duino.com/viewtopic.php?f=7&t=197&p=1372#p1372

oofus, 2020-04-08:
Is there any progress on implementing PlugableUSB ? Is it even going to happen ?

fpistm, 2020-04-08:
Hi @oofus ,
Time is always the missing ingredients. So, currently, there is no progress on this.
Any contribution are welcome 😉

oofus, 2020-04-11:
Hi @fpistm 
Thanks for the answer, at least I know the current status now.

Using and implementing are two quite different things. Just being willing to "give it a go" shouldn't be the criteria by which you pass an interview. You don't want any of my code near the core libraries !!!

Cheers

HamidSaffari, 2020-06-04:
Hi there.
Would you tell us where we can find some examples of that [USB Middleware library](https://github.com/stm32duino/Arduino_Core_STM32/tree/master/system/Middlewares/ST/STM32_USB_Device_Library).
Or give us one example code like read/write from a mass strong host device by using the MSC lib.
even for STM32CubeIDE so we can start porting it for Arduino.

HelloDB, 2020-06-24:
Thank you for your work. Can you provide an example of keyboard that can be used in BIOS environment

fpistm, 2020-06-25:
> 
> 
> Thank you for your work. Can you provide an example of keyboard that can be used in BIOS environment

Hi and welcome.
There is already a HID Kbd  example. Don't know if it can work in BIOS environment.

HelloDB, 2020-06-26:
Hi, I tested the keyboard in BIOS environment. It works well, but I can't find an example of AbsoluteMouse. I'm trying to modify the Mouse library

fpistm, 2020-06-26:
Fine.
For absolute mouse, I never tried.
I guess you will have to modify the library?

fpistm, 2020-06-26:
In fact it seems it will requires more dev as currently the mouse only send the relative move.
There is no get report. What is your goal? get the absolute position or send it ?

HelloDB, 2020-06-26:
I want to make a remote mouse, sending absolute coordinate data is easier to control it than using relative coordinate data

HelloDB, 2020-06-26:
> In fact it seems it will requires more dev as currently the mouse only send the relative move.
> There is no get report. What is your goal? get the absolute position or send it ?

I generate the mouse position in the webpage, send it to ESP32 through the server and then send it to STM32, and then STM32 controls the mouse movement of another computer

HelloDB, 2020-06-26:
I added `moveTo(uint16_t x, uint16_t y, uint8_t wheel)` and modified `HID_MOUSE_ReportDesc` `HID_MOUSE_REPORT_DESC_SIZE` and still cannot control the mouse

MS1987, 2020-09-25:
@fpistm , time flies. How about the function of USB Host now?

fpistm, 2020-09-25:
@MS1987 

> @fpistm , time flies. How about the function of USB Host now?

and it will continue. Time is always the missing ingredient.

rhapsodyv, 2020-10-03:
Hi, @fpistm 

I'm an active marlin contributor.

And I just got usb host with mass storage working on marlin, using your middleware + conf generated in cube mx for otg on gpio.

I would to like to know what is the api/interface that I should follow to send a PR to you.
Or general guidelines for the integration of the usb host.
Any info will help me to send a good PR for you. :-)

senceryazici, 2020-10-03:
That’s great! I’ll be looking up to it @rhapsodyv

HamidSaffari, 2020-10-03:
@rhapsodyv 
Here is the [USB Middleware library](https://github.com/stm32duino/Arduino_Core_STM32/tree/master/system/Middlewares/ST/STM32_USB_Device_Library)
and here is the [USB API documents](https://os.mbed.com/docs/mbed-os/v6.3/apis/usbmsd.html) for [ARM mbed OS](https://github.com/ARMmbed/mbed-os) which would be nice if we can port and use it inside Arduino core.

rhapsodyv, 2020-10-04:
I just pushed a working version, #1196. Right now, it just expose the default `USBH_MSC_` interface from `STM32_USB_Host_Library`.
There're some work to do, and it isn't a standard interface yet, but I think it's good for a first version. And it's working! I'm using it in Marlin.

I commented with a sample usage too.

Hyratel, 2021-11-09:
still no examples for Joystick via USB HID Middleware, and MHeironymous's Joystick library relies on HID.h (arduino core) -> PluggableUSB.h (arduino core) -> USBAPI.h (arduino core)

shreesha97, 2022-02-22:
Is USB MSC + CDC ready??
Any example available??

Levi--G, 2022-04-26:
@Hyratel I don't know if its still relevant to you but i just "completed" a basic library to add the traditional HID/PluggableUSB support. It is in nowhere ready for merging in the core, but if MHeironymous's joystick emulation is all you need you can use it right now. ;)

If anyone else comes across this and is knowledgeable in usb feel free to contact me to make the library more compatible. :)

Hyratel, 2022-05-15:
@Levi--G I'd definitely be interested in what you have thusfar and yes - MHJoy is about all I'm after for this. it's about the only thing blocking the 32F401 from being a viable competitor to the 32F103 or Pro Micro in the Simpit and Custom Joystick scene (MHJoy is kinda Hammery (very few iterative loops, everything is done in a very sequential way; it makes a good primer to HID descriptor construction) but also very understandable to an amateur like myself, it has a very clean structure)

Levi--G, 2022-05-15:
@Hyratel Well i noticed a while ago that my library actually works "by accident" if you just use the HID class it will work, but while working on receiving code i noticed i would have to redo a lot of code already in the framework... So now im running into the issue my library would need to replace framework files to work, which would either be manually or via some scripting in platformio, not compatible with arduino ide. So atm im still figuring things out :/
if you use platformio and only are interested in basic HID with SEND only and no USBSerial you can use my lib here: https://registry.platformio.org/libraries/levi--g/USBLibrarySTM32
But a proper redo with receiving and proper endpoint handling will take some time sadly :(

Hyratel, 2022-05-15:
@Levi--G I got it working on F401 with your LibHack, so it's rough-and-ready for that. I look forward to it being given bidirectional, but even this much is a watershed moment for the chip's usability

SunboX, 2022-09-19:
I'm looking for a USB MIDI Device or Host example for the F103C6 or F103C8. Is there already one available?

I've currently this one working:
https://github.com/arpruss/USBComposite_stm32f1/blob/master/examples/midiout/midiout.ino

But it uses the Maple core and is incompatible  with other libraries that are using the STM32Duino core.

fpistm, 2022-09-19:
Hi @SunboX,

unfortunately not.
IIRW, On the forum some ones talk about this but I guess there is no progress.

Levi--G, 2022-09-19:
@SunboX 
Well... i wrote a library to support all the original arduino usb libraries (i linked it a bit up) but i did not test MIDI yet, and installation is kindof hard atm since there is a bug in platformio, ill see if i can get it working yet and maybe you can test it then? It might work or not, but might be worth trying?

Levi--G, 2022-09-21:
@SunboX Hi, sorry to ping you twice, but i just tested [my library](https://github.com/Levi--G/USBLibrarySTM32) with the default arduino MIDIUSB and it works, I included a sample in the repository, it might not be ideal (especially with the extended installation procedure) but it might at least work with the new core :)

fpistm, 2022-09-21:
> @SunboX Hi, sorry to ping you twice, but i just tested [my library](https://github.com/Levi--G/USBLibrarySTM32) with the default arduino MIDIUSB and it works, I included a sample in the repository, [...]

Nice @Levi--G 
Thanks for sharing.

sstaub, 2023-08-13:
Are there any news? I see there is no real progress since some years.

Levi--G, 2023-08-15:
Well I would be prepared to do the work to integrating my "library" as an alternative USB implementation to provide compatibility with the traditional arduino libraries and check at least a few boxes. But i need someone from the team (maybe @fpistm ?) to guide me towards how to implement it in a way where it would not interfere with the original implmentation. Like what boards need to be supported (i ran into issues with DMA on some boards), what defines to use, make sure it doesn't interfere with the other implementation and set up testing and documentation and such... Not to mention code review and writing style i didn't even check for. 😓

fpistm, 2023-08-28:
USB_CON is used to enabled USB.
So You can simply try to use new define and add menu option to the boards.txt

Levi--G, 2023-10-13:
Yea the problem is I tried some different devices like stm32F401 or stm32F411 but there it does not seem to work, I don't know what goes wrong. I tried debugging with bmp and capturing packets, but the communication does not seem to work. The setup goes correctly so the hid devices are made and registered fine, but after that none of the endpoints are transmitting data. I have seen these devices don't use DMA for their communication, maybe that is the difference, but I cannot seem to find the issue as the calls made are identical to the usbcomposite library. So I am stuck not knowing if it has any use if only one device is confirmed working and at least 2 are not...

Hyratel, 2023-10-17:
> F401

I had good results using the Arduino IDE, not VSCode @Levi--G (do not like VScode for Assorted Personal Reasons). the only change I had to make was https://github.com/Levi--G/USBLibrarySTM32/issues/1

gigaj0ule, 2024-09-15:
TinyUSB has STM32 implementations, and many devices implemented already like MIDI and RNDIS.

It’s actively maintained too — could this be a good way to upgrade the USB subsystem?

https://github.com/hathach/tinyusb

fpistm, 2024-09-16:
> TinyUSB has STM32 implementations, and many devices implemented already like MIDI and RNDIS.
> 
> It’s actively maintained too — could this be a good way to upgrade the USB subsystem?
> 
> https://github.com/hathach/tinyusb

Yes it could. I know it and already follow it. Anyway, the main issue is the time. This is a community project and any contributions are welcome.
Since some times I've updated the core to solve some issue and now USB support is provided as a built-in library , it should ease the process to add new USB stack.

Levi--G, 2025-05-10:
Shameless plug that my library (https://github.com/Levi--G/USBLibrarySTM32) reached 1.0.0 status. Meaning that all STM32 USB (F0,F1,F3,L0,L1,...) / USB_OTG_FS (F2,F4,F7,L4...) devices are now supported and many popular boards are now tested and stable for AVR compatible pluggable libraries:
- Full PluggableUSB compatible for all supported boards
- MIDIUSB, CustomHID, Mouse, joystick and other libraries compatible at the same time!
- Automatic optimal native fifo/PMA management
- Using built-in native stm32 HAL with no extra dependencies (no TinyUSB)
- Many adjustable compatibility and performance options (Buffer options, EP size, ...)
- More features than standard AVR implementation (calls for checking/reiterating connection, ...)
- and more...

So if you landed here needing a way to get a standard (AVR) library working go give that a try.

rtek1000, 2025-05-12:
> [@Levi--G](https://github.com/Levi--G)
> 
> (this is long overdue, considering that the chip has been around for almost 10 years now)

Perhaps support for USB and STM32 is still crawling because many of these uCs had hardware design errors, and the manufacturer itself must have been discouraged. I left this information noted in the repository of a project I was developing with STM32F407 and STM32CubeIDE, at this link, in case anyone wants to know the details:

https://github.com/rtek1000/Datalogger_2039?tab=readme-ov-file#stm32-internal-hardware-usb

(Honestly I see more progress in implementing USB support on the ESP32-S3)

Levi--G, 2025-05-12:
> Perhaps support for USB and STM32 is still crawling because many of these uCs had hardware design errors, and the manufacturer itself must have been discouraged. 

Well arduino compatible device-only mode is here now, and the issue was not the hardware, its actually really stable in device mode. The big problem was the difference in USB architecture between the existing arduino library (polling driven) and the stm hal (interrupt/direct access driven). Well that and 32-bit and 16-bit alignment issues, and the different implementations (PMA/fifo), and the lack of time on my side,... i can go on for a bit 😉 .

I myself see little use for an arduino with usb in host mode, because you'll most likely want the speedup of using the hal/interrupts directly anyway, so that will most likely be the way to go there. So thats not in scope for me atm. USB_OTG_HS in device mode i can add as soon as i get a device, but this means that right now 2/3 of the stm32 family should be supported. Better late than never!