stm32duino/Arduino_Core_STM32 #688: May the F030 core require some optimization?

r-pv, 2019-10-08:
Hi, I'm using the F030 line on some projects and I like it but I always feel that the code gets too big on this mcu line, so I made a small test running some code and I got this results:

**On the arduino Uno:**
Sketch uses 4000 bytes (12%) of program storage space. Maximum is 32256 bytes.
Global variables use 91 bytes (4%) of dynamic memory, leaving 1957 bytes for local variables. Maximum is 2048 bytes.

**on my variants:**
_stm32f030K6_
Sketch uses 15444 bytes (47%) of program storage space. Maximum is 32768 bytes.
Global variables use 208 bytes (5%) of dynamic memory, leaving 3888 bytes for local variables. Maximum is 4096 bytes.
_stm32f103ve_
Sketch uses 11616 bytes (2%) of program storage space. Maximum is 524288 bytes.
Global variables use 224 bytes (0%) of dynamic memory, leaving 65312 bytes for local variables. Maximum is 65536 bytes.

**on the stock variants:**
_bluepill_
Sketch uses 10796 bytes (16%) of program storage space. Maximum is 65536 bytes.
Global variables use 216 bytes (1%) of dynamic memory, leaving 20264 bytes for local variables. Maximum is 20480 bytes.
_stm32f030f4_
Sketch uses 15400 bytes (93%) of program storage space. Maximum is 16384 bytes.
Global variables use 208 bytes (5%) of dynamic memory, leaving 3888 bytes for local variables. Maximum is 4096 bytes.
_stm32f407ve_
Sketch uses 10336 bytes (1%) of program storage space. Maximum is 524288 bytes.
Global variables use 232 bytes (0%) of dynamic memory, leaving 130840 bytes for local variables. Maximum is 131072 bytes.
_stm32f103re_
Sketch uses 11400 bytes (2%) of program storage space. Maximum is 524288 bytes.
Global variables use 224 bytes (0%) of dynamic memory, leaving 65312 bytes for local variables. Maximum is 65536 bytes.

So, for the F030 line the code gets too big, even relative to the other stm32 mcu's, so may require some optimization or is it something that I'm missing it?

The code I'm using it is not mean to be anything functional yet, I just trying to see what mcu I should be using on my next project, but I'm guessing that you could replicate this problem with any code, so the code it's:

```
#include <AutoPID.h>
#include <rTM1668.h>

rTM1668 rLED(PA8,PA11,PA12,6);

#define POT_PIN PA0
#define OUTPUT_PIN PB0

#define OUTPUT_MIN 0
#define OUTPUT_MAX 255
#define KP .12
#define KI .0003
#define KD 0

double temperature, setPoint, outputVal;
AutoPID myPID(&temperature, &setPoint, &outputVal, OUTPUT_MIN, OUTPUT_MAX, KP, KI, KD);

void setup(void){
	pinMode(POT_PIN, INPUT);
	pinMode(OUTPUT_PIN, OUTPUT);
	analogWriteFrequency(40000);

	myPID.setBangBang(4);
	//set PID update interval to 4000ms
	myPID.setTimeStep(40);
}

void loop(void){
	setPoint = analogRead(POT_PIN);
	myPID.run(); //call every loop, updates automatically at certain time interval
	analogWrite(OUTPUT_PIN, outputVal);
	rLED.clear();
	rLED.dpn((short)temperature,0,3);
	rLED.dpn((short)setPoint,3,2);
	delay(250);
}
```
and there's a hal_conf_extra.h disabling everything but the ADC:

```
//#define HAL_ADC_MODULE_DISABLED
#define HAL_I2C_MODULE_DISABLED
#define HAL_RTC_MODULE_DISABLED
#define HAL_SPI_MODULE_DISABLED
#define HAL_TIM_MODULE_DISABLED
#define HAL_DAC_MODULE_DISABLED
#define HAL_EXTI_MODULE_DISABLED
#define HAL_ETH_MODULE_DISABLED
#define HAL_SD_MODULE_DISABLED
#define HAL_QSPI_MODULE_DISABLED
```
The serial was disabled on all cases.
The only that was changed was the pinout where was required, like with the uno and the f030f4.

Cheers.

fpistm, 2019-10-09:
Hi @r-pv 
I've testing your code (without `rTM1668.h` as I'm not able to find the lib you used) with Disco F030R8, I've got 14632 bytes.
Analyzing the `.map` to see what use FLASH, it seems this is due to AutoPID which made a lot of computation with double. Then requires some instructions set from the libgcc:

Other STM32 you compare have a FPU (Floating Point Unit) and are not based on the same cortex M families which explains the difference.

So, I guess the only optimization which can be performed could be the HAL RCC which could be replaced by LL call.

r-pv, 2019-10-09:
> Hi @r-pv
> I've testing your code (without `rTM1668.h` as I'm not able to find the lib you used) with Disco F030R8, I've got 14632 bytes.
> Analyzing the `.map` to see what use FLASH, it seems this is [...]

Hi, thanks for your reply, it makes sense now. Regarding to the HAL RCC optimization it's a feature that it can be used now or it's something that may be on the future of the core?.

fpistm, 2019-10-09:
Yes, LL are fully available, you can try on your side by redefining the WEAK `void SystemClock_Config(void)`.
STM32CubeMX could help to generate it using LL instead of HAL RCC

r-pv, 2019-10-09:
Awesome, thanks, I'll give it a try. Will I loose some functionality by changing this?, I'm not using anything of the HAL functions directly on arduino ide, just standard arduino code.

fpistm, 2019-10-09:
No you should not loose functionality. This is just the way the clock config is done.

r-pv, 2019-10-09:
> 
> 
> No you should not loose functionality. This is just the way the clock config is done.

Thanks.

closed 2019-10-10 by fpistm