stm32duino/Arduino_Core_STM32 #691: stm32f030f4p6 Serial doesn't seem to work

mcer12, 2019-10-09:
Serial.print() doesn't output anything at any baud rate. When I use SoftwareSerial at the RX and TX gpios instead of hardware, it works. I use serial to flash the chip so it's not a connection issue. Am I missing something?

> #include <SoftwareSerial.h>
> 
> SoftwareSerial mySerial(10, 9); // RX, TX
> 
> void setup() {
>   mySerial.begin(9600);
> }
> 
> void loop() { // run over and over
>     mySerial.println("TEST");
>     delay(1000);
> }

Basic Serial doesn't work.

> void setup() {
>   Serial.begin(9600);
> }
> 
> void loop() { // run over and over
>     Serial.println("TEST");
>     delay(1000);
> }

fpistm, 2019-10-09:
Hi @mcer12 
I guess you use the DEMO board ?
I did not test it as I do not have it. It is a contribution.
Looking at the variant, I can see this:
https://github.com/stm32duino/Arduino_Core_STM32/blob/6808f06a3c183f22784f0486a3034644535c5b2b/variants/DEMO_F030F4/variant.h#L53-L54

So I guess this is the one you used to flash over serial.
Anyway, the default Serial instance has been mapped to:
https://github.com/stm32duino/Arduino_Core_STM32/blob/6808f06a3c183f22784f0486a3034644535c5b2b/variants/DEMO_F030F4/variant.h#L83-L88

Probably because PA9/PA10 are reserved for I2C.
So, you can change the default Serial like this:

```C++
Serial.setRx(PA10);
Serial.setTx(PA9);
Serial.begin(9600);
```

or kept default pins and connect to PA3/PA2

mcer12, 2019-10-09:
I see, it works now, thank you. This is pretty confusing, especially looking at the pinout of the demo board
https://user-images.githubusercontent.com/20950920/48308359-6e531480-e56b-11e8-9e92-98dba779e01b.jpg
Maybe it would be worth to add this as a switch in the compiler options, if user wants to use SERIAL1 for serial or i2c?

closed 2019-10-09 by mcer12

fpistm, 2019-10-09:
Well, all can be done. Anyway as there is only one I2C, I guess this is the best option.