stm32duino/Arduino_Core_STM32 #695: SPI.beginTransaction() does not work - help

litian85734, 2019-10-12:
Dear All sirs:
    I use STM32F401 SPI HW mode drive the TFT LCD, but the speed is very low even worse than soft SPI mode,
I try to use SPI.beginTransaction(SPISettings(XXXX, MSBFIRST, SPI_MODE0)) to change the SPI speed,  but seems no use,    Anybody meet same issue?

Thanks

fpistm, 2019-10-12:
Hi @litian85734 
could you be more precise  (setup, LCD, core version, board, ...) and share your sketch.

stas2z, 2019-10-18:
Ive played with st7735 spi lcd last week, using f401cc blackpill, and can say SPI.beginTransaction works fine, ive made a test using 10kb bitmap drawing in a loop which changed spi clock, drawing speed changed as expected.
Also ive tested software spi bitbanging in compare with hw spi and it was much slower. So probably your lib have an issue with hw spi implementation

BennehBoy, 2019-10-19:
+1 There a re a _lot_ of broken SPI implementations out their in user
library land.

fpistm, 2019-10-21:
No feedback since more than a week and it works.
Close it

closed 2019-10-21 by fpistm