stm32duino/Arduino_Core_STM32 #696: SerialUSB  no print-out

anatom74, 2019-10-12:
HW: Nucleo board STM32F4RE411

I wanted to use USB CDC for serial communication and set this accordingly: 
[image]

Running the code below I get LED blinking but no print out on com port (COM3)
`void setup() {
  // put your setup code here, to run once:
 SerialUSB.begin();
 pinMode(LED_BUILTIN, OUTPUT);
 delay(2000);
}

void loop() {
  // put your main code here, to run repeatedly:
  SerialUSB.println("TEST");
  delay(400);
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
}`

The COM port is ST-LINK VCP (COM3):
[image]

Am I using this right?

fpistm, 2019-10-12:
Hi @anatom74 
This is not the USB CDC. This is an hardware serial connected to stlink. So disable USB cdc then use Serial.

anatom74, 2019-10-13:
oh, my bad. works now.
Thanks @fpistm.

closed 2019-10-13 by anatom74

fpistm, 2019-10-13:
Welcome