stm32duino/Arduino_Core_STM32 #697: Provide provisions for user defined TIMx_IRQ_Handlers.

geosmall, 2019-10-12:
**Is your feature request/improvement related to a problem? Please describe.**
Attempts to make full use of timer features in applications (i.e brushless motor control) require direct access to Timer IRQ Handlers.  Current design of HardwareTimer.cpp takes full control of Timer IRQ Handlers.  No user level compile switch is available to turn off these features.  Any attempts to define sketch level Timer IRQ Handlers generates  multiple definition of `TIMx_IRQHandler' errors.
 
**Describe the solution you'd like to see**
Provide user level switch to turn off HardwareTimer (and associated pin PWM) functionality for advanced users.  I've hacked a quick and dirty solution with a user flag in build_opt.h with #ifndef EXCLUDE_HARDWARE_TIMER_MODULE in HardwareTimer.cpp, analog.cpp and wiring_digital.c files to get rid of the multiple definition errors, but have not thoroughly tested yet.

STM32 peripherals have many powerful features.  A design philosophy to be considered for this core would be that features that take over IRQ Handlers (TIM, I2C, etc.) can still be switched off by advanced users looking to take full control of the peripheral at the sketch level.

fpistm, 2019-10-14:
Hi @geosmall 
Right, I've already think about that and maybe I have a solution, anyway this will need some extra study.
Any proposal are welcome.

ABOSTM, 2019-10-14:
Hi @geosmall,
There is an alternative:
You can directly replace NVIC vector with an IRQ handler of your choice:
      _`NVIC_SetVector(irq_event_n, handler);`_
Just be warned that it will break any usage of concerned NVIC vector.
This NVIC API is directly usable in arduino sketch.

geosmall, 2019-10-17:
> There is an alternative:
> You can directly replace NVIC vector with an IRQ handler of your choice:
> _`NVIC_SetVector(irq_event_n, handler);`_
Hi @ABOSTM,
This approach sounds promising.  I'm having trouble finding documentation on NVIC_SetVector.   Is it presently implemented in STM32 core.  I'm working STM32F0 at present, with no VTOR register..

ABOSTM, 2019-10-17:
Hi @geosmall,
NVIC_SetVector() is defined in CMSIS so it is not part of this core,
but it comes with CMSIS which is automatically installed with this core.
For information about this function: official ARM documentation:
[https://arm-software.github.io/CMSIS_5/Core/html/group__NVIC__gr.html#gab43c1c59d5c081f1bc725237f4b1f916](https://arm-software.github.io/CMSIS_5/Core/html/group__NVIC__gr.html#gab43c1c59d5c081f1bc725237f4b1f916)

You are right STM32F0 doesn't support it as it is based on Cortex-M0.
And you are also right saying you need first to relocated to RAM in order to use NVIC_SetVector().

geosmall, 2019-10-18:
Apparently doable even on Cortex-M0 using relocation to RAM after carving out enough space for the vectors with modified linker script.  https://community.st.com/s/question/0D50X00009XkfRgSAJ/stm32f0-no-interrupt-after-vectortable-relocation  So a good bit of work.

I thinking user #defines would be a better more versatile approach long term and work across variants.

geosmall, 2019-10-19:
I was able to get NVIC_SetVector()  working on STM32F051K6 with the following sketch and linker script:

Test sketch:
`
/*
  DEMO to:
  Remap Wraith32 STM32F0051K6 interrupt table to RAM,
  then use NVIC_SetVector() to insert our own sketch based
  SysTick interrupt handler.  
*/

#define APPLICATION_ADDRESS     (uint32_t)0x08000000

__IO uint32_t VectorTable[48] __attribute__((section(".RAMVectorTable")));

extern "C" void My_SysTick_IRQHandler(void) {

  HAL_IncTick();

}

// the setup function runs once when you press reset or power the board
void setup() {

  // Relocate by software the vector table to the internal SRAM at 0x20000000
  // Copy the vector table from the Flash (mapped at the base of the application
  // load address 0x08000000) to the base address of the SRAM at 0x20000000
  for(uint32_t i = 0; i < 48; i++) {
    VectorTable[i] = *(__IO uint32_t*)(APPLICATION_ADDRESS + (i<<2));
  }

  /* Enable the SYSCFG peripheral clock*/
  __HAL_RCC_SYSCFG_CLK_ENABLE();

  // Remap SRAM at 0x00000000
  __disable_irq();
  __HAL_SYSCFG_REMAPMEMORY_SRAM();
  __enable_irq();

  NVIC_SetVector(SysTick_IRQn, (uint32_t)My_SysTick_IRQHandler);

  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);

}

// the loop function runs over and over again forever
void loop() {

  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second

}
`

Linker script:
`
/*
******************************************************************************
**
**  File        : LinkerScript.ld
**
**  Author      : Auto-generated by STM32CubeMx, modified by G. Small
**
**  Abstract    : Linker script for STM32F051K6Tx series
**                32Kbytes FLASH and 256 bytes SRAM, 7936 bytes RAM
**
**                Set heap size, stack size and stack location according
**                to application requirements.
**
**                Set memory bank area and size if external memory is used.
**
**  Target      : STMicroelectronics STM32
**
**  Distribution: The file is distributed “as is,” without any warranty
**                of any kind.
**
*****************************************************************************
*/

/* Entry Point */
ENTRY(Reset_Handler)

/* Highest address of the user mode stack */
_estack = 0x20002000;    /* end of RAM */
/* Generate a link error if heap and stack don't fit into RAM */
_Min_Heap_Size = 0x200;      /* required amount of heap  */
_Min_Stack_Size = 0x400; /* required amount of stack */

/* Specify the memory areas */
MEMORY
{
  SRAM  (xrw) : ORIGIN = 0x20000000,  LENGTH = 256
  RAM (xrw)   : ORIGIN = 0x20000100,  LENGTH = 7936
  FLASH (rx)  : ORIGIN = 0x8000000,   LENGTH = 32K
}

/* Define output sections */
SECTIONS
{
  /* The startup code goes first into FLASH */
  .isr_vector :
  {
    . = ALIGN(4);
    KEEP(*(.isr_vector)) /* Startup code */
    . = ALIGN(4);
  } >FLASH

  /* The program code and other data goes into FLASH */
  .text :
  {
    . = ALIGN(4);
    *(.text)           /* .text sections (code) */
    *(.text*)          /* .text* sections (code) */
    *(.glue_7)         /* glue arm to thumb code */
    *(.glue_7t)        /* glue thumb to arm code */
    *(.eh_frame)

    KEEP (*(.init))
    KEEP (*(.fini))

    . = ALIGN(4);
    _etext = .;        /* define a global symbols at end of code */
  } >FLASH

  /* Constant data goes into FLASH */
  .rodata :
  {
    . = ALIGN(4);
    *(.rodata)         /* .rodata sections (constants, strings, etc.) */
    *(.rodata*)        /* .rodata* sections (constants, strings, etc.) */
    . = ALIGN(4);
  } >FLASH

  .ARM.extab   : { *(.ARM.extab* .gnu.linkonce.armextab.*) } >FLASH
  .ARM : {
    __exidx_start = .;
    *(.ARM.exidx*)
    __exidx_end = .;
  } >FLASH

  .preinit_array     :
  {
    PROVIDE_HIDDEN (__preinit_array_start = .);
    KEEP (*(.preinit_array*))
    PROVIDE_HIDDEN (__preinit_array_end = .);
  } >FLASH
  .init_array :
  {
    PROVIDE_HIDDEN (__init_array_start = .);
    KEEP (*(SORT(.init_array.*)))
    KEEP (*(.init_array*))
    PROVIDE_HIDDEN (__init_array_end = .);
  } >FLASH
  .fini_array :
  {
    PROVIDE_HIDDEN (__fini_array_start = .);
    KEEP (*(SORT(.fini_array.*)))
    KEEP (*(.fini_array*))
    PROVIDE_HIDDEN (__fini_array_end = .);
  } >FLASH

  /* used by the startup to initialize data */
  _sidata = LOADADDR(.data);

  /* Initialized data sections goes into RAM, load LMA copy after code */
  .data : 
  {
    . = ALIGN(4);
    _sdata = .;        /* create a global symbol at data start */
    *(.data)           /* .data sections */
    *(.data*)          /* .data* sections */

    . = ALIGN(4);
    _edata = .;        /* define a global symbol at data end */
  } >RAM AT> FLASH

  
  /* Uninitialized data section */
  . = ALIGN(4);
  .bss :
  {
    /* This is used by the startup in order to initialize the .bss secion */
    _sbss = .;         /* define a global symbol at bss start */
    __bss_start__ = _sbss;
    *(.bss)
    *(.bss*)
    *(COMMON)

    . = ALIGN(4);
    _ebss = .;         /* define a global symbol at bss end */
    __bss_end__ = _ebss;
  } >RAM

  /* User_heap_stack section, used to check that there is enough RAM left */
  ._user_heap_stack :
  {
    . = ALIGN(8);
    PROVIDE ( end = . );
    PROVIDE ( _end = . );
    . = . + _Min_Heap_Size;
    . = . + _Min_Stack_Size;
    . = ALIGN(8);
  } >RAM

  

  /* Remove information from the standard libraries */
  /DISCARD/ :
  {
    libc.a ( * )
    libm.a ( * )
    libgcc.a ( * )
  }

  .ARM.attributes 0 : { *(.ARM.attributes) }
  .RAMVectorTable : { *(.RAMVectorTable) } >SRAM AT> FLASH

}
`

geosmall, 2019-10-20:
Assuming use of NVIC_SetVector() per above to manage IRQ_Handlers at the sketch level, I run into the next issue.  The application I am trying to port originated as an STM32CubeMx project.  As such it makes use of HAL_XXX_MspInit() functions.  As these functions are claimed within STM32 core I get a series of multiple definition linker errors.

The above will make porting of STM32CubeMx generated code or examples an onerous task for Arduino sketch writers.  I can work around this by writing local replacement HAL_XXX_Init() functions that then allow me to insert my own local HAL_XXX_MspInit() functions, but this is all feels like unnecessary work.

I am convinced that the correct design philosophy is to provide user level build_opt.h switches to turn off TIM/ADC/Etc... functionality for advanced user applications.

Link errors generated:
```
Linking everything together...
"C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\tools\arm-none-eabi-gcc\8.2.1-1.7/bin/arm-none-eabi-gcc" -mcpu=cortex-m0  -mthumb -g -Og --specs=nano.specs -Wl,--defsym=LD_FLASH_OFFSET=0 -Wl,--defsym=LD_MAX_SIZE=32768 -Wl,--defsym=LD_MAX_DATA_SIZE=8192 -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common "-TC:\Users\geoma\OneDrive\Arduino\hardware\STM32\stm32\variants\WRAITH32_F051K6/ldscript.ld" "-Wl,-Map,C:\Users\geoma\AppData\Local\Temp\arduino_build_500636/f051bldc.ino.map"   "-LC:\Users\geoma\AppData\Local\Arduino15\packages\STM32\tools\CMSIS\5.5.1/CMSIS/DSP/Lib/GCC/" -larm_cortexM0l_math -o "C:\Users\geoma\AppData\Local\Temp\arduino_build_500636/f051bldc.ino.elf" "-LC:\Users\geoma\AppData\Local\Temp\arduino_build_500636" -Wl,--start-group "C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\sketch\eeprom.c.o" "C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\sketch\main.c.o" "C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\sketch\stm32f0xx_hal_msp.c.o" "C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\sketch\f051bldc.ino.cpp.o" "C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\core\PeripheralPins.c.o" "C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\core\variant.cpp.o" -Wl,--whole-archive "C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\core\core.a" -Wl,--no-whole-archive -lc -Wl,--end-group -lm -lgcc -lstdc++
c:/users/geoma/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\core\core.a(timer.c.o): in function `HAL_TIM_Base_MspInit':

C:\Users\geoma\OneDrive\Arduino\hardware\STM32\stm32\cores\arduino\stm32/timer.c:30: multiple definition of `HAL_TIM_Base_MspInit'; C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\sketch\stm32f0xx_hal_msp.c.o:C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\sketch/stm32f0xx_hal_msp.c:219: first defined here

c:/users/geoma/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\core\core.a(timer.c.o): in function `HAL_TIM_Base_MspDeInit':

C:\Users\geoma\OneDrive\Arduino\hardware\STM32\stm32\cores\arduino\stm32/timer.c:50: multiple definition of `HAL_TIM_Base_MspDeInit'; C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\sketch\stm32f0xx_hal_msp.c.o:C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\sketch/stm32f0xx_hal_msp.c:344: first defined here

c:/users/geoma/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\core\core.a(analog.cpp.o): in function `HAL_ADC_MspInit':

C:\Users\geoma\OneDrive\Arduino\hardware\STM32\stm32\cores\arduino\stm32/analog.cpp:572: multiple definition of `HAL_ADC_MspInit'; C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\sketch\stm32f0xx_hal_msp.c.o:C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\sketch/stm32f0xx_hal_msp.c:75: first defined here

c:/users/geoma/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\core\core.a(analog.cpp.o): in function `HAL_ADC_MspDeInit':

C:\Users\geoma\OneDrive\Arduino\hardware\STM32\stm32\cores\arduino\stm32/analog.cpp:653: multiple definition of `HAL_ADC_MspDeInit'; C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\sketch\stm32f0xx_hal_msp.c.o:C:\Users\geoma\AppData\Local\Temp\arduino_build_500636\sketch/stm32f0xx_hal_msp.c:123: first defined here

collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Wraith32 speed controller.
```

fpistm, 2019-10-21:
@geosmall 

> I believe that the correct design philosophy is to provide user level build_opt.h switches to turn off TIM / ADC / COMP / Etc... functionality for advanced user applications.

I guess this is the simplest way. 

Anyway why use Arduino if you generate you code thanks CubeMX why not use directly CubeFw with CubeIDE or other?

geosmall, 2019-10-21:
> Anyway why use Arduino if you generate you code thanks CubeMX why not use directly CubeFw with CubeIDE or other?

Two reasons:
1. Board independence, easy to switch between variants
2. Toolchain is managed, Arduino_Core_STM32 is a good way to share projects with beginner programmers without getting bogged down in toolchain configs.

fpistm, 2019-11-02:
@geosmall 
let me know if  #750 answer your request.

fpistm, 2019-11-06:
@geosmall 
any update on this, please?

geosmall, 2019-11-10:
Sorry, was on travel.  I will test and come back here.

fpistm, 2019-11-10:
No worry 😉

closed 2019-11-28 by fpistm

fpistm, 2019-11-28:
Wiki:
https://github.com/stm32duino/wiki/wiki/HAL-configuration#hal-module-only

Bambofy, 2020-03-02:
> 
> 
> Wiki:
> https://github.com/stm32duino/wiki/wiki/HAL-configuration#hal-module-only

How do we know what parts of the core this will affect?

Bambofy, 2020-03-02:
When defining #HAL_TIM_MODULE_ONLY i receive these errors on building:
 

> 
> board.h:8: In file included from
> wiring.h:41: from
> Arduino.h:32: from
> uart.c:38: from
>  
>  
> board.h:8: In file included from
> analog.h: 55:66: error: unknown type name 'TimerCompareFormat_t
> timer.c:15: from
>    55 | void pwm_start(PinName pin, uint32_t clock_freq, uint32_t value, TimerCompareFormat_t resolution)
>  
>    |                                                                  ^~~~~~~~~~~~~~~~~~~~
> analog.h: 55:66: error: unknown type name 'TimerCompareFormat_t
>    55 | void pwm_start(PinName pin, uint32_t clock_freq, uint32_t value, TimerCompareFormat_t resolution)
>    |                                                                  ^~~~~~~~~~~~~~~~~~~~
>  
> board.h:8: In file included from
> wiring.h:41: from
> Arduino.h:32: from
>  
> analog.cpp:39: In file included from
> low_power.c:39: from
>  
>  
> analog.h: 55:66: error: unknown type name 'TimerCompareFormat_t
> analog.h: 55:66: error: 'TimerCompareFormat_t' has not been declared
>    55 | void pwm_start(PinName pin, uint32_t clock_freq, uint32_t value, TimerCompareFormat_t resolution)
>    55 | void pwm_start(PinName pin, uint32_t clock_freq, uint32_t value, TimerCompareFormat_t resolution)
>    |                                                                  ^~~~~~~~~~~~~~~~~~~~
>    |                                                                  ^~~~~~~~~~~~~~~~~~~~
> Error compiling libraries
> Build failed for project

fpistm, 2020-03-02:
Yes seems logical...
Did you read the wiki ?
Where did you add you define...

fpistm, 2020-03-02:
> How do we know what parts of the core this will affect?

Sorry, 
Just saw your first comment...

So, all using the HAL module you used. This is for advanced application, so it expect you know what you do and want.

About your issue:
https://github.com/stm32duino/Arduino_Core_STM32/issues/839

Bambofy, 2020-03-02:
No worries :)

I've just used the HardwareTimer library for now instead ;) So should it build if i define both:

```
#define HAL_TIM_MODULE_ENABLED
#define HAL_TIM_MODULE_ONLY

```
Like this? I will test it in a moment.

fpistm, 2020-03-02:
There is a bug as I mentioned: #839

And again please use the forum.

commit 16d7b2ba (2019-11-28): [SPI] Allow to enable HAL module only

Allow HAL SPI module usage without any usage by the core.

Fixes #697
commit 19884107 (2019-11-28): [UART] Allow to enable HAL module only

Define `HAL_UART_MODULE_ONLY` in `build_opt.h` or `hal_conf_extra.h`
allow HAL UART module usage without any usage by the core.

Fixes #697
commit 2697be12 (2019-11-28): [DAC] Allow to enable HAL module only

Define `HAL_DAC_MODULE_ONLY` in `build_opt.h` or `hal_conf_extra.h`
allow HAL DAC module usage without any usage by the core.

Fixes #697
commit 34a757d3 (2019-11-28): [RTC] Allow to enable HAL module only

Define `HAL_RTC_MODULE_ONLY` in `build_opt.h` or `hal_conf_extra.h`
allow HAL RTC module usage without any usage by the core.

Fixes #697