stm32duino/Arduino_Core_STM32 #698: The new name for stimer_t

MagoKimbra, 2019-10-12:
**Please, before reporting any issue**
- Make sure you are using the latest Arduino_Core_STM32 version.
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
- Make sure the issue is not already reported/fixed on GitHub or discussed on the [stm32duino forum](http://stm32duino.com)
- Submit a GitHub issue only for reporting a problem related to the [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32).
- **Avoid** to submit a GitHub issue for project troubleshooting.
 
Any questions/feedback/suggestions should be discussed on the [stm32duino forum](http://stm32duino.com):
  * questions on the [STM32 Core](http://stm32duino.com/viewforum.php?f=48)
  * bugs/enhancements on the [STM core: Bugs and enhancements](http://stm32duino.com/viewforum.php?f=49)

When reporting any issue, please try to provide all relevant information:

**Describe the bug**

This is not a real bug.
I write the MK4duo firmware for 3D printers and I have recently inserted the RUMBA32 card on the arduino stm32. With 1.6.1 everything is ok, but now with 1.7.0 I have errors. In summary there is no longer the definition of stimer_t, I cannot find the new name or how it was transformed, you could kindly give me an indication. Thanks a lot.

fpistm, 2019-10-12:
Hi, this is now replaced by HardwareTimer.

MagoKimbra, 2019-10-12:
Uhm ok i see it.
Question pause and resume, is similar to HAL_NVIC_DisableIRQ and HAL_NVIC_EnableIRQ.
But for know if is it in pause or not?

MagoKimbra, 2019-10-13:
Ok I almost got it all sorted out, but I have a few things I can't understand.
I hope you can help me.

First: With this void I read the count of the timer:
```
FORCE_INLINE static uint32_t HAL_timer_get_count(const uint8_t timer_num) {
  return __HAL_TIM_GET_COUNTER(&TimerHandle[timer_num].handle);
}
```
Now it becomes so:
```
FORCE_INLINE static uint32_t HAL_timer_get_count(const uint8_t timer_num) {
  return MK_timer[timer_num]->getCount();
}
```
MK_timer is new HardwareTimer is array 0 for Stepper, 1 for Temperature.

First I set the count with this void:
```
FORCE_INLINE static void HAL_timer_set_count(const uint8_t timer_num, const uint32_t count) {
  __HAL_TIM_SET_AUTORELOAD(&TimerHandle[timer_num].handle, count);
  if (HAL_timer_get_count(timer_num) >= count)
    TimerHandle[timer_num].handle.Instance->EGR |= TIM_EGR_UG; // Generate an immediate update interrupt
}
```
Now it should be like this:
```
FORCE_INLINE static void HAL_timer_set_count(const uint8_t timer_num, const uint32_t count) {
  MK_timer[timer_num]->setOverflow(count);
}
```
But how do I force the interrupt as before? There is a way to know the instance and then put the register EGR |= TIM_EGR_UG?
Now i do with
```
if (timer_num == STEPPER_TIMER_NUM) TIM5->EGR |= TIM_EGR_UG
else TIM8->EGR |= TIM_EGR_UG
```
TIM5 use for stepper and TIM8 use for Temperature (because TIM 6 and 7 use for tone ans servo right?)

Last question pause() stop interrupt and resume() restart interrupt, but i f i want know the state in pause o resume?

Thanks for your answers.

fpistm, 2019-10-14:
I will let  @ABOSTM answered to this as he implement the HT.

ABOSTM, 2019-10-14:
Hi @MagoKimbra,

Few answers to your questions:

> But how do I force the interrupt as before?

You can use API:
`HardwareTimer::refresh();`
It will writre register (and generate update event) like TIM8->EGR |= TIM_EGR_UG

> TIM 6 and 7 use for tone ans servo right?

That is right that we try to use TIIM6 and TIM7  (when available) for servo and tone, because they are the simpliest Hardware timers, and we let more complex timers for people who wants to make more complex things.
This means that on other boards that don't have TIM6 or TIM7, other Hardware timer may be used. Ex: on NUCLEO_F103RB, we use TIM3 and TIM4.

> pause() stop interrupt and resume() restart interrupt

Not exactly: pause() will not only disable interrupt, but also disable the whole timer.

> i want know the state in pause o resume?

Currently there is no API in HardwareTimer library to know whether timer is paused or resume.
There is a possibility, using STM32 cube HAL/LL API:
`__STATIC_INLINE uint32_t LL_TIM_IsEnabledCounter(_HardwareTimerObj.handle.Instance)`
But this should be done within Hardware timer class, because _HardwareTimerObj is a private attibut.
You are welcome to contribute :-)

MagoKimbra, 2019-10-14:
Indeed last night concerning everything I used refresh and it works, perfectly ...
Since the firmware must work on different processors, I have a HAL for each processor inside the firmware, so to disable and activate interrupts, I call a HAL_disable_interrupt function that in the specific case of the STM32 pauses the timer, in this function I will declare a Boolean variable you are about to set the timer status when it is disabled and when it is enabled, I will check this variable to find out.
Maybe you could add a timer function IsState () to subsequent patches to know in what state it is, pause, running, or other.
Thanks.

fpistm, 2019-10-15:
> Maybe you could add a timer function IsState () to subsequent patches to know in what state it is, pause, running, or other.

Hi @MagoKimbra 
Please, open a new issue for this request.
Do not hesitate to implement it and provide a PR if you can.
Any contributions are welcome :wink:

I close this one as it is answered

closed 2019-10-15 by fpistm