stm32duino/Arduino_Core_STM32 #702: USB Virtual Serial port on win 10

ladyada, 2019-10-13:
when uploading a simple sketch to test serial over USB for my Feather F405 definition, the device created does not seem to be USB CDC compatible

note it does not appear under the COM ports

[image]

if i force it to WinUSB
[image]

then it shows up under generic USB devices

[image]

seems...odd? like shouldn't it a generic CDC USB device? perhaps im missing something in my definition?

code:
```
int led = 13;

void setup() {                
  pinMode(led, OUTPUT);     
  SerialUSB.begin(115200);
}

// the loop routine runs over and over again forever:
void loop() {
  SerialUSB.println(millis());
  digitalWrite(led, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(100);               // wait for a second
  digitalWrite(led, LOW);    // turn the LED off by making the voltage LOW
  delay(100);               // wait for a second
}
```

[image]

fpistm, 2019-10-14:
Hi @ladyada 
You configuration is correct.
It seems more an issue with driver used.
In theory no specific driver is required and the one from Windows is enough:
[image]

As the default VID:PID is correct (STM: VCP), I guess an other driver claims it is better than the generic one.

Just FYI, as USB CDC is set as _generic Serial_, you can use the `Serial` name instead of `SerialUSB`.

ladyada, 2019-10-14:
hmm, is there a way to change the vid pid? i will use our own

fpistm, 2019-10-14:
To change the VID, add in the https://github.com/ladyada/Arduino_Core_STM32/blob/7bf7f6b55a5685e93aaa275a3c134822f1fe6e08/boards.txt#L987

`GenF4.menu.pnum.FEATHER_F405.build.vid=0x239A`

For the PID, currently this is hardcoded but, it should be easy to fix:
https://github.com/ladyada/Arduino_Core_STM32/blob/7bf7f6b55a5685e93aaa275a3c134822f1fe6e08/cores/arduino/stm32/usb/usbd_desc.c#L52

In the meantime, you can edit it on your side to check if this solved your driver issue.

Did you try under Linux to see if you get the same behavior?

ladyada, 2019-10-14:
ok yeah it was the dfuse driver being grabby -  i fixed it like this:

[image]

closed 2019-10-14 by ladyada

fpistm, 2019-10-15:
Fine.