stm32duino/Arduino_Core_STM32 #706: Automatic restart in DFU Bootloader mode

ladyada, 2019-10-16:
When using the built in USB DFU bootloader, lets auto-reset into the bootloader mode from USB Serial using something like 1200-baud knocking

stas2z, 2019-10-18:
Uh, that will be great, ive tried to implement smth like this with F401 (using maple "LEAF" sequence as a trigger), but stucked cuz it just reboots instead of running bootloader, seems like BL code checks for BOOT0 not pulling up and jumping to fw stored

fpistm, 2019-10-18:
I have a first working version :wink:

akamensky, 2020-01-19:
Are there any updates on this issue? I've got this Adafruit feather with stm32 on it, and it is a bit of inconvenience that I cannot just straight work from Arduino IDE

matthijskooijman, 2020-01-22:
There is an initial working, but rough, version at #710. I've been using that, along with some custom board files to use dfu-util, for a while now (so I'm not sure if that PR works out of the box yet, but it certainly works).

fpistm, 2020-01-22:
It works but it is not enough generic and I would avoid the backup register usage.

ramboerik, 2021-05-01:
Hi @fpistm !

What is the status of this feature? I see that you have pushed some code after your last comment that seems to solve it.

fpistm, 2021-05-01:
I have to rebase and test. But IIWR I have some dev to do.

ramboerik, 2021-05-01:
Thanks for the info. It would be a nice feature to have 👍

BCsabaEngine, 2021-06-23:
It would be a great option for testing (test upload etc) with strange sideeffects to. When production upload I can press the boot0 button.

mikeysklar, 2026-08-29:
This works from sketch code today, no core change needed. `USBSerial::baud()` and `dtr()` are already public, and the core leaves `VTOR` at 0, so the only missing piece is the SYSCFG remap.

```cpp
static void jump_to_system_bootloader(void) {
  __disable_irq();
  SysTick->CTRL = 0;                       // NVIC->ICER cannot mask SysTick
  HAL_RCC_DeInit();
  HAL_DeInit();
  __HAL_RCC_SYSCFG_CLK_ENABLE();
  __HAL_SYSCFG_REMAPMEMORY_SYSTEMFLASH();  // BOOT0 low means 0x0 is main flash
  SCB->VTOR = 0;                           // do not set 0x1FFF0000, the ROM does it
  __DSB(); __ISB();
  __enable_irq();
  __set_MSP(*(volatile uint32_t *)0x1FFF0000);
  ((void (*)(void)) * (volatile uint32_t *)0x1FFF0004)();
  while (1) {}
}

void loop() {
  if (Serial.baud() == 1200 && !Serial.dtr()) {
    delay(20);
    jump_to_system_bootloader();
  }
}
```

Tested on a Feather STM32F405 Express, core 3.0.0.

The jump lands in the ROM every time. But about two times in three the ROM runs for 1.5s and then resets back into the sketch. AN2606 Rev 61 Figure 32/33 shows why: after `USB cable detected -> yes` the next decision is `HSE detected`, and the no branch is a box labelled "Generate System reset". The ROM measures the crystal against the HSI, and section 28.2.1 warns that is marginal:

> it is better to use 8 MHz instead of 25 MHz

This board has a 12 MHz crystal.

The failure is a clean reset back into the sketch, so retrying is safe. Retrying until DFU appeared, 6 trials took 1, 1, 2, 2, 5 and 12 attempts. All succeeded. So a hook should retry rather than give up after one touch.