stm32duino/Arduino_Core_STM32 #707: Custom Bootloader

serkanc, 2019-10-16:
I Want to write custom bootloader for my project, to update flash via Sd card.
I will use stm32duino bootlader, but to make firmware update from sd, i want to jump to address end of stm32f103re before eeprom usage... How can I make applicaction for this bootloader, and how can I jump to this application address in application ?

fpistm, 2019-10-17:
Hi @serkanc 
here it is a place for STM32 core issue.
This is not a forum to talk about user application
The core provide a way to relocate binary (VECT_TAB offset) and related linker script that's all.
Your use case is not specifically related to Arduino API but more on HAL/LL and CMSIS usage.
This would be a god topic for the stm32duino.com forum but it is currently down (should be up and running befor end of the year)  so I advise you to ask on ST community in a generic way.

closed 2019-10-17 by fpistm

serkanc, 2019-10-18:
https://github.com/stm32duino/Arduino_Core_STM32/issues/524
I have done everything according to this instructions.  In Arduino I'm using this code

typedef void (*funcPtr)(void);
  uint32_t jumpAddr = usrAddr + 0x04;
  funcPtr usrMain = (funcPtr) jumpAddr;
  SCB_BASE->VTOR = usrAddr;
  __asm__("ldr r0, =0x08020000");
  __asm__("msr msp, r0");

But I cannot jump to this address ?