stm32duino/Arduino_Core_STM32 #708: stm32duino in Arduino on Raspberry Pi

odo2063, 2019-10-17:
**Is your feature request/improvement related to a problem? Please describe.**
yes

**Describe the solution you'd like**
Being able to install stm32duino in Arduino on Raspberry Pi.

**Describe alternatives you've considered**
none

**Additional context**
It would be nice to be able to compile and flash stm32duino boards from a Raspberry Pi. At the moment we doing a longterm test where all devices are only available by wifi or GSM. So it would be nice to be able to implement improvements in field. It would also great in schools. the raspi4 is powerful enough to be cheap desktop for schoolusage and then it would be nice if your boards could be used besides it.

fpistm, 2019-10-18:
Hi @odo2063,
As I do not have a Raspberry, I will not be able to work on this stuff.
I guess the main works is to have an arm-none-eabi-gcc toolchain for this target and also check all tools used by the core for upload.

If anyone want contribute to this this will be greatly appreciated.

odo2063, 2019-10-18:
the arm-none-eabi-gcc seems easy...
https://packages.debian.org/buster/gcc-arm-none-eabi

What else do you need?

i could also give you SSH to Raspi if you like...or you find something with https://www.st.com/content/st_com/en/products/microcontrollers-microprocessors/legacy-mpus/spear-arm-cortex-a9-microprocessors.html

fpistm, 2019-10-18:
Thanks but I will not have access with SSH due to security policies.
About arm-none-eabi-gcc this require several investigations:
* Which one have to be used depending of the host arch? Honestly, I don't know Raspi ecosystem, what I know is Arduino seems able to handle twice of them (arm64 and armhf)
 * arm64
 * armel
 * armhf
* Which options has been used to build ? (libraries provided? lto support?....), moreover this is not the same version and unfortunately I don't think Arduino IDE was able to deal with as we are able to only reference one.
* Test if all options used is well supported
* Need to be repackage to be installed thanks Arduino but I'm not be able to test.
* ...

> What else do you need?

All tools (binaries and scripts) tested and updated in the https://github.com/stm32duino/Arduino_Tools  repo.

odo2063, 2019-10-21:
Hi I found this one 
https://github.com/vanbwodonk/gcc-arm-embedded-build-armhf

odo2063, 2019-10-21:
> Which one have to be used depending of the host arch? Honestly, I don't know Raspi ecosystem, what I know is Arduino seems able to handle twice of them (arm64 and armhf)

For now consider Raspian as standard Debian on armhf....this will maybe change to arm64 for the Raspi4 in the future

odo2063, 2019-10-21:
```
./dfu-util --version
dfu-util 0.8
```
...seems to compile

I can run any script you want me to...but I am a newbie when it comes to build up a toolchain with my own hands...

fpistm, 2019-10-21:
Currently, I've no time frame to investigate this at least until next year.
If anyone wants try to work on this will be welcome.

koendv, 2019-12-01:
Just FYI:
Raspberry already has the following debian packages:

- binutils-arm-none-eabi 
- gcc-arm-none-eabi
- libnewlib-arm-none-eabi libnewlib-dev 
- libstdc++-arm-none-eabi-newlib 

After installing these packages, you can compile stm32 code on Raspberry. Eg. you can compile this: [gdb debugger for blue pill](https://github.com/blacksphere/blackmagic). 
What's missing (I think) is the STM32 headers in the arduino IDE.

fpistm, 2019-12-02:
@koendv 
the main issue is to have n all in one toolchain package (aligned with the gcc version used by other host OS) and as I said I could not test it. So I could integrate it if someone is able to provide and test it.

koendv, 2019-12-02:
Fine. If you provide a STM32 board package that assumes arm-none-eabi-g++, arm-none-eabi-ld, dfu-util, ... are in /usr/bin, I can try to compile some examples ("blink") and get the examples running on black pill/blue pill. OK?

fpistm, 2019-12-02:
When I said  "integrate", I mean store it on GitHub and reference it in the boards JSON files.
Not provide it.

koendv, 2019-12-04:
OK.  odo2063, if I put an arm-none-eabi toolchain up, are you willing to test it?

odo2063, 2019-12-05:
I could give you SSH to a raspi if you like...with X-Forwarding...

koendv, 2019-12-05:
OK. Your turn. https://github.com/koendv/stm32duino-raspberrypi

gcc-arm-none-eabi-9-2019-q4-major, dfu-util 0.9, hid-flash 2.2.1, built for armv7l-linux-gnu (raspberry in 32-bit mode). Build takes 16.5 hours. I could recompile this for aarch64 if needed.

odo2063, 2019-12-06:
I'll compile it that weekend...can we meet at IRC or something like that so I can hand you over the login credentials, because I don't know what to do any further?

koendv, 2019-12-06:
OK. I've got a raspberry, I've already compiled the toolchain, and you can download toolchain and stm32tools it from my github. What comes next is installing toolchain in the arduino ide.

If I look at the package_stm_index.json file, there are 4 things needed for a functioning toolchain. 

- "STM32 Cores 1.7.0"
can be used mostly as-is. Maybe need to edit platforms.txt because compiler version changed.
- "arm-none-eabi-gcc"
Already compiled for raspberry. Download from my github. Probably need to change the name of the top-level directory so the naming convention is similar to 
- "STM32Tools"
Already compiled for raspberry. Download from my github. 
- "CMSIS"
no changes needed.

There are some issues here: it seems arduino packages need to be signed before they can be installed. And because I've compiled arm-none-eabi-gcc-9.2.1 you'll probably need to change "8.2.1-1.7" in platforms.txt to "9.2.1". But, summarizing: it's no longer a problem of *compiling*, it's a problem of packaging and installing. Perhaps it would be best if some small .json package index file were created for the arm/raspberry platform, for testing.

I will try to compile toolchain and tools for aarch64 this weekend.

odo2063, 2019-12-07:
i got a raspi4 ;-) and if this goes productive we could do a crosscompile...

fpistm, 2019-12-08:
You can use the platform.local.txt to override some config.
About the toolchain, thanks the help of @ilg-ul, I moved to 9.2.1. He is currently thinking to provide an toolchain for arm host, see https://github.com/stm32duino/Arduino_Core_STM32/issues/708

koendv, 2019-12-08:
OK. That is good news. You can download the [gcc-arm-none-eabi toolchain and STM32Tools](https://github.com/koendv/stm32duino-raspberrypi/releases) for armv7l (raspberry pi in 32-bit mode) and aarch64 (raspberry pi in 64-bit mode). There's also [build notes](https://github.com/koendv/stm32duino-raspberrypi/blob/master/README.md). Anything else needed?

ilg-ul, 2019-12-08:
I did some research and I learned that the first time arm64 was supported by a major Debian distribution was Debian 9 Stretch (`ldd --version` -> 2.24), and Ubuntu 16.04 (which unfortunately picked a slightly older `ldd --version` -> 2.23).

Due to the relative order of the system libraries versions, I would say that binaries built on Debian 9 might fail on Ubuntu 16.

Thus, although personally I'd prefer to base my binaries on Debian distributions, it is safer to go for the oldest, which means that for Arm we'll probably have to use Ubuntu 16.04.

This is in line with the Arm build procedure, which also uses Ubuntu 16.04 for the `aarch64` binaries, available starting with 9-2019-q4-major.

And, for consistency reasons, I'd use the same 16.04 for both 32 and 64-bit variants (`armhf` and `aarch64`).

Any comments on this?

For the moment I installed an Ubuntu 16.04 arm64 on QEMU and I'll use it to update my xPack build scripts to add support for Arm binaries.

Later on I'll probably need a hardware solution, since I expect the build times to be unbearable long in emulation.

koendv, 2019-12-08:
OK. Estimate 3 days build time. 
Does this imply the STM32Tools have to be recompiled under Ubuntu?

ilg-ul, 2019-12-08:
> I guess the advantage will be the 32-bit build will be the more general `armhf`, and not the raspberry-specific `armv7l`.

right

> Does this imply the STM32Tools have to be recompiled under Ubuntu?

I know nothing about these tools, but if stm32duino is to run on generic Arm 32-bit machines, i guess they have to be  `armhf` too.

The comment about avoiding the raspberry-specific `armv7l` is a good point and should be a gain if indeed `armhf` binaries run on all Raspberry Pis, large and small.

But the question has another meaning too, choosing Ubuntu 16.04, anything with `ldd --version` older than 2.23 probably will not work. I have no idea if there are such systems, for 64-bit probably not, but for 32-bit there might be, I really have no idea.

On the other side, given that making a build environment based on the old Ubuntu 16 is already difficult, and going even further back in time to Ubuntu 14 is almost 'mission impossible', if there are such very old Arm 32-bit systems not able to run Ubuntu 16 `armhf` binaries, I'm afraid they'll be collateral damages...

fpistm, 2019-12-09:
> > Does this imply the STM32Tools have to be recompiled under Ubuntu?
> 
> I know nothing about these tools, but if stm32duino is to run on generic Arm 32-bit machines, i guess they have to be `armhf` too.

Yes some binaries will have to be built for the targeted hosts (ex: DFU utils).

> But the question has another meaning too, if choosing Ubuntu 16.04, anything with `ldd --version` older than 2.23 probably will not work. I have no idea if there are such systems, for 64-bit probably not, but for 32-bit there might be, I really have no idea.
> 
> On the other side, given that making a build environment based on the old Ubuntu 16 is already difficult, and going even further back in time to Ubuntu 14 is almost 'mission impossible', if there are such very old Arm 32-bit systems not able to run Ubuntu 16 `armhf` binaries, I'm afraid they'll be collateral damages...

I'm not aware of arm based distribution anyway I guess supporting all possibles arch is not possible. 
So, I guess your proposal seems reasonable and aligned with ARM choice to use Ubuntu 16.04

koendv, 2019-12-09:
OK. The latest version of dfu-utils is v0.9. If we have to recompile for armhf, let's compile the latest version, so we don't have to revisit the subject for a year or longer.

ilg-ul, 2019-12-09:
> seems reasonable and aligned with ARM choice to use Ubuntu 16.04

Good, we'll stick to Ubuntu 16, ldd --version -> 2.23 for both `armhf` and `aarch64` for now.

> Yes some binaries will have to be built for the targeted hosts (ex: DFU utils).

Probably it would be good to build them using the same Ubuntu 16. The binaries do not need to be build necessarily on a Pi, since I'm not sure there is such an old official Ubuntu 16 running on Pi, but can also be build on QEMU, for usual projects the speed should not be a limitation, as it is for toolchains; I'll publish the details of my build environments on a separate page, for those who want to play with it.

beta-tester, 2020-05-21:
i just tried to install STM32 Cores 1.9.0 on my Raspberry Pi
(newest Raspbian Buster updated and upgrades and Arduino IDE 1.8.12).

but i get the error message `Tool STM32Toolsis not available for your operating system.`

does it mean that there is still no support for STM32 on a Raspbian Pi or am i missing something?

ESP32 and Digispark support i could add sucessfully to Arduino IDE on my RPi.
ir would be nice to have STM32 as well.

koendv, 2020-05-21:
On Thu, 21 May 2020 04:13:49 -0700
beta-tester <notifications@github.com> wrote:

> i just tried to install STM32 Cores 1.9.0 on my Raspberry Pi
> (newest Raspbian Buster updated and upgrades and Arduino IDE 1.8.12).
>
> but i get the error message `Tool STM32Toolsis not available for [...]

I've just updated. Maybe you tried to install just when I was updating
the git?

koen

beta-tester, 2020-05-21:
i deleted every stm related stuff from my RPi and re-tried the process:

- in Arduino IDE i added the link to the package json file:
[image]
- and then tried to install the STM32 Cores:
[image]
but the installation does not go on, because of the message:
`Tool xpack-arm-none-eabi-gcc is not available for your operation system.`
(the message now is different to the previous one - maybe i messed too much around previously)

do i have to compile something before i can continue installation?

beta-tester, 2020-05-21:
i think i got it... i took the regular link for package json
(`https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json`).

when i take the raspberry pi specific link
/`https://raw.githubusercontent.com/koendv/stm32duino-raspberrypi/master/BoardManagerFiles/package_stm_index.json`)
the installation succeeds.

koendv, 2020-05-21:
On Thu, 21 May 2020 08:21:16 -0700
beta-tester <notifications@github.com> wrote:

> i think i got it... i took the regular link for package json
> (`https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json`).
>
> when i take the raspberry pi specific link [...]

OK. Good.

koen

RandoSY, 2020-08-20:
I've been experimenting with the arduino-cli (Arduino command line tool) on the Raspberry Pi, seems OK to build ATMEL SAMD Arduino programs. 

Has anyone tried that approach? Also gcc-arm-none-eabi is a available in the the Buster repo...  Works great with 
[Piconomixs FW](https://github.com/piconomix/piconomix-fwlib](url))

I was able to build the large Piconomixs "cli-explorer" app on a tiny 256 mb of ram, Raspberry Pi 1!

[image]

Took a while, but compiled perfectly.

ekawahyu, 2020-09-27:
@RandoSY I was testing arduino-cli on Raspberry Pi 3 model B and had some issue with insufficient memory during installation of stm32duino. But that has been fixed with https://github.com/arduino/arduino-cli/issues/975 (in case you wanted to test it on your Raspberry Pi 1 and see if you can give them feedback about the fix on your Raspi?)

I don't have Raspberry Pi 1, but I am gonna try to do it on Raspberry Pi Zero W

ilg-ul, 2020-10-12:
> but the installation does not go on, because of the message:
> Tool xpack-arm-none-eabi-gcc is not available for your operation system.

There must be something wrong in your configurations, since the arm-none-eabi-gcc xPack **is** available on most platforms.

ilg-ul, 2020-10-12:
BTW, any progress with stm32duino on Arm platforms? You no longer have the excuse of the missing toolchain, the Arm versions are available for several months already.

Not to mention that Eclipse Embedded CDT is also available for Aarch64 for some time.

fpistm, 2020-10-13:
Currently the toolchain for arm is not provided but will be for the 2.0.0.
The ARM support will be added don't worry but this required some update on tools part which required some time and tests. Moreover I'm busy on other task.
So, this will come be patient. 😉

ilg-ul, 2020-10-13:
> Currently the toolchain for arm is not provided but will be for the 2.0.0.

In 2.0.0 please be sure you use the correct 32/64-bit version.
And could you also switch the Windows toolchain to the 64-bit version?

fpistm, 2020-10-13:
> And could you also switch the Windows toolchain to the 64-bit version?

Well, currently I don't know if it should be switch, unfortunately some user always run on 32 bits version.

koendv, 2020-10-15:
> ARM support ... required some update on tools part
This sounds as if ST-Link will be available on raspberry?

ilg-ul, 2020-10-15:
> some user always run on 32 bits version.

Is Arduino for Arm able to differentiate between 32/64-bit systems and install the proper toolchain and tools?

fpistm, 2020-10-15:
>  for Arm able to differentiate between 32/64-bit systems and install the proper toolchain and tools?

Unfortunately not. Only we are on Windows else I would deploy both toolchain 32/64-bit.
I've checked which arm version they provide with the new core based on MBed for the new portenta H7 board and it's a v7 win32. So I guess I will kept win32 release.

ilg-ul, 2020-10-15:
Sorry for the misunderstanding, the question was related to Linux Arm, I currently do not have Windows Arm binaries for the toolchain.

So, is the **Linux** Arm version of Arduino able to differentiate between 32/64-bit systems?

fpistm, 2020-10-15:
Sorry read to quickly. yes for ARM host OS this is OK for both system.

ilg-ul, 2020-10-15:
Great, thank you. 

Please be sure you configure the correct toolchain for 32/64-bit Arm Linux systems, since they're both available.

BennehBoy, 2021-03-14:
@koendv silly question, but I'm guessing that rpi 1 isn't supported by your package file?  I get segfaults and illegal instructions so I guess I've got the armv6 versions?  I wasn't seriously going to use a pi 1 to develop on, just had one lying about so decided to dust it off....

koendv, 2021-03-15:
On Sun, 14 Mar 2021 13:50:03 -0700
Ben Anderson ***@***.***> wrote:

> @koendv silly question, but I'm guessing that rpi 1 isn't supported
> by your package file?  I get segfaults and illegal instructions so I
> guess I've got the armv6 versions?  I wasn't seriously going to use a
> pi 1 to develop on, just had one lying about so decided to dust it
> off....
> 

No, AFAIK you'd have to write the board package for rpi1 yourself.
I'd type gcc -v on the rpi, to get the compiler options that were used
to build gcc for the rpi1, compare that with the compiler options of
the gcc used for the arduino build, and work from there.

koen

BennehBoy, 2021-03-15:
OK, thought as much.  The toolchain does get installed just I guess the embedded logic doesn't make any distinction between rpi1 & 2+ (just 32 & 64 bit).

BennehBoy, 2021-03-15:
Works lovely on an rpi4 - but I did have to amend the rename the maple_upload bash script to maple_upload.dh and also amend line 39 to point to the correct location and filename for dfu-util.

`  "${DIR}/dfu-util/dfu-util" -d "${usbID}" -a "${altID}" -D "${binfile}" ${dfuse_addr} -R`

Works a charm afterward.

fpistm, 2021-05-21:
Just FYI, latest release now provides the arm none eabi toolchains for arm host:
https://github.com/stm32duino/BoardManagerFiles/blob/62a6f583bd7a10192f57e7e3afe729a12f73a028/package_stmicroelectronics_index.json#L74-L87

Tools will come later.

ilg-ul, 2021-05-21:
Koen also has a build with the new Arduino IDE for AArch64.

Emeric-dufour, 2021-07-28:
Hello all,

I have trouble with the combination of Arduno-cli, a raspberry pi 4b and stm32duino. 
Installing STM32duino gives "Error during install: tool STMicroelectronics:STM32Tools@2.0.0 not available for the current OS".

# Bug Report

### Current behavior

Behavior is as shown.

While trying to install core STMicroelectronics:stm32, the packages download proceeds as expected, and all the arm-none-eabi-gcc packages are correctly downloaded.

But the STM32Tools download that follows is leading to an error: 
"Error during install: tool STMicroelectronics:STM32Tools@2.0.0 not available for the current OS". 

This makes it impossible for me to install stm32duino.

```
hlfc@hlfc-4:~ $ arduino-cli core install STMicroelectronics:stm32
Downloading packages...
STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 774.28 KiB / 140.30 MiB    STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 1.85 MiB / 140.30 MiB    1.STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 3.02 MiB / 140.30 MiB    2.STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 4.15 MiB / 140.30 MiB    2.STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 4.15 MiB / 140.30 MiB    2.STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 4.99 MiB / 140.30 MiB    3.STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 6.22 MiB / 140.30 MiB    4.STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 7.37 MiB / 140.30 MiB    5.STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 8.33 MiB / 140.30 MiB    5.STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 8.33 MiB / 140.30 MiB    5.STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 9.33 MiB / 140.30 MiB    6.STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 10.46 MiB / 140.30 MiB    7STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 11.52 MiB / 140.30 MiB    8STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 12.57 MiB / 140.30 MiB    8STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 12.57 MiB / 140.30 MiB    8STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 13.63 MiB / 140.30 MiB    9STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 14.68 MiB / 140.30 MiB   10STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 15.79 MiB / 140.30 MiB   11STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 17.10 MiB / 140.30 MiB   12STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 17.10 MiB / 140.30 MiB   12STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 18.40 MiB / 140.30 MiB   13STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 19.65 MiB / 140.30 MiB   14STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 20.82 MiB / 140.30 MiB   14STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 21.80 MiB / 140.30 MiB   15STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 21.80 MiB / 140.30 MiB   15STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 22.74 MiB / 140.30 MiB   16STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 24.30 MiB / 140.30 MiB   17STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 25.35 MiB / 140.30 MiB   18STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 26.44 MiB / 140.30 MiB   18STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 26.44 MiB / 140.30 MiB   18STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 27.49 MiB / 140.30 MiB   19STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 28.65 MiB / 140.30 MiB   20STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 29.66 MiB / 140.30 MiB   21STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 30.72 MiB / 140.30 MiB   21STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 30.72 MiB / 140.30 MiB   21STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 31.96 MiB / 140.30 MiB   22STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 33.16 MiB / 140.30 MiB   23STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 34.44 MiB / 140.30 MiB   24STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 35.52 MiB / 140.30 MiB   25STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 35.52 MiB / 140.30 MiB   25STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 36.66 MiB / 140.30 MiB   26STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 37.66 MiB / 140.30 MiB   26STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 38.63 MiB / 140.30 MiB   27STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 39.32 MiB / 140.30 MiB   28STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 40.71 MiB / 140.30 MiB   29STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 40.71 MiB / 140.30 MiB   29STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 41.13 MiB / 140.30 MiB   29STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 41.46 MiB / 140.30 MiB   29STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 43.49 MiB / 140.30 MiB   31STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 44.58 MiB / 140.30 MiB   31STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 45.76 MiB / 140.30 MiB   32STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 45.76 MiB / 140.30 MiB   32STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 46.83 MiB / 140.30 MiB   33STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 47.80 MiB / 140.30 MiB   34STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 48.86 MiB / 140.30 MiB   34STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 49.96 MiB / 140.30 MiB   35STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 49.96 MiB / 140.30 MiB   35STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 51.02 MiB / 140.30 MiB   36STMicroelectronics:xpack-arm-none-eabi-gcc@9.3.1-1.3 52.02 MiB / 140.30 MiB   37STMicroelectr [... cut ...]
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
```
hlfc@hlfc-4:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```
```
hlfc@hlfc-4:~ $ uname -a
Linux hlfc-4 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux
```

If somebody has some advice for me, that would be amazing thanks!

ilg-ul, 2021-07-28:
> xpack-arm-none-eabi-gcc@9.3.1-1.3 

BTW, since December 2020 there is also a [10.x](https://xpack.github.io/blog/2020/12/19/arm-none-eabi-gcc-v10-2-1-1-1-released/) toolchain release.

camelator, 2022-07-25:
Hi,
not sure to understand the problem... so sorry if my answer is not relevant or stupid:
in my configuration, I am using plateformio with a black magic probe flashed on a STM32F103. By this way, you can use the IDE you like, the hardware you like (raspberry, etc...) and the framework you like (STM32 duino, ...). The black magic probe is linked to the hardware by USB and allow debugging.
as far I know, it is working on raspberry.

fpistm, 2022-07-25:
Hi @camelator 
here we talk about Arduino IDE not PIO.

fpistm, 2025-12-04:
Hi @ilg-ul 
right linux oriented.
Thanks for the insight.