stm32duino/Arduino_Core_STM32 #711: HardwareTimer prevents overriding HAL_TIM_PeriodElapsedCallback

benlye, 2019-10-20:
**Describe the bug**
Attempting to override `HAL_TIM_PeriodElapsedCallback` gives a compiler error as `HAL_TIM_PeriodElapsedCallback` is defined in HardwareTimer.cpp without `__weak` definition.

**To Reproduce**
Create a simple sketch which defines an override function for `HAL_TIM_PeriodElapsedCallback` and attempt to compile.

For example, adding this function in the sketch:

```
void HAL_TIM_PeriodElapsedCallback(TIM_HandleTypeDef *htim)
{
  if (htim->Instance == TIM2)
  {
    HAL_GPIO_TogglePin(GPIOA, GPIO_PIN_1);
  }
}
```

Results in this error:
```
c:/users/blye/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\blye\AppData\Local\Temp\arduino_build_330887\core\core.a(HardwareTimer.cpp.o): in function `HAL_TIM_PeriodElapsedCallback':

HardwareTimer.cpp:(.text.HAL_TIM_PeriodElapsedCallback+0x0): multiple definition of `HAL_TIM_PeriodElapsedCallback'; C:\Users\blye\AppData\Local\Temp\arduino_build_330887\sketch\ArduinoBlink1.ino.cpp.o:ArduinoBlink1.ino.cpp:(.text.HAL_TIM_PeriodElapsedCallback+0x0): first defined here

collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic STM32F1 series.
```

**Expected behavior**
Compiler should compile the code without error

**Desktop (please complete the following information):**
 - OS: Windows
 - Arduino IDE version: 1.8.10
 - STM32 core version:1.7.0
 - Tools menu settings if not the default: Serial Disabled
 - Upload method: N/A

**Board (please complete the following information):**
 - Name: Blue Pill STM32F103

**Additional context**
Add any other context about the problem here.

benlye, 2019-10-20:
Related to #697, I guess.

fpistm, 2019-10-21:
`HAL_TIM_PeriodElapsedCallback ` is already a weak in the HAL, that's why it can't be a weak in Arduino" layer. 
And yes it is the same issue than #697.

closed 2019-10-21 by fpistm

benlye, 2019-10-21:
So I should wait for a solution to the other issue, or is there a workaround I can use until then?

fpistm, 2019-10-21:
The other/ thisissue is a feature request, this is not a bug. 
This is a community project, any contribution are welcome to extend advanced user usage.

Workaround can be to comment the one from HardwareTimer but all dependent feature will be broken.

mrguen, 2019-12-07:
@benlye do you have a workaround for this? I am also looking to execute a handler more directly.

benlye, 2019-12-07:
@mrguen my workaround was to disable the HAL_TIM module altogether by adding a hal_conf_extra.h file containing `#define HAL_TIM_MODULE_DISABLED`.

mrguen, 2019-12-07:
@benlye ok thank you very much.
 I was wondering why TIM1_IRQHandler() was not executed... This should be appropriate since I use the TIM1 and TIM2 instances directly, so I can position any register value.

fpistm, 2019-12-07:
#697 is fixed you could use all HAL TIM now

mrguen, 2019-12-07:
@fpistm does it mean this handler 

void TIM1_IRQHandler(void) {
  Serial.println("IRQ");
  cptBoucle++;
}

should be executed without registering any handler but just positioning  TIM1->DIER = TIM_DIER_UIE;

mrguen, 2019-12-07:
@benlye how do you activate the IRQ / does TIM1->DIER = TIM_DIER_UIE; is enough? / I can't make it work.

mrguen, 2019-12-07:
Ok, I have got it 

Enable TIM1 update IRQ:

```
  TIM1->DIER = TIM_DIER_UIE;                          // enable interrupt on update
  NVIC_EnableIRQ(TIM1_UP_TIM10_IRQn);        // enable IRQ
  asm volatile("cpsie i");                                      // activates global interrupt ( if needed )
```

And the handler

```
extern "C" {
    void TIM1_UP_TIM10_IRQHandler(void) {
    if (TIM1->SR & TIM_SR_UIF) {
      // Reset the update interrupt flag
      TIM1->SR &= ~(TIM_SR_UIF);
      // do something
    }
  }
}
```