stm32duino/Arduino_Core_STM32 #712: clean chip STM32f103c8t6 how to

tatoswinka, 2019-10-21:
Hallo. I'm very sorry for writing this topic here but I can't find a solution to the problem. I have a clean chip STM32f103c8t6. I program it with ST-Link v2. If I create a "blink" code using STM32CubeMX and Kein uVision 5 everything is ok. If I create a code using the Arduino IDE, unfortunately it doesn't work. I suppose the chip must be properly configured for the code from the Arduino IDE. How to do it ?. I am beginner and I can't use Kein. I would like to use the chip in the driver I build - without the entire BluePill board. Regards Robert

fpistm, 2019-10-21:
BluePill variant is mainly a pins mapping. This should works for your board as it is based on the same MCU.
If the blink does not work, probably you do not have the same XTal (value, type) than the BluePill, so simply implement your own `SystemClock_Config` as it is a weak function:
https://github.com/stm32duino/Arduino_Core_STM32/blob/2b723f85337a0684573de03a9d471f861f271435/variants/PILL_F103XX/variant.cpp#L105

BluePill use the HSE as input clock with a value of 8MHz.
This value can also be overridden, see:
https://github.com/stm32duino/wiki/wiki/HAL-configuration#other-hal-configuration

Finally you can do your own variant to get the pins mapping you want:
https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board)

tatoswinka, 2019-10-21:
Thank you very much for the quick reply, but I think I'm too weak to run it. I have boards installed in the Arduino IDE. I choose the right clock frequency. I thought that if a board is available, the clock and pin mapping sets itself. I don't know how to understand the variant setting. I work on Windows, not Linux - the description is rather for linux. What do I have to do specifically? Please, if possible, explain to an amateur. The chip is installed on a prototype board with 8MHz quartz. I have power supply pins and pins for the ST-Link programmer. On pin number 2 (PC13) an LED is connected. The code from the Arduino IDE works on a blue board with the same chip, it does not work on my board.

fpistm, 2019-10-21:
The description is not for Linux. It is mainly file management and the only example command is 'cp' which means copy so I guess it is not so hard on Windows.

Based on your input the built-in sketch example for the BluePill F103C8  should work.
The LED_BUILTIN is set to PC13 and the clock config is for an HSE at 8MHz.
If not then I could not help as this is your custom hardware without any schematics or relevant information.

tatoswinka, 2019-10-21:
[image]

This is simple code:
`void setup() {

  pinMode(PC13, OUTPUT);
}
void loop() {
  digitalWrite(PC13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(200);               // wait for a second
  digitalWrite(PC13, LOW);    // turn the LED off by making the voltage LOW
  delay(200);

}`

Programming original bluepill:
_      -------------------------------------------------------------------
                       STM32CubeProgrammer v2.2.0                  
      -------------------------------------------------------------------

ST-LINK SN  : 57FF6C065185525327580787
ST-LINK FW  : V2J34S7
Voltage     : 3.26V
SWD freq    : 4000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x410
Device name : STM32F101/F102/F103 Medium-density
Flash size  : 128 KBytes
Device type : MCU
Device CPU  : Cortex-M3

Memory Programming ...
Opening and parsing file: BlinkSTM32.ino.bin
  File          : BlinkSTM32.ino.bin
  Size          : 13972 Bytes
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 13]
Download in Progress:

File download complete
Time elapsed during download operation: 00:00:01.297

RUNNING Program ... 
  Address:      : 0x8000000
Application is running
Start operation achieved successfully_

This is programming of clear chip:
_      -------------------------------------------------------------------
                       STM32CubeProgrammer v2.2.0                  
      -------------------------------------------------------------------

ST-LINK SN  : 57FF6C065185525327580787
ST-LINK FW  : V2J34S7
Voltage     : 3.25V
SWD freq    : 4000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x410
Device name : STM32F101/F102/F103 Medium-density
Flash size  : 128 KBytes
Device type : MCU
Device CPU  : Cortex-M3

Memory Programming ...
Opening and parsing file: BlinkSTM32.ino.bin
  File          : BlinkSTM32.ino.bin
  Size          : 13972 Bytes
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 13]
Download in Progress:

File download complete
Time elapsed during download operation: 00:00:01.221

RUNNING Program ... 
  Address:      : 0x8000000
Application is running
Start operation achieved successfully_

fpistm, 2019-10-21:
Unfortunately I will not be able to help.

tatoswinka, 2019-10-21:
I suspect it could be a Chinese fake chip and this could be a problem, was bought on aliexpress. I will disassemble by hot air from the original bluepill and solder it to my project board. Thank You for help. Regards.

stas2z, 2019-10-22:
@tatoswinka 
try to disable LTO

tatoswinka, 2019-10-22:
I will try, but I think I have already done so

stas2z, 2019-10-22:
@tatoswinka if disabling LTO will not  ressurect it, probably its smth wrong with your custom board xtal, check crystal resonator and shunt capacitors. You can transfer your cubemx  SystemClock_Config routine to bluepill variant temporary as a reference (backup variant folder before copy/pasting), i can guess yr cubemx project configured to use internal (HSI) clock without pll so everything run fine at 8mhz and dont use external crystal

tatoswinka, 2019-10-22:
I did it exactly based on this tutorial: [https://www.youtube.com/watch?v=aOjLb3T_wAg&t=111s](url)
Blinking works, an external clock of 8x9 = 72MHz is set here.
I have external 8MHz crystal (between pins 5 and 6) with capacitors 22pF.

stas2z, 2019-10-22:
@tatoswinka clock init in that tutorial is the same as done in blackpill variant
so only LTO can be an issue here

tatoswinka, 2019-10-22:
I think there is something wrong with crystal. I also have a chip STM32F103RBT6, 64pins, if I upload the code selecting such a chip does not work. If I choose Nucleo F103 64MHz - WORKS !!, If I choose Nucleo F103 72MHz - it doesn't work
WORKING:
[image]

DOES NOT WORK:
[image]

tatoswinka, 2019-10-22:
yes is possible, I will try to use 10pF in my final board.
Thanks all for help
[https://www.dropbox.com/s/d1fl1459uxnaozz/20191022_174334.mp4?dl=0](url)
50ms blinking delay

fpistm, 2019-10-23:
@tatoswinka fine
Thanks @stas2z for support.

closed 2019-10-23 by fpistm