stm32duino/Arduino_Core_STM32 #714: Please update st-link / st-flash tools for compatibility with BluePill

AndKe, 2019-10-21:
updating these two tools fixes a lot of problems:  (copying freshly compiled version)

```
andre@loke:~$ cp /usr/local/bin/st-info .arduino15/packages/stm32duino/tools/stm32tools/2019.10.5/linux/stlink/st-info 
andre@loke:~$ cp /usr/local/bin/st-flash .arduino15/packages/stm32duino/tools/stm32tools/2019.10.5/linux/stlink/st-flash
```

fpistm, 2019-10-21:
Those tools are no more supported with this core since 1.5.0.

AndKe, 2019-10-21:
My stm32duino library for Arduiono seems to be version 2019.10.9 , what exactly is not supported since what was 1.5.0 ?

fpistm, 2019-10-21:
I understood, you use Roger core (https://github.com/rogerclarkmelbourne/Arduino_STM32) packaged here:
http://dan.drown.org/stm32duino/package_STM32duino_index.json
where we find 2019.10.5 version of the stm32tools.

closed 2019-10-21 by fpistm

AndKe, 2019-10-21:
Now I see that this is not the project I used.
I understand that this project does more or less the same job, most likely better, as it seems to me STM32duino is no longer developed?

Should I uninstall/remove the stm32duino boards, before removing it, and replacing with this project ?
Please advise.
Thank you.

fpistm, 2019-10-21:
You could have both installed.
Each core is different. This one support all series and is more Arduino compatible.