stm32duino/Arduino_Core_STM32 #720: STM32F103C8  64/128k devices missing third serial port.

AndKe, 2019-10-22:
I see option for Serial and Serial1 , yet a BluePill should have three USARTs - what is going on here ?
I wish there were a spreadsheed/table that shows supported features of different ST's ?

stas2z, 2019-10-22:
Create build_opt.h file in yr sketch dir with following content
`-DENABLE_HWSERIAL2 -DENABLE_HWSERIAL3 `

fpistm, 2019-10-23:
Hi  @AndKe 
All U(S)ART from a STM32 are managed.
By default Arduino instantiate one named `Serial` which is mapped to the one defined in the variant:
https://github.com/stm32duino/Arduino_Core_STM32/blob/2b723f85337a0684573de03a9d471f861f271435/variants/PILL_F103XX/variant.h#L111

For BP USART1 is used, so `Serial1` is instantiate and mapped to `Serial`. If you enable USB CDC to superseded Serial and kept Serial enabled  then `Serial1` is always instantiate and `SerialUSB` is mapped on `Serial`

If you need more HW Serial then see:
https://github.com/stm32duino/wiki/wiki/API#hardwareserial

closed 2019-10-23 by fpistm

fpistm, 2019-10-24:
I've just tested and it works.
Regarding your log:
`Using previously compiled file:`
As it is specified in the wiki for build_opt.h, if you already build before add it then it is ignored.
Close and reopen Arduino IDE.

AndKe, 2019-10-24:
I am embarrassed, but happy.  - thanks for the swift help