stm32duino/Arduino_Core_STM32 #721: STM32WB - no interrupts from SysTick_Handler() when run from OTA bootloader.

denisn73, 2019-10-23:
Please help me.
I assembled the project in IAR, it works fine.
I want to add the OTA bootloader, for this I place the main program and vectors table at addr 0x8007000, and at address 8x0000000 I load the OTA bootloader programm from STM32Cube examples for stm32wb series.
After switching from the bootloader to the main program, the clock is normally configured, here is no error, all HAL_OK, but the function HAL_getTick() return "0", it is no interrupts from SysTick_Handler(). As a result, the millis() function always returns 0.
What could be wrong with timing configuration and how to set it up correctly?
P.S.: Coprocessor already programmed with latest BLE stack.
Company: "Vega-Absolute, Russian, Novosibirsk"

fpistm, 2019-10-23:
Hi  @denisn73 
could you be more precise. about how you achieve this?
I guess you update the ldscript.ld?

denisn73, 2019-10-23:
no, I did not use ldscript. I manually assembled the project in IAR, including all the necessary files, and I used the startup_stm32wb55xx_cm4.s file. I set the start address through the linker file stm32wb55xx_flash_cm4.icf

fpistm, 2019-10-23:
Unfortunately, we support only Arduino not IAR.
So I will not able to help.

closed 2019-10-28 by fpistm