stm32duino/Arduino_Core_STM32 #722: [New Variant] request list

fpistm, 2019-10-23:
Hereafter the list of all new variants requested

**Board support requested:**
 - [ ] [STM32F469iDiscovery](https://www.st.com/en/evaluation-tools/32f469idiscovery.html) (#571)
 - [ ] [STM32F407ZGT6 Mini](https://stm32-base.org/boards/STM32F407ZGT6-VCC-GND-Large) (#635)
 - [ ] [STM32F103ZET6 Mini](https://item.taobao.com/item.htm?&id=41867070778) (#635)
 - [ ] [NUCLEO-L010RB](https://www.st.com/en/evaluation-tools/nucleo-l010rb.html) (#662)
 - [ ] [NUCLEO-F746ZG](https://www.st.com/en/evaluation-tools/nucleo-f746zg.html) (#715)

**Note that user can add a STM32 based board following this [wiki](https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board))**

Any contribution is welcome, so do not hesitate to submit a PR.

r-pv, 2019-10-23:
Hi, could you expand a bit the **Add new board** wiki with info about how to set properly the analogs?, because I'm having problems with my variant for the [F030C8](https://github.com/r-pv/rSTM32F030C8T6_ProMiniPlus/tree/master/RBOARD_F030C8T6), when I tried to read PA0 ( analogRead(PA0) ) it really reads the PA3, for the moment I solve it by using analogRead(PA_0). So I was under the impression that only has to match the:
NUM_ANALOG_FIRST on the variant.h file with the duplicates on the digitalPin array on variant.cpp, but it seems that may be more to it, right?. Thanks.

Hedda, 2019-10-28:
Request support for STM32MP1 (STM32 MP1) that comes onboard STM32MP157A-DK1 development board

https://www.st.com/en/evaluation-tools/stm32mp157a-dk1.html

STM32MP157A-DK1 developmnet board also comes in STM32MP157C-DK2 discovery kit

https://www.st.com/en/evaluation-tools/stm32mp157c-dk2.html

STM32MP157A-DK1 and STM32MP157C-DK2 Discovery kits are $99 with LCD or $69 without
https://www.st.com/en/evaluation-tools/stm32mp157a-dk1.html
https://www.st.com/en/evaluation-tools/stm32mp157c-dk2.html

STM32MP157A-EV1 Evaluation kit for PCB engineers does have a steeper price at $399
https://www.st.com/en/evaluation-tools/stm32mp157a-ev1.html

Several other boards with STM32MP1 have been announced but none released yet AFIK.

justind000, 2019-10-29:
Picked up [STM32G0316-DISCO](https://www.st.com/en/evaluation-tools/stm32g0316-disco.html) the other day. It'd be nice to port over. It's close to an ATTiny85 alternative.

MCUdude, 2019-11-18:
It would be great if we could have more generic chip options. I'm particularly interested in the 48 and 64 pin versions of the STM32F401 and STM32F411, but these will be used on custom hardware and not board like the Nucleo. The F401 and F411 are also very reasonably priced on LCSC.com.

I've been a pro AVR user for years now, but have very experience with STM32. Let's say I start working on a pin definition file for all 48-pin F401C's. Can we use the same pin definition files for all variants (F401CB/CC/CD/CD)? And are there guidelines on how the Arduino pinout should look like? Perhaps similar/identical to other supported STM32's?

kbumsik, 2019-11-18:
@MCUdude 
As far as I understand about this project, it can support all generic chips, since it pulled from official STM32 Cube libraries.
The problem is that it is just impossible to define generic pinouts for all possible chips because Arduino platform flatten the pinout definitions (like PA14 => Arduino Digial 1 pin). You should modify the source code to define your own board.
The good news is that you don't need to do much works since Arduino_Core_STM32 already supports STM32F401 Series, so you only need to define a new [variant](https://github.com/stm32duino/Arduino_Core_STM32/tree/master/variants). You can see [the wiki](https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board)) to figure out how to define your own variant.

> Can we use the same pin definition files for all variants (F401CB/CC/CD/CD)?

You should look at the datasheet to see if they have the same pinouts, but as far as I know they have different memories and the same package/pinouts.

MCUdude, 2019-11-18:
> The problem is that it is just impossible to define generic pinouts for all possible chips because Arduino platform requires flattened pinout definitions (like PA14 => Arduino Digital 1 pin)

Luckily, (unlike on 8-bit AVRs) I can still use the generic pin names (PA14) instead of the Arduino Pin number. This makes the actual Arduino pinout not so relevant to me, since I will always have full control over the generic pin names.

> The good news is that you don't need to do much works since Arduino_Core_STM32 already supports STM32F401 Series, so you only need to define a new variant. You can see the wiki to figure out how to define your own variant and see here to learn how to work with the source code.

Thanks! I'll read through the Wiki and see what I can do. I prefer working in the PlatformIO IDE, so I hope a new variant for 48 pin 401/411's will be added if I do it right, since I'll rely on an official release. In PlatformIO it's also easy to change the microcontroller variant as long as I have a working variants file for the particular target. @fpistm any thought on adding more generic targets to this core?

> You should look at the datasheet to see if they have the same pinouts, but as far as I know they have different memories and the same package/pinouts.

I haven't really read through the datasheet, but according to ST's tables the only difference is flash and RAM.

kbumsik, 2019-11-18:
> Luckily, (unlike on 8-bit AVRs) I can still use the generic pin names (PA14) instead of the Arduino Pin number. This makes the actual Arduino pinout not so relevant to me, since I will always have full control over the generic pin names.

I mean, this project is Arduino Core that uses Arduino API. Even if you physically have access to the generic pin names, Arduino API forces you to use Arduino pin numbers. So if you want more flexibility of choice of pins you might have to define your own Arduino pins to work with.

MCUdude, 2019-11-18:
> I mean, this project is Arduino Core that uses Arduino API. Even if you physically have access to the generic pin names, Arduino API forces you to use Arduino pin numbers. So if you want more flexibility of choice of pins you might have to define your own Arduino pins to work with.

Yes, I'm forced to use Arduino Pin numbers. But the PA[x], PB[x]..Pn[x] macros are defined in a way that they match their Arduino Pin number. On the Nucleo 401 board I have I don't care about digital pin 13. I use the PA5 macro instead (which is defined as 13).

I have lots of experience when it comes to "wrapping" the Arduino API around AVRs. I currently host six different repos here on GitHub, and some have become quite popular.

> I don't know how PlatformIO IDE works with third-party Arduino cores, but working with the source does not require a specific IDE, since you replace the official release with the git repo.

I've been involved with platformIO since three on my Arduino cores ([MightyCore](https://github.com/MCudude/MightyCore), [MiniCore](https://github.com/MCudude/MiniCore) and [MegaCore](https://github.com/MCudude/MegaCore)) are available through PlatformIO. What they do is that they will make a copy of the latest release and include it in one of their repos. It _is_ possible to add custom core files, but it's so much easier when it is already available as an option in their project wizard.

fpistm, 2019-11-18:
@MCUdude 
You can add an GenF4 menu, this is not a problem.
About Arduino pinout this is not an issue. We only try to do a correct pins mapping when there is an Arduino connector else this is not important. As you said it is easier to use PYn naming.
Only the Analog pins have to be contiguous.
You can refers to other GenYx series added as an example.

MCUdude, 2019-11-18:
@fpistm thanks! I'll put some time and effort into this. Does any of you guys know if there exist a good ready-made F401/f411 48-pin development board? If not I'd probably have to roll my own.

It would also be great if it was possible to use the standard stlink programmer for the generic "boards", preferably the cheap $3 knockoff's. They do work great in PlatformIO, but not in Arduino IDE. I do have a black magic probe and currently waiting for a Jlink in the mail, but it's always nice to be able to use cheap and highly available tools. Is there any particular reason why support for stlink v2 was dropped?

fpistm, 2019-11-18:
STlinkv2 is supported, you have to install the STM32CubeProgrammer.

MCUdude, 2019-11-19:
Sorry for the noob questions, but I'm slowly starting to getting into things.

The Nucleo board I currently have uses the internal HFI (high-frequency internal oscillator) instead of an external crystal. To gain more accurate timing I soldered in an 8 MHz crystal, some caps and added a few 0R resistors. With a little fiddling, I was able to reconfigure the clock, and I'm now running with an external crystal! 

However, one question still remains; Why would I need a 32.768 kHz crystal as well? The Nucleo board has this crystal populated, but as far as I know, it's not in use by default? Can this be used to drive the main high-frequency clock? Is it only for RTC use? Do I really need to use this at all?

The reason I'm asking is that even the cheapest boards from China has this crystal. And they are trying to cut costs everywhere, so I figured it maybe has to be there for a reason?

fpistm, 2019-11-19:
AFAIK for F401 it i mainly used for RTC and allow to have a better accuracy than the LSI.
I guess, this also allow to reach a better consumption in some Low Power mode.
Anyway for this kind of question you should ask on ST community. Here this is for issue related to the core and this issue is to list all variant requests.

MCUdude, 2019-11-19:
> AFAIK for F401 it i mainly used for RTC and allow to have a better accuracy than the LSI.
I guess, this also allow to reach a better consumption in some Low Power mode.

Thanks for the clarification. I understand you would like to keep this on-topic, so I'll rather ask in the community forums instead, no worries!

Back to the generic pinouts, Since I only have a 64-pin (Nucleo) 32F401R chip at the moment to work with, I started working on a generic pinout for this chip first. Is it OK for you guys if I keep the peripheral names as close to the metal as possible? This means we'll have `Serial1`, `Serial2` and `Serial6` as available objects, and the user can use `setRx()` and `setTx()` to change the pin location for each peripheral? If this is documented properly (I can help out) it should be fairly easy for the end-user to configure this properly.

fpistm, 2019-11-19:
By default, only one serialx instance is defined and mapped to `Serial`.
Most of the user only use one serial. I really don't think you should instantiate all possible Serial.
Mainly if the user need to change the mapping.
It is simplest to define it than changing the default pins.

kargeor, 2020-01-03:
Can you add support for new BlackPill STM32F401CCU6?

fpistm, 2020-01-03:
Hi @kargeor 
as the F401 version exists it should not be difficult to add this one.

TzOk83, 2020-01-10:
The STM32F411CEU6 "Black Pill 2" apparently has 2 different board revisions:
WeACT V1.3 and WeAct V2.0.
The V2.0 has an additional "KEY" button, which binds pin A0 to GND when pressed.

I really can't wait support for this board...

dooodgers, 2020-03-01:
Is support for NUCLEO-H745ZI-Q planned in the near future?

BennehBoy, 2020-03-01:
> The STM32F411CEU6 "Black Pill 2" apparently has 2 different board revisions:
> WeACT V1.3 and WeAct V2.0.
> The V2.0 has an additional "KEY" button, which binds pin A0 to GND when pressed.
> 
> I really can't wait support for this board...

Do you have any links to schematics?

BennehBoy, 2020-03-01:
I just knocked up a quick Pill 411CE variant only to realise that it is already done.

There's an error in the variant.h however, it has the USER BUTTON (KEY) mapped to PC13, this is the builting LED pin, KEY is on pin A0.

fpistm, 2020-03-02:
@BennehBoy 
there is no error.
LED is well defined to PC13 and no user button is defined for the ARDUINO_BLACKPILL_F411CE.
Anyway it seems there is one user button based on schemtaics I've found anyway I didn't see it on any pictures I've saw of the board (Weact 1.3 and 2.0).

fpistm, 2020-03-02:
Hi @dooodgers 

> Is support for NUCLEO-H745ZI-Q planned in the near future?

I did not plan to add it specifically. Moreover, H745 is the Dual-core lines.
Currently there is no dedicated management for Dual-core lines.

TzOk83, 2020-03-02:
> Anyway it seems there is one user button based on schemtaics I've found anyway I didn't see it on any pictures I've saw of the board (Weact 1.3 and 2.0).

https://github.com/mcauser/WEACT_F411CEU6/blob/master/docs/STM32F411CEU6_schematics.pdf

elsni, 2020-03-02:
Support for the HY-ministm32v would be really great, is it hard to add?
https://www.hotmcu.com/hyministm32v-dev-board-32-tft-lcd-module-p-5.html

Hedda, 2020-03-02:
FYI, STMicro has updated their STM32MP1 Family with models that have faster 800 MHz Cortex-A7 Processors, through the Cortex-M4 core still runs at same 209 MHz speed as previous models.

https://www.cnx-software.com/2020/02/27/stmicro-updates-stm32mp1-family-with-800-mhz-cortex-a7-processors/

fpistm, 2020-03-02:
@elsni 

> Support for the HY-ministm32v would be really great, is it hard to add?
> https://www.hotmcu.com/hyministm32v-dev-board-32-tft-lcd-module-p-5.html

This should not be too hard. refers to the Wiki how to add a new board. You can also create your own package and use this core.

N0rbert, 2020-03-03:
I do not know the STM32 platform well for now, but it would be really great if you support the STM32F031K6T6 as included in https://www.st.com/en/evaluation-tools/nucleo-f031k6.html (it has Arduino Nano pin-to-pin headers).

N0rbert, 2020-03-03:
And having support of STM32F413ZHT6 as included in https://www.st.com/en/evaluation-tools/32f413hdiscovery.html would be great too.

fpistm, 2020-03-04:
@N0rbert 

As it is mentioned in the OP:

> **Note that user can add a STM32 based board following this [wiki](https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board))**
> 
> Any contribution is welcome, so do not hesitate to submit a PR.

stevewells20, 2020-04-13:
I too am interested in the Nucleo H745ZI. I realize it is a dual core system (Similarly to the ESP32), which is not currently supported. For a starting point, would there be a simple way to modify the existing code for the H743ZI to be compatible with the H745ZI, since they share the same Cortex M-7 CPU?

The ESP32's dual core API is quite easy to use and would be a great example to shoot for for dual core support in this project.

Hackswell, 2020-04-23:
Q1)
I have a question about the STM32F4xx definitions.  For example, in Generic_F411Cx/variants.h, it lists one of the pins as:
`#define PB9  25  // | 25      |        |           | TWI1_SDA, (TWI2_SDA) | SPI2_SS                        |           |`
However, the documentation from STM clearly states that PB9 is on pin 46 for the 48-pin package. As a matter of fact, pin 25 isn't listed for PB9 for any of the chip packages!

Q2)
In variant.cpp, why is this set to 192, when the STM32F4xx is clocked at 100MHz, not 96MHz?
`  RCC_OscInitStruct.PLL.PLLN            = 192;           // VCO output clock = 192 MHz (1 MHz * 192)`

fpistm, 2020-04-24:
@Hackswell 
I've answered you on the forum. Please avoid to duplicate your questions and use only the forum for question/support. Thanks 😉

lutfisan, 2020-11-10:
please add support for NUCLEO-L552ZE-Q

fpistm, 2020-11-10:
> 
> 
> please add support for NUCLEO-L552ZE-Q

L5 introduction is planned. The this board could be added. Added to the list

AnthonyDiGirolamo, 2020-12-15:
Hoping to get variant support for the https://www.st.com/en/evaluation-tools/32f429idiscovery.html board including setting up the 64-Mbit SDRAM.

I'm guessing one could copy most of this variant to start? https://github.com/stm32duino/Arduino_Core_STM32/tree/master/variants/NUCLEO_F429ZI to start. 

The only other variant I can see that uses SDRAM is https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/DAISY_SEED/ldscript.ld#L74 Is that a good example of how to setup SDRAM for the STM32F429I-DISC1?

fpistm, 2020-12-15:
@AnthonyDiGirolamo 
Yes it will come. But before adding any new variant I have to end the #1091 PR as all the variant will have a different structure for genericity. WiKi will be updated as well.

fpistm, 2021-01-22:
Hi @matthijskooijman 
No need to submit a PR. I'm currently reworking the variant definition for 2.0.0 release.
The Nucleo G031K8 will be part of this.

matthijskooijman, 2021-01-22:
@fpistm Any preview I could maybe test? In need to do some tests with this particular chip/board in the coming days to see if it's feasible for a product we're working on :-)

fpistm, 2021-01-22:
Yes it could be possible.
Let me few hours to clean up my branch and I will provided it to you.

P.S.: all my best wishes for this new year. :wink:

matthijskooijman, 2021-01-22:
Awesome, thanks for your lightning quick and helpful (as always) reply! Best wishes for you as well! :-)

harry1453, 2021-02-12:
Please could you add generic F7 targets, such as STM32F722, STM32F745, STM32F765, etc?

fpistm, 2021-02-12:
Hi @harry1453 
added to the list, with new release it will be lore easy to add generic variant as the main files required to create a variant is automatically generated and provided. Only clock config and ldscript need to be added.

Hedda, 2021-02-12:
### STMicroelectronics STM32 H7 (STM32H7) Discovery Kits

STM32H735G-DK Discovery Kit with STM32H735IG = https://www.st.com/en/evaluation-tools/stm32h735g-dk.html

STM32H745I-DISCO Discovery Kit with STM32H745XI https://www.st.com/en/evaluation-tools/stm32h745i-disco.html

STM32H747I-DISCO Discovery Kit STM32H747XI https://www.st.com/en/evaluation-tools/stm32h747i-disco.html

STM32H750B-DK Discovery Kit with STM32H750XB https://www.st.com/en/evaluation-tools/stm32h750b-dk.html

### STMicroelectronics Nucleo boards with STM32 H7 (STM32H7):

NUCLEO-H743ZI ( STM32H743ZI / STM32H743ZIT6U ) https://www.st.com/en/evaluation-tools/nucleo-h743zi.html

NUCLEO-H745ZI-Q ( STM32H745ZI / STM32H745ZIT6U ) https://www.st.com/en/evaluation-tools/nucleo-h745zi-q.html

NUCLEO-H755ZI-Q ( STM32H755ZI/STM32H755ZIT6U ) https://www.st.com/en/evaluation-tools/nucleo-h755zi-q.html

NUCLEO-H753ZI ( STM32H753ZI/STM32H753ZIT6U ) https://www.st.com/en/evaluation-tools/nucleo-h753zi.html

NUCLEO-H723ZG ( STM32H723ZG / STM32H723ZGT6U ) https://www.st.com/en/evaluation-tools/nucleo-h723zg.html

NUCLEO-H7A3ZI-Q ( STM32H7A3ZI / STM32H7A3ZIT6QU ) https://www.st.com/en/evaluation-tools/nucleo-h7a3zi-q.html

PS: Also see https://github.com/MarlinFirmware/Marlin/issues/19751 which has more links of both generic and custom boards with STM32 H7 (STM32H7).

Gamester17, 2021-02-15:
NUCLEO-G474RE development board has STM32G474RE MCU

https://www.st.com/en/evaluation-tools/nucleo-g474re.html

NUCLEO-G491RE development board has STM32G491RE MCU

https://www.st.com/en/evaluation-tools/nucleo-g491re.html

fpistm, 2021-02-15:
Hi @Gamester17 
Nucleo G474RE is already available.

I've added the NUCLEO-G491RE in the list.

keystoneclimber, 2021-02-23:
I'd love to see support for the STM32L4R9AI (ultra low power MCU) and it's associated STM32L4R9I-DISCO board for circular LCD development!

rei-vilo, 2021-03-05:
Seeed Studio has launched a new range of LoRa-E5 boards based on the STM32WLE5JC:

* [LoRa-E5 Development Kit - based on LoRa-E5 STM32WLE5JC](https://www.seeedstudio.com/LoRa-E5-Dev-Kit-p-4868.html)
* [LoRa-E5 mini (STM32WLE5JC) Dev Board, LoRaWAN protocol](https://www.seeedstudio.com/LoRa-E5-mini-STM32WLE5JC-p-4869.html)

Up to now, only the [STM32Cube MCU Package for STM32WL series ](https://my.st.com/content/my_st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32cube-mcu-mpu-packages/stm32cubewl.license=1608693595598.product=STM32CubeWL.version=1.0.0.html#overview) is listed as SDK. 

Adding support for those boards could broaden the user’s base and foster new applications. Thank you!

ttimasdf, 2021-03-15:
Added support for STM32F105x8/B/C. The pinmap is generated through [genpinmap](https://github.com/stm32duino/Arduino_Tools/tree/master/src/genpinmap), clock config generated by STM32Cube.

https://github.com/ttimasdf/Arduino_Core_STM32/tree/variant-generic-stm32f105x

The board config file is also available for testing. see platformio/platform-ststm32#499

fpistm, 2021-03-15:
Hi @ttimasdf 
I will add it with the new format.

williamalexmoran, 2021-07-01:
Hello, I want to add a new variant so I can use my STM H723ZG with Arduino. I was a little confused by the naming within the variants folder. Would my board (NUCLEO-H723ZG) be in the folder "H723Z(E-G)I_H730ZBI_H733ZGI" or in the folder "H723Z(E-G)T_H733ZGT"?

fpistm, 2021-07-01:
Hi @williamalexmoran 
as stated in the databrief: (https://www.st.com/en/evaluation-tools/nucleo-h723zg.html)
[image]

it is a `STM32H723ZGT6` so you should use the `H723Z(E-G)T_H733ZGT`

Pestimist, 2021-09-29:
Hello,

I would like to add support for Nucleo-L4P5ZG like stated over there in the forums: [https://www.stm32duino.com/viewtopic.php?f=12&t=1063](https://www.stm32duino.com/viewtopic.php?f=12&t=1063)

I somehow succeeded to modify the files so that a project compiles for the Nucleo-L4P5ZG and successfully downloads to the board.

But I am completely new to github and don't know how to use it properly...

- Do I have to clone this repository to my local disk?
- If so, what local folder should I use? the current stm32-arduino folder in my arduino installation?
- How do I get my modified files up here?

fpistm, 2021-09-29:
Hi @Pestimist 
First you should read the https://github.com/stm32duino/Arduino_Core_STM32/blob/master/CONTRIBUTING.md
Mainly Pull Requests section.

Pestimist, 2021-09-29:
ok, thank you!

bertrik, 2021-10-17:
Hi all,

I'm working on adding support for the Seeed studios LoRa E5 mini board.
It uses an STM32WLE5JCIx, with has an SX1262 LoRa transceiver integrated on an internal "subghz" SPI bus.
This board has serial pins on PB6/PB7, a LED on PB5, user button, and a radio Tx/Rx switch on pins PA4/PA5.

The plan is to add this under 'Generic STM32WL series', then as variant 'Seeed LoRa E5 mini', a bit like the 'Generic node SE (TTI)" from TheThingsIndustries.

Work in progress will appear at my fork:
https://github.com/bertrik/Arduino_Core_STM32

HammadK44, 2021-12-17:
Hi, 
Could someone please help me with STM32MP1.
I have updated PlatformIO to the latest version now but I am still not getting the option for STM32MP1 board in the Board Selector.
[image]

fpistm, 2021-12-17:
Hi @HammadK44, you should ask to pio.

alin2585, 2022-04-16:
> Hello, I want to add a new variant so I can use my STM H723ZG with Arduino. I was a little confused by the naming within the variants folder. Would my board (NUCLEO-H723ZG) be in the folder "H723Z(E-G)I_H730ZBI_H733ZGI" or in the folder "H723Z(E-G)T_H733ZGT"?

How is your progress, please?

alin2585, 2022-04-17:
hi.i'm working on the variant of Nucleo n723zgt6.i don't know how to edit the ldscript.ld. the original config is as follows.
could you please tell me how to edit? thanks.

[image]

[31 more replies]