stm32duino/Arduino_Core_STM32 #723: DFU mode via Software

MagoKimbra, 2019-10-23:
Is a question.
I have a Rumba32 on which I am programming the MK4duo firmware.
Unfortunately every time you download you have to put the processor in DFU mode by pressing two keys. There is a software way to add to the firmware that puts the processor in DFU mode to be updated by the arduino IDE without having to press the keys.
Many printers mount the boards in closed places of difficult access, it is complicated every time to update the firmware press those keys, so I wanted to do a gcode command that put the card in DFU mode to be updated by arduino ide ...
Thanks for the reply.

fpistm, 2019-10-23:
Hi @MagoKimbra 
See #706.

MagoKimbra, 2019-10-24:
Ok. I see. Thank's!

closed 2019-10-24 by MagoKimbra