stm32duino/Arduino_Core_STM32 #724: Board "unresponsive" on Serial after plugin

odo2063, 2019-10-23:
- Make sure you are using the latest Arduino_Core_STM32 version.
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
check
- Make sure the issue is not already reported/fixed on GitHub or discussed on the [stm32duino forum](http://stm32duino.com)
check
- Submit a GitHub issue only for reporting a problem related to the [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32).
think so
- **Avoid** to submit a GitHub issue for project troubleshooting.
no
 
When reporting any issue, please try to provide all relevant information:

**Describe the bug**

Board does not start or is unresponsive to serialcommands after connecting to PC by USB-Cable.
After pressing the reset button everything is fine.

tested on Nucleo L4R5ZI and Nucleo G071RB
Minimal code:
```
void setup() {
  Serial.begin(115200);
  delay(1000); 
  Serial.println("Ready");
} 

void loop() {

}
```
**To Reproduce**

Steps to reproduce the behavior:
1. unplug USB-cable
2. replug USB-cable
3. See nothing (no serial output)

**Expected behavior**
see serial output

**Additional**
Found the Problem while testing reboots with Raspi. The readout-programm didn't work. the curiouse thing is, I have more complex procedure on a second Nucleo and that Nucleo does not fail to start. :-/

fpistm, 2019-10-23:
Hi @odo2063 
you use the Serial over STLink right ?

odo2063, 2019-10-23:
I have not changed the jumper settings.
I use standard MicroUSBcable from nucleo board to PC.
For the test i leave arduino serialmonitor open.
Arduino version is 1.8.10

the real program is with serialcommands to which it is unresponsive.

fpistm, 2019-10-23:
I guess the main issue is that the Serial monitor was opened and so when you plug/unplug the descriptor is no more correct as the port disappear and appear again.
I don't know your host OS. but:
On Linux, I'm on /dev/ttyACM0 then if the Serial monitor is open, when the board reenumerate after the unplug/plug I get a /dev/ttyACM1.
On Windows, I have to close the serial monitor and reopen. Even in the Arduino port menu I get the COMxx name twice (evene more if I made the test several time).

fpistm, 2019-10-23:
Else try to update the STLink firmware using the STM32CubeProgrammer

odo2063, 2019-10-23:
okeeey....now the whole board seems to be bricked :-/
no filesystem to mount and just red LED flashing ST-Link....is there an easy way to unbrick?

fpistm, 2019-10-23:
After update of the STLink firmware ?

odo2063, 2019-10-23:
no...after i tried to upload a different sketch while in "unresponsive" state...

fpistm, 2019-10-23:
I guess the STLink is in a bad state. This could be issue wiht USB enumeration...
For Nucleo L4R5ZI
[image]

As I said, try to update the STLink firmware.

odo2063, 2019-10-23:
stm32cubeprogrammer(linux) tells me "Could not find or load main class"...

fpistm, 2019-10-23:
Argh... I will not able to help.
Maybe a Java runtime issue...

odo2063, 2019-10-23:
ok...evrything is back and running...firmwareupdate done...but the main Problem: Device not responsive after Systemreboot(raspi) still persists. So if you unplug and replug (this is essentially what raspi is doing on reboot) the device wants the resetbutton to be pressed...this will not be possible in this project(at least not by hand)...

fpistm, 2019-10-23:
Fine you get back work the board. 
But I don't understand that:
> Device not responsive after Systemreboot(raspi)

odo2063, 2019-10-23:
When raspi reboots it powers done the USB ports. So for the boards it's like unplug and replug. But if you do unplug and replug the board withthe simple code on it wants you to push the resetbutton. else you get the behavior from the first post. It is "unresponsive".

odo2063, 2019-10-23:
the dirty workaround would be to trigger the resetbutton by GPIO...but i dont want to do this atm...

fpistm, 2019-10-23:
Unfortunately, in this case I could not help, seems an issue with stlink not the core.

odo2063, 2019-10-23:
the strange thing is, The board with the complex code starts up fine...:-/

fpistm, 2019-10-23:
If you unplug an plug and it's works then there is an issue with the stlink not ready when app is ready when your raspi power on the USB. And I've no way to handle that in app side.
Your minimal sketch print only one time and is probably missed. What happen if you print in the loop with a delay

odo2063, 2019-10-23:
here is the code with the unresponsive behavior it does not respond to the serialcommands, the app on the raspi is asking over and over again "who" without response after reboot:

```
#include "Adafruit_VL53L0X.h"
#include <SerialCommands.h>

Adafruit_VL53L0X lox = Adafruit_VL53L0X();
VL53L0X_RangingMeasurementData_t measure;

bool meas = false;
unsigned long time_old = 0, time_new;
char serial_command_buffer_[32];

SerialCommands serial_commands_(&Serial, serial_command_buffer_, sizeof(serial_command_buffer_), "\n", " ");

//This is the default handler, and gets called when no other command matches. 
void cmd_unrecognized(SerialCommands* sender, const char* cmd)
{
  sender->GetSerial()->print("Unrecognized command [");
  sender->GetSerial()->print(cmd);
  sender->GetSerial()->println("]");
}

  //Serialcommand Functions
void cmd_meas_on(SerialCommands* sender)
{
  meas = true;  
}

void cmd_meas_off(SerialCommands* sender)
{
  meas = false;  
}

void cmd_who(SerialCommands* sender)
{
  Serial.println("vl53");  
}

SerialCommand cmd_meas_on_("start", cmd_meas_on);
SerialCommand cmd_meas_off_("stop", cmd_meas_off);
SerialCommand cmd_who_("who", cmd_who);

void setup() {
  delay(5000);
  Serial.begin(115200);
  delay(1000);
   
  
  if (!lox.begin()) {
    Serial.println("Failed to boot VL53L0X");
    while(1);
  }

  delay(1000);
  serial_commands_.SetDefaultHandler(cmd_unrecognized);
  serial_commands_.AddCommand(&cmd_meas_on_);
  serial_commands_.AddCommand(&cmd_meas_off_);
  serial_commands_.AddCommand(&cmd_who_); 

  Serial.println("Ready");
}

void loop() {
  serial_commands_.ReadSerial();

  if(meas)
  {
    lox.getSingleRangingMeasurement(&measure, false);
    time_new = millis();
    
    Serial.print(measure.RangeMilliMeter);
    Serial.print(", ");
    Serial.println(time_new-time_old);
    time_old = time_new;
    delay(6);
  }
  else {
    delay(100);
  }
}
```

...the board that start's "fine", does actually have issues with the first messages that being exchanged. the first chars are rubbish(sender and reciever side).

fpistm, 2019-10-23:
And it works when reset?

odo2063, 2019-10-23:
works when reset is pushed yes

fpistm, 2019-10-23:
I never tested this lib: serialcommands
I see while 1 in your code, try to print something to see if you see something

odo2063, 2019-10-23:
the while(1), is for when the sensor is not present, i get that absend message. before while(1)...
...and nothing, so no Timimng problem with that sensor:
```
  if (!lox.begin()) {
    Serial.println("Failed to boot VL53L0X");
    while(1)
    {
      Serial.println("oh no");
      delay(100);
    }
  }
```

fpistm, 2019-10-23:
Well power issue with USB during raspi boot and a wrong timing which cause unexpected behavior

odo2063, 2019-10-23:
I should get some Nano-sized Nucleos tommorrow so i will report back tommorrow...have a nice evening...

odo2063, 2019-10-24:
do you know a compileroption that could free me 328 bytes of ram?

fpistm, 2019-10-24:
Try disabling some HAL not required. See Wiki.
Next release will produce a smaller memory bin consumption (FLASH and RAM) sue to remove of whole-archive option (already in the master).

fpistm, 2019-10-24:
https://github.com/stm32duino/wiki/wiki/HAL-configuration#core-version--150-1

odo2063, 2019-10-24:
did this (stm32f0)...
```
#define HAL_MODULE_ENABLED
#define HAL_ADC_MODULE_DISABLED
#define HAL_CAN_MODULE_DISABLED
#define HAL_CEC_MODULE_DISABLED
#define HAL_COMP_MODULE_DISABLED
#define HAL_CORTEX_MODULE_ENABLED
#define HAL_CRC_MODULE_ENABLED
#define HAL_DAC_MODULE_DISABLED
#define HAL_DMA_MODULE_ENABLED
#define HAL_FLASH_MODULE_ENABLED
#define HAL_GPIO_MODULE_ENABLED
#define HAL_I2C_MODULE_ENABLED
#define HAL_I2S_MODULE_DISABLED
#define HAL_IRDA_MODULE_DISABLED
#define HAL_IWDG_MODULE_DISABLED
#define HAL_PCD_MODULE_DISABLED
#define HAL_PWR_MODULE_ENABLED
#define HAL_RCC_MODULE_ENABLED
#define HAL_RTC_MODULE_DISABLED
#define HAL_SMARTCARD_MODULE_DISABLED
#define HAL_SMBUS_MODULE_DISABLED
#define HAL_SPI_MODULE_DISABLED
#define HAL_TIM_MODULE_DISABLED
#define HAL_TSC_MODULE_DISABLED
#define HAL_UART_MODULE_ENABLED
#define HAL_USART_MODULE_DISABLED
#define HAL_WWDG_MODULE_DISABLED
```
no change..just need uart(on USB) and I²C...

fpistm, 2019-10-24:
Where you did this?

odo2063, 2019-10-24:
stm32f0xx_hal_conf_default.h

if i remove a character from one of the includes i get an error...

fpistm, 2019-10-24:
So, what I understand is you don't read the wiki and modify the core file....

fpistm, 2019-10-24:
> Extra HAL configuration can be enabled/disabled in variant.h (if required) or in a file named (at sketch level):
> 
> hal_conf_extra.h

anyway this should work also... but I advise to avoid to modify the core file

odo2063, 2019-10-24:
ok, same list in hal_conf_extra.h and `#include "hal_conf_extra.h"` in main file...reduced the overflow from 328 to 248 bytes...

fpistm, 2019-10-24:
Why you want reduce ram size? Provide the full sketche and lib dependencies you used.

odo2063, 2019-10-24:
I ordered some Nucleoboards and only looked for the flash size...my fault. Since there seems no quick way I would just use my DevBoard. if really want to have a look here's the code:
```
#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BNO055.h>
#include <utility/imumaths.h>
#include "filter1.h"
#include "filter2.h"
#include "filter3.h"
#include "hal_conf_extra.h"
#include <SerialCommands.h>
//#include "MathHelpers.h"
#include <EEPROM.h>

/* Set the delay between fresh samples */
#define BNO055_SAMPLERATE_DELAY_MS (1)

Adafruit_BNO055 bno = Adafruit_BNO055();

filter1Type *filter1 = filter1_create(); //TP vor 1. Integration
filter2Type *filter2 = filter2_create(); //HP vor 1. Integration
filter2Type *filter3 = filter2_create(); //HP nach 1. Integration
filter2Type *filter4 = filter2_create(); //HP nach 2. Integration
filter3Type *filter5 = filter3_create(); //TP gegen Filterschwingen

uint i = 1, j = 1, k = 0;

uint8_t sys, gyro, accel, mag = 0;

const uint  value1 = 2908; // Versatz 1/3 ////4408; //Versatz 1/4

            
float   wtf[8], 
            filtered1[value1], 
            envelope1[] = {0, 0, 0},  
            decay = 0.9998,
            mean1[] = {0, 0},
            mean3[] = {0, 0};
            //varianz[] = {0, 0},
            //stdabw[] = {0, 0};

float a = 0, v = 0, s = 0;

unsigned long time_new, time_old;

bool meas = false;

int eeAddress = 0;
adafruit_bno055_offsets_t calibData;

void displaySensorOffsets(const adafruit_bno055_offsets_t &calibData)
{
    Serial.println("done");
    /*
    delay(200);
    //Serial.print("Accelerometer: ");
    Serial.println(calibData.accel_offset_x); //Serial.print(" ");
    Serial.println(calibData.accel_offset_y); //Serial.print(" ");
    Serial.println(calibData.accel_offset_z); //Serial.print(" ");

    //Serial.println("\nGyro: ");
    Serial.println(calibData.gyro_offset_x); //Serial.print(" ");
    Serial.println(calibData.gyro_offset_y); //Serial.print(" ");
    Serial.println(calibData.gyro_offset_z); //Serial.print(" ");

    //Serial.print("\nMag: ");
    Serial.println(calibData.mag_offset_x); //Serial.print(" ");
    Serial.println(calibData.mag_offset_y); //Serial.print(" ");
    Serial.println(calibData.mag_offset_z); //Serial.print(" ");

[... 364 lines cut ...]
          delay(BNO055_SAMPLERATE_DELAY_MS);
          time_old = time_new;    
    
        }
  
        else
        {
          time_old = millis();
          delay(10);
          k += 1;
        }
    
    //}
  }
  else
  {
  k = 0;
  delay(100);
  }
}
```

the main topic of this thread i would close with "these are Developmentboards and no production boards"(no offense), since the nano-sized Nucleo also fails to start up. I will now implement that the working board will hardwarereset the not working board with GPIO action.

Thank a lot for your precious time. mille merci.

fpistm, 2019-10-24:
You can try to put all string in flash usin F("...")

odo2063, 2019-10-24:
all strings are hardcode F("") or removing them doesn't make any diffrence....

fpistm, 2019-10-25:
Which board you try to use?

fpistm, 2019-10-25:
Building your core and regarding the map file. 
You declare an array of 2908 float!
So this is normal on small RAM it does not fit.
So, please be aware of what you made...

Example on a Nucleo F030R8 which have 8Kb of RAM, the build result is less than 14Kb RAM .
Regarding the map file:
| Name	| Size dec	| Size hex| Section | Address |
| --- | --- | --- | --- | --- |
| filtered1	|11632 | 0x2d70 | .bss | 0x20000204 |
| .bss.eeprom_buffer | 1024 | 0x400 | .bss | 0x200030c1 |

So 12.6k are used by your sketch for 1 array and the eeprom emulation feature (as the Flash size page is 1kB).
Whatever you try you will not decrease those sizes. Core uses less than  1.4Kb of RAM for all other task.
So you need a minimum  of 16Kb for your use case.

odo2063, 2019-10-25:
I try to use a F072RB...for dev i had a G071RB...the Array is needed because of the Timeshift of the Highpassfilter of 2908 sample. And i need 16kb + 348b...it is totally my fault, i haven't looked up the RAM size of the new boards. i thought that if you have a lot of flash you will have enough RAM...the other thing is i also got the F303K, which i just realised when it arrived is a Cortex-M4 with 16kb RAM... i think i should create Table with all needed infos of all Boards...

odo2063, 2019-10-25:
**WORKAROUND for the actual Topic is to Hardware Reset the Boards with RaspiGPIO**

closed 2019-10-25 by odo2063