stm32duino/Arduino_Core_STM32 #726: BluePill , ST-Link V2 , DFU programming trouble.   - Missing ST-Linkv2 support.

AndKe, 2019-10-23:
So I love the work you've put down into this.
BluePill is compact, easily to integrate into other projects, and easily available.
Until now - I've used ST-Link v2 to program them.
I can burn the famous generic_boot20_pc13.bin bootloader, and try to program using DFU , but it fails most of the time, or the 1eaf:0003 USB device gets stuck(even after removing the USB cable)
Full description here: https://forum.arduino.cc/index.php?topic=642320.new#new

So I have two questions: 
-why did you remove ST-Link support around version 1.5
-how can I, reliably, program a BluePill now, on Ubuntu 19.04 /19.10  without it ? - preferably by using the bluepill's USB, or the DIO /DCLK pins (as with St-Link.)

If I should sum it up as a feature request; "please bring back ST-Link support."

stas2z, 2019-10-23:
for flashing with stlink choose STM32CubeProg (SWD)

AndKe, 2019-10-23:
Thank you, that worked perfectly.
I am horrified about the amount of bloat and "subscribe/accept" needed to just upload the firmware, I did really expect a tool of few megabytes at most. 
Hoping ST is not becoming like Microsoft, where everything grows beyond being reliable. :)

closed 2019-10-23 by AndKe

stas2z, 2019-10-23:
> Thank you, that worked perfectly.
> I am horrified about the amount of bloat and "subscribe/accept" needed to just upload the firmware, I did really expect a tool of few megabytes at most.
> Hoping ST is not becoming like Microsoft, where everything grows beyond being reliable. :)

Cubeprog is fat but its crossplatform and supports all stm chips