stm32duino/Arduino_Core_STM32 #727: Question on variant

fpistm, 2019-10-24:
Hi, could you expand a bit the **Add new board** wiki with info about how to set properly the analogs?, because I'm having problems with my variant for the [F030C8](https://github.com/r-pv/rSTM32F030C8T6_ProMiniPlus/tree/master/RBOARD_F030C8T6), when I tried to read PA0 ( analogRead(PA0) ) it really reads the PA3, for the moment I solve it by using analogRead(PA_0). So I was under the impression that only has to match the:
NUM_ANALOG_FIRST on the variant.h file with the duplicates on the digitalPin array on variant.cpp, but it seems that may be more to it, right?. Thanks.

_Originally posted by @r-pv in https://github.com/stm32duino/Arduino_Core_STM32/issues/722#issuecomment-545621796_

fpistm, 2019-10-24:
Hi @r-pv
you issue is that some analog pins have an number less than `NUM_ANALOG_INPUTS`
in your case it is equal to 10. So when you request `PA0` you get this value:
`#define PA0   3`
which in fact in the Ax equivalent to `A3` .
All Analog pin PYn must have a number greater or equal to `NUM_ANALOG_INPUTS`
This is what it means [here](https://github.com/r-pv/rSTM32F030C8T6_ProMiniPlus/blob/master/RBOARD_F030C8T6/variant.h#L126)
`// First analog pin value (A0) must be greater than or equal to NUM_ANALOG_INPUTS`

Maybe it could be rewrote like that:
`// First analog pin value (A0)  and all analog pin PYn must have a number greater or equal to  NUM_ANALOG_INPUTS`

r-pv, 2019-10-24:
Thanks for the reply, ups I got that wrong, I was thinking that the number was referring to the duplicates index and that was the reason they were written at last of the digitalPin array.

fpistm, 2019-10-24:
The reason to duplicate is to have all analog pins contiguous in order to be able to loop on it:

```
for (int i=0; i< NUM_ANALOG_INPUTS; i++) {
Serial.print(analogRead(i);
}
```

If in your definition they already are contiguous hen no need to duplicate.

r-pv, 2019-10-24:
awesome, mr @fpistm, I can fix my variants now. And sorry for keep taking your time.

fpistm, 2019-10-24:
Welcome

closed 2019-10-24 by fpistm