stm32duino/Arduino_Core_STM32 #729: GPIO PULLDOWN problem on PA11 and PA12

sutech, 2019-10-24:
Hi everybody
I am working on a custom board base on STM32f103c8 chip.
I use USART1, SPI1, I2C1 and GPIOs of course.
Everything is fine except the PULLDOWN feature of the two pins PA11 and PA12 which are ought to be used as Ext.Interrupts. it seems the internal PULLDOWN feature is not enabled by the following code:
` pinMode(ipbutton,INPUT_PULLDOWN);`

though the rest of the code works fine...
I reduced my code to a minimal form and try different gpios , and the result was it only happens for the two PA11 & PA12 pins so I thought it might be due to the USART module (the CTS and RTS pins are on these pins in the package) so I comment the codes related to the Serial module. still no success.
Here is the minimal Code, Does anyone has any ideas why this happens?

```
 #define LED1 PB15
#define LED2 PB14
#define LED3 PA8
#define ipbutton PA11 //

//#define Serial Serial1
int buttonvalue;

void setup() {
  // put your setup code here, to run once:
//Serial.begin(57600);
//  Serial.println("\n  Begin");
    pinMode(LED1, OUTPUT);
    pinMode(LED2, OUTPUT);
    pinMode(LED3, OUTPUT);
    pinMode(ipbutton,INPUT_PULLDOWN);                //Sets pin 2 as input 

}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(LED1, HIGH);
  delay(300);
  digitalWrite(LED1, LOW);
  delay(400);
  buttonvalue = digitalRead(ipbutton);   //Reads the status of 
  if(buttonvalue == HIGH)           
  {
 //     Serial.println("pressed");
       digitalWrite(LED2, HIGH);  

  }
  else
  {
 //    Serial.println("\n  Fine!");
            digitalWrite(LED2, LOW);  

    }
} 
```

sutech, 2019-10-24:
Thanks for your response Frederic.
No , never worked with usb on STM. Actually the code I posted is all I expect to work but does not.

fpistm, 2019-10-24:
Ok so it is not enabled in the arduino menu.
I will test tomorrow but as other worked I don't see why this one doesnt work.
Which variant you used?

sutech, 2019-10-24:
Its so kind of you.
Looks weird to me too, I tried to triple check everything though the code looks too simple to be buggy.
**STM32f103c8** 
[image]

fpistm, 2019-10-24:
OK I've found the issue 😉
You do not use this core. You use this one:
https://github.com/rogerclarkmelbourne/Arduino_STM32

closed 2019-10-24 by fpistm

Broman3100, 2019-10-25:
I guess you were right, I apologize for the inconvenience and thank you for taking your time to help me.

sutech, 2019-10-26:
Seriously?! 
But I never intend to use that repository.

Thank You

WrongVoltage, 2024-04-16:
Put Serial.end(); at the beginning of the Setup void.