stm32duino/Arduino_Core_STM32 #730: Arduino SD card library does not work (NUCLEO-L452RE-P)

Bambofy, 2019-10-24:
I have connected a micro sd card module to the SPI pins on this board at:

SCK - D13 - PB13
MISO - D12 - PB14
MOSI - D11 - PB15
CS - D10 - PA11

However the SD card library does not initialize the sd card. Using the Cardinfo sketch the result is:

> 17:27:24.472 -> Initializing SD card...initialization failed. Things to check:
> 17:27:26.517 -> * is a card inserted?
> 17:27:26.552 -> * is your wiring correct?
> 17:27:26.586 -> * did you change the chipSelect pin to match your shield or module?

The relavent code is as follows:
```
// include the SD library:
#include <SPI.h>
#include <SD.h>

// set up variables using the SD utility library functions:
Sd2Card card;
SdVolume volume;
SdFile root;

// change this to match your SD shield or module;
// Arduino Ethernet shield: pin 4
// Adafruit SD shields and modules: pin 10
// Sparkfun SD shield: pin 8
// MKRZero SD: SDCARD_SS_PIN
const int chipSelect = PA11;
void setup() {  
  pinMode(chipSelect, OUTPUT);
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  Serial.print("\nInitializing SD card...");

  // we'll use the initialization code from the utility libraries
  // since we're just testing if the card is working!
  if (!card.init(SPI_HALF_SPEED, chipSelect)) {
    Serial.println("initialization failed. Things to check:");
    Serial.println("* is a card inserted?");
    Serial.println("* is your wiring correct?");
    Serial.println("* did you change the chipSelect pin to match your shield or module?");

    while (1) {
              digitalWrite(PB13, HIGH);   // turn the LED on (HIGH is the voltage level)
      delay(1000);  
    }
  } else {
    Serial.println("Wiring is correct and a card is present.");

    while (1) {
        digitalWrite(PB13, HIGH);   // turn the LED on (HIGH is the voltage level)
      delay(1000);                       // wait for a second
      digitalWrite(PB13, LOW);    // turn the LED off by making the voltage LOW
      delay(1000);  
    }
  }

  // print the type of card
  Serial.println();
  Serial.print("Card type:         ");
  switch (card.type()) {
    case SD_CARD_TYPE_SD1:
      Serial.println("SD1");
      break;
    case SD_CARD_TYPE_SD2:
      Serial.println("SD2");
      break;
    case SD_CARD_TYPE_SDHC:
      Serial.println("SDHC");
      break;
    default:
      Serial.println("Unknown");
  }

  // Now we will try to open the 'volume'/'partition' - it should be FAT16 or FAT32
  if (!volume.init(card)) {
    Serial.println("Could not find FAT16/FAT32 partition.\nMake sure you've formatted the card");
    while (1);
  }

  Serial.print("Clusters:          ");
  Serial.println(volume.clusterCount());
  Serial.print("Blocks x Cluster:  ");
  Serial.println(volume.blocksPerCluster());

  Serial.print("Total Blocks:      ");
  Serial.println(volume.blocksPerCluster() * volume.clusterCount());
  Serial.println();

  // print the type and size of the first FAT-type volume
  uint32_t volumesize;
  Serial.print("Volume type is:    FAT");
  Serial.println(volume.fatType(), DEC);

  volumesize = volume.blocksPerCluster();    // clusters are collections of blocks
  volumesize *= volume.clusterCount();       // we'll have a lot of clusters
  volumesize /= 2;                           // SD card blocks are always 512 bytes (2 blocks are 1KB)
  Serial.print("Volume size (Kb):  ");
  Serial.println(volumesize);
  Serial.print("Volume size (Mb):  ");
  volumesize /= 1024;
  Serial.println(volumesize);
  Serial.print("Volume size (Gb):  ");
  Serial.println((float)volumesize / 1024.0);

  Serial.println("\nFiles found on the card (name, date and size in bytes): ");
  root.openRoot(volume);

  // list all files in the card with date and size
  root.ls(LS_R | LS_DATE | LS_SIZE);
}
```

fpistm, 2019-10-24:
Hi @Bambofy 
I will check that tomorrow.

Bambofy, 2019-10-24:
> Hi @Bambofy
> I will check that tomorrow.

Thanks 👍

fpistm, 2019-10-25:
Hi @Bambofy 
I've tested and have no error.
So probably wiring issue, SD card (file system, defective, type not supported?)  or something like that.

fpistm, 2019-10-25:
The same than you Nucle-L452RE-P

fpistm, 2019-10-25:
Maybe your shield has a pull up/down resistor on one of the line?

Bambofy, 2019-10-25:
[image]
[image]

[image]

Bambofy, 2019-10-25:
Powweing it through usb port on computer

Bambofy, 2019-10-25:
> 
> 
> ~On your last photo it seems both cable are on GND ??? (Blue and yellow)~
> No this is correct.
> Well I don't know how to help. Did you check all your cable are correct?
> I've tested with 2 SD cards shield without any issue.

ok thank you very much for helping, it must be the sd card i have!

Bambofy, 2019-10-25:
Do i have to change any of the jumpers from the default settings?

fpistm, 2019-10-25:
Sorry to could not help more.

fpistm, 2019-10-25:
> Do i have to change any of the jumpers from the default settings?

No. Did you made hardware modifications? Or in the core?

Bambofy, 2019-10-25:
none

> 
> 
> > Do i have to change any of the jumpers from the default settings?
> 
> No. Did you made hardware modifications? Or in the core?

none, what sd card do you use?

fpistm, 2019-10-25:
If same setup works with a mega then it should work also. Did you enable LTO in the optimisation menu?

BennehBoy, 2019-10-25:
Are you sure this isn't a logic voltage level issue?  There's a voltage
regulator on that SD board, could it be stepping 3.3v down to ~ 2.2v?  I
know this issue has bitten a lot of people in STM32 space previously, I
usually buy non regulated SD boards.

fpistm, 2019-10-25:
Maybe. Don't know.
You can check 5v provided by the board and also check each SPI lines.

fpistm, 2019-10-25:
I will test another SPI device, power the shield with another source (with GND interconnect), check all cables used, try another SPI pins for the boards,...
There is a lot of test which can be performed but you are on your own for this.

fpistm, 2019-10-25:
FYI I used an SDHC of 8GB.

Bambofy, 2019-10-25:
Hi, will test this out today

Bambofy, 2019-10-25:
[image]

Bambofy, 2019-10-25:
Got it working!

fpistm, 2019-10-25:
Nice.
Anyway, seems you get an hardware issue on first pin.
did you try to simply toggle the pin to check if they are functional?

Bambofy, 2019-10-25:
> Nice.
> Anyway, seems you get an hardware issue on first pin.
> did you try to simply toggle the pin to check if they are functional?

have not tried testing the pins but i'll check them with a multimeter tomorrow

Bambofy, 2019-10-28:
Hi, turns out that this version of the STM32 chip NUCLEO-L452RE-P is not supported by stm32duino yet.

fpistm, 2019-10-29:
> Hi, turns out that this version of the STM32 chip NUCLEO-L452RE-P is not supported by stm32duino yet.

In the repo it is supported, will be release with 1.8.0.

fpistm, 2019-10-31:
Any update on this ?
I think it could be linked to a voltage issue.
As this board use SMPS per default maybe with this shield you have an issue.
Did you try to power the SD shield with an external source ?

fpistm, 2019-11-05:
@Bambofy 
As mentioned before, I could not reproduce this and probably linked to an hardware issue.
So I close it as I will not be able to help more, I provided some track, hope this help.

closed 2019-11-05 by fpistm