stm32duino/Arduino_Core_STM32 #731: Variant.h for RUMBA32

MagoKimbra, 2019-10-24:
Please enter the TIMER_SERIAL define in variant.h of rumba32.
If the SoftwareSerial.h library is included, as is the case with the TMCStepper library, SoftwareSerial takes the timer 7 for the STM32F446 processor which is the same one used by the servo and in fact no longer works.
I put #define TIMER_SERIAL 8 in variant.h so the servo works, but I don't know if the SoftwareSerial library works, it should be verified ...

fpistm, 2019-10-24:
Hi @MagoKimbra 
in this case you could provide a PR.
This is well explained here:
https://github.com/stm32duino/Arduino_Core_STM32/blob/2f1fd8e19d35a9805f81e5054a726d3b0b6b748a/libraries/SoftwareSerial/src/SoftwareSerial.cpp#L42-L51

So I guess there is no TIM18 for F446 and then TIM7 is used.

MagoKimbra, 2019-10-24:
Yes, TIM18 not define for STM32F466 and use the first TIM7.
I tried to put the define in my HAL firmware, but it is not seen by SoftwareSerial.cpp.

fpistm, 2019-10-24:
Don't know your HAL firmware but if set in variant.h this should be ok

MagoKimbra, 2019-10-24:
Yes yes in variant.h the servo works perfectly. I tried to put the define in my firmware which includes the TMCstepper.h library which in turn includes the Softwareserial.h library, but unfortunately the define is not seen, I said this.

fpistm, 2019-10-24:
ok.
Other way is to use build_opt.h or maybe the hat_conf_extra.h.

fpistm, 2019-10-28:
@MagoKimbra 
which timer you want for TIMER_Serial per default ?

MagoKimbra, 2019-10-28:
TIM9. Thank's.
I use TIM2 for stepper, SysTick for the rest. There are still TIM5 32bit, TIM3, TIM4, TIM10-TIM14 at 16bit for other things.

fpistm, 2019-10-28:
Well,  I will not modify the variant as this will fix it for all users and they do not want necessarily to use this one.
In your case you have to add `hal_conf_extra.h` with 
`#define TIMER_SERIAL TIM9`

closed 2019-10-28 by fpistm

MagoKimbra, 2019-10-28:
Where i add hal_conf_extra.h??? in Myfirmware or in library STM32???

fpistm, 2019-10-28:
When using Arduino IDE, simply adding it at sketch level works.
While the fle in the include path this will work.