stm32duino/Arduino_Core_STM32 #732: PA12 pin not toggling on stm32f103c series

Broman3100, 2019-10-25:
The digitalWrite(PA12, LOW/0) function for PA12 pin doesn't work on generic stm32f103 board, (f103c8), I know it's not just my board problem since I used keil and it worked just fine so I figured the problem is in stmduino.

fpistm, 2019-10-25:
Hi @Broman3100 
Could you be more precise.
Which core version you used?
Build option set in the Arduino menu, ...
All relevant information to help

Broman3100, 2019-10-25:
I used arduino 1.8.9( windows store 1.8.21.0), I'm confident I used newest version of stm32duino and build options are following:
[image]

fpistm, 2019-10-25:
Ok that's what I think.
Same issue than here: #729
You do not use this core. You use this one:
https://github.com/rogerclarkmelbourne/Arduino_STM32

closed 2019-10-25 by fpistm

Broman3100, 2019-10-25:
I guess you were right, I apologize for the inconvenience and thank you for taking your time to help me.

fpistm, 2019-10-25:
Welcome