stm32duino/Arduino_Core_STM32 #734: Half Duplex serial example won't compile.(invalid conversion from 'int' to 'void*')

AndKe, 2019-10-25:
I tried to compile the half duplex example found here : https://github.com/stm32duino/wiki/wiki/API#hardwareserial (since I failed to apply it to real application)

I replaced all Serial4 with Serial2 for BluePill

I get this error:
```

/home/andre/Arduino/arduino-1.8.10/arduino-builder -dump-prefs -logger=machine -hardware /home/andre/Arduino/arduino-1.8.10/hardware -hardware /home/andre/.arduino15/packages -hardware /home/andre/Arduino/projects/hardware -tools /home/andre/Arduino/arduino-1.8.10/tools-builder -tools /home/andre/Arduino/arduino-1.8.10/hardware/tools/avr -tools /home/andre/.arduino15/packages -built-in-libraries /home/andre/Arduino/arduino-1.8.10/libraries -libraries /home/andre/Arduino/projects/libraries -fqbn=STM32:stm32:GenF1:pnum=BLUEPILL_F103C8,upload_method=swdMethod,xserial=generic,usb=CDCgen,xusb=FS,opt=oslto,rtlib=nano -vid-pid=0483_5740 -ide-version=10810 -build-path /tmp/arduino -warnings=default -build-cache /tmp/arduino_cache_583206 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=/home/andre/.arduino15/packages/STM32/tools/CMSIS/5.5.1 -prefs=runtime.tools.CMSIS-5.5.1.path=/home/andre/.arduino15/packages/STM32/tools/CMSIS/5.5.1 -prefs=runtime.tools.STM32Tools.path=/home/andre/.arduino15/packages/STM32/tools/STM32Tools/1.3.1 -prefs=runtime.tools.STM32Tools-1.3.1.path=/home/andre/.arduino15/packages/STM32/tools/STM32Tools/1.3.1 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/andre/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7 -prefs=runtime.tools.arm-none-eabi-gcc-8.2.1-1.7.path=/home/andre/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7 -verbose /home/andre/Arduino/projects/temp_sketch_oct25a/temp_sketch_oct25a.ino
/home/andre/Arduino/arduino-1.8.10/arduino-builder -compile -logger=machine -hardware /home/andre/Arduino/arduino-1.8.10/hardware -hardware /home/andre/.arduino15/packages -hardware /home/andre/Arduino/projects/hardware -tools /home/andre/Arduino/arduino-1.8.10/tools-builder -tools /home/andre/Arduino/arduino-1.8.10/hardware/tools/avr -tools /home/andre/.arduino15/packages -built-in-libraries /home/andre/Arduino/arduino-1.8.10/libraries -libraries /home/andre/Arduino/projects/libraries -fqbn=STM32:stm32:GenF1:pnum=BLUEPILL_F103C8,upload_method=swdMethod,xserial=generic,usb=CDCgen,xusb=FS,opt=oslto,rtlib=nano -vid-pid=0483_5740 -ide-version=10810 -build-path /tmp/arduino -warnings=default -build-cache /tmp/arduino_cache_583206 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=/home/andre/.arduino15/packages/STM32/tools/CMSIS/5.5.1 -prefs=runtime.tools.CMSIS-5.5.1.path=/home/andre/.arduino15/packages/STM32/tools/CMSIS/5.5.1 -prefs=runtime.tools.STM32Tools.path=/home/andre/.arduino15/packages/STM32/tools/STM32Tools/1.3.1 -prefs=runtime.tools.STM32Tools-1.3.1.path=/home/andre/.arduino15/packages/STM32/tools/STM32Tools/1.3.1 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/andre/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7 -prefs=runtime.tools.arm-none-eabi-gcc-8.2.1-1.7.path=/home/andre/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7 -verbose /home/andre/Arduino/projects/temp_sketch_oct25a/temp_sketch_oct25a.ino
Using board 'GenF1' from platform in folder: /home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0
Using core 'arduino' from platform in folder: /home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0
bash -c "[ -f /home/andre/Arduino/projects/temp_sketch_oct25a/build_opt.h ] || (mkdir -p /tmp/arduino/sketch && touch /tmp/arduino/sketch/build_opt.h)"
Detecting libraries used...
/home/andre/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/arm-none-eabi-g++ -mcpu=cortex-m3 -mthumb @/tmp/arduino/sketch/build_opt.h -c -Os -flto -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -I/home/andre/Arduino/projects/temp_sketch_oct25a -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/cores/arduino/avr -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/cores/arduino/stm32 -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/cores/arduino/stm32/LL -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/cores/arduino/stm32/usb -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/cores/arduino/stm32/usb/hid -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/cores/arduino/stm32/usb/cdc -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/system/Drivers/STM32F1xx_HAL_Driver/Inc/ -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/system/Drivers/STM32F1xx_HAL_Driver/Src/ -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/system/STM32F1xx/ -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -w -x c++ -E -CC -DSTM32F1xx -DARDUINO=10810 -DARDUINO_BLUEPILL_F103C8 -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"BLUEPILL_F103C8\"" -DSTM32F103xB -DUSBCON -DUSBD_VID=0x0483 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"BLUEPILL_F103C8\"" -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DHAL_UART_MODULE_ENABLED -I/home/andre/.arduino15/packages/STM32/tools/CMSIS/5.5.1/CMSIS/Core/Include/ -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/ -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/ -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/cores/arduino -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/variants/PILL_F103XX /tmp/arduino/sketch/temp_sketch_oct25a.ino.cpp -o /dev/null
Generating function prototypes...
/home/andre/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/arm-none-eabi-g++ -mcpu=cortex-m3 -mthumb @/tmp/arduino/sketch/build_opt.h -c -Os -flto -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -I/home/andre/Arduino/projects/temp_sketch_oct25a -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/cores/arduino/avr -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/cores/arduino/stm32 -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/cores/arduino/stm32/LL -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/cores/arduino/stm32/usb -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/cores/arduino/stm32/usb/hid -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/cores/arduino/stm32/usb/cdc -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/system/Drivers/STM32F1xx_HAL_Driver/Inc/ -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/system/Drivers/STM32F1xx_HAL_Driver/Src/ -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/system/STM32F1xx/ -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -w -x c++ -E -CC -DSTM32F1xx -DARDUINO=10810 -DARDUINO_BLUEPILL_F103C8 -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"BLUEPILL_F103C8\"" -DSTM32F103xB -DUSBCON -DUSBD_VID=0x0483 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"BLUEPILL_F103C8\"" -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DHAL_UART_MODULE_ENABLED -I/home/andre/.arduino15/packages/STM32/tools/CMSIS/5.5.1/CMSIS/Core/Include/ -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/ -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/ -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/cores/arduino -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/variants/PILL_F103XX /tmp/arduino/sketch/temp_sketch_oct25a.ino.cpp -o /tmp/arduino/preproc/ctags_target_for_gcc_minus_e.cpp
/home/andre/Arduino/arduino-1.8.10/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
[... cut ...]
/home/andre/Arduino/projects/temp_sketch_oct25a/temp_sketch_oct25a.ino: In function 'void test_uart(int)':
temp_sketch_oct25a:35:11: error: 'class HardwareSerial' has no member named 'enableHalfDuplexRx'
   Serial2.enableHalfDuplexRx();
           ^~~~~~~~~~~~~~~~~~
temp_sketch_oct25a:44:13: error: 'class HardwareSerial' has no member named 'enableHalfDuplexRx'
     Serial3.enableHalfDuplexRx();
             ^~~~~~~~~~~~~~~~~~
exit status 1
invalid conversion from 'int' to 'void*' [-fpermissive]

```

fpistm, 2019-10-26:
As it is mentioned in the wiki:
> Available in core version greater than 1.7.0

currently this is available in the master but not yet released thanks the board manager. This will be available in the next release 1.8.0

closed 2019-10-26 by fpistm

AndKe, 2019-10-29:
how long till next release?
If I upgrade to master manually, will it break further Arduino boards-manager updates?

fpistm, 2019-10-29:
Don't know exactly. But probably not before december.
And no this should not break update.