stm32duino/Arduino_Core_STM32 #736: STM32G071/081 Analog Pin Mapping Errors

steenerson, 2019-10-28:
**Describe the bug**
Several ADC pins report incorrect values on STM32G0x1 series boards, I have tried separate hardware with G071RBT6 and G081RBT6 and both exhibit similar issues. On my Nucleo-G071RB board, 3 are not working correctly:

PB11/A4 - always returns 0
PB12/A5 - sketches performing analogRead of PB12 seem to be reading physical pin PA0 instead of PB12. Continuity tests with multimeter match the schematic - PB12 is physically connected to socket A5 and they are not shorted to PA0. With the board completely unplugged from power/peripherals I measure 10-15Mohm between PA0 and PB12, leading me to believe it is a software issue and not hardware.
PB0/D10 - similarly, sketches reading PB0 seem to be reading physical pin PB10.

On a board I made with a STM32G081RBT6 and a custom variant, I'm attempting to use all 16 analog pins and at least 4 are reading incorrectly - PB11, PB12, PC4 and PC5 readings don’t match the measured voltage on the pin. The variant I made has all of the analog pins in variant.h and variant.cpp in the same order as in the PinMap_ADC section of PeripheralPins.c, which seems to make PB0 work correctly compared to the Nucleo board, but PB11/PB12/PC4/PC5 haven’t worked in any order I’ve tried.

**To Reproduce**
void setup() {
  analogReadResolution(12);
  Serial.begin(115200);
  pinMode(PA0, INPUT);
  pinMode(PA1, INPUT);
  pinMode(PA4, INPUT);
  pinMode(PA6, INPUT);
  pinMode(PA7, INPUT);
  pinMode(PB0, INPUT);
  pinMode(PB1, INPUT);
  pinMode(PB2, INPUT);
  pinMode(PB10, INPUT);
  pinMode(PB11, INPUT);
  pinMode(PB12, INPUT);
}

void loop() {
  Serial.print("PA0/A0 volts = ");
  Serial.println(analogRead(PA0) / 1241.2);
  Serial.print("PA1/A1 volts = ");
  Serial.println(analogRead(PA1) / 1241.2);
  Serial.print("PA4/A2 volts = ");
  Serial.println(analogRead(PA4) / 1241.2);
  Serial.print("PA6/D12 volts = ");
  Serial.println(analogRead(PA6) / 1241.2);
  Serial.print("PA7/D11 volts = ");
  Serial.println(analogRead(PA7) / 1241.2);
  Serial.print("PB0/D10 volts = ");
  Serial.println(analogRead(PB0) / 1241.2);
  Serial.print("PB1/A3 volts = ");
  Serial.println(analogRead(PB1) / 1241.2);
  Serial.print("PB2/CN10_22 volts = ");
  Serial.println(analogRead(PB2) / 1241.2);
  Serial.print("PB10/CN10_28 volts = ");
  Serial.println(analogRead(PB10) / 1241.2);
  Serial.print("PB11/A4 volts = ");
  Serial.println(analogRead(PB11) / 1241.2);
  Serial.print("PB12/A5 volts = ");
  Serial.println(analogRead(PB12) / 1241.2);
  Serial.println();
  delay(1000);
}

Steps to reproduce the behavior:
1. Run above code
2. Open 115200 baud serial monitor
2. Use short jumper wire to induce 3.3V on analog pins

**Expected behavior**
Plugging 3.3V into PB0, PB11 and PB12 will cause them to read 3.3V (4095), but instead they read .5-1v with high variability, as if floating. 

Plugging 3.3V into PA0 and PB10 will only cause those pins to read 3.3V, but instead each causes 2 pins to read 3.3V.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Windows
 - Arduino IDE version: 1.8.10
 - STM32 core version: 1.7.0
 - Tools menu settings if not the default: 
 - Upload method: STLink (SWD)

**Board (please complete the following information):**
 - Name: Nucleo G071RB
 - Hardware Revision: MB1360-G071RB-C01
 - Extra hardware used if any: jumper wire

**Additional context**

fpistm, 2019-10-28:
Hi @steenerson 
Thanks for this report.
PB0 should not be available as its number is less than analog pin number.
For other (PB11/12), I have the same result than you and trying with STM32CubeIDE gave me the same result.
So, at this time it seems not linked to core implementation.

fpistm, 2019-10-29:
I've made a PR to fix issue with `PB0`.
For `PB11` and `PB12`, I"ve raised an issue internally. I will give a feedback when I will get some news.

steenerson, 2019-10-29:
Great news and progress, thanks!

fpistm, 2019-10-31:
Hi @steenerson 

The STM32G0 ADC has a new feature on sequencer: can be set to "not fully configurable "legacy mode (for compatibility with STM32F0 and L0) or "fully configurable" (new mode).

The new sequencer fully configurable is more flexible but it has the drawback to manage channels only in range [0-14].

That's why  PB11/12 (ADC_IN15/16) and also PC4/5 was not functional.
I've made  the fix and now all Ax pins are working.

steenerson, 2019-11-01:
Ah that makes sense, thanks! It's easy to setup in CubeMX/IDE but it's proving a bit over my head in Arduino, I found a function in the HAL driver that looks like it might be made for this, but adding it to setup isn't working - it compiles without errors but doesn't change pin behavior or the output of GetSequencerConfigurable:

LL_ADC_REG_SetSequencerConfigurable(ADC1, LL_ADC_REG_SEQ_FIXED);

I'll keep working at it, but do you have any hints to get it working?

fpistm, 2019-11-01:
In CubeMX , there is an issue with `LL_ADC_REG_SetSequencerConfigurable`, it have to be call before the `LL_ADC_Init`.
Anyway, for Arduino this is now fixed with #737.
Did you test it ?

closed 2019-11-01 by fpistm

steenerson, 2019-11-01:
Sorry about that, I missed the commit you added to #737 yesterday and was trying to set 'not fully configurable' on my own.

I can confirm that all analog pins on the updated nucleo-g071 variant are now reading as expected, and on my custom board all 16 analog pins and the internal channels are working as expected too. Thank you very much for the fix!

fpistm, 2019-11-01:
Welcome
No worry

commit 1fb52633 (2019-11-01): [Nucleo-G071RB] Fix analog pin issue

Fixes #736
commit 856da01c (2019-11-01): [Nucleo-G071RB] Fix PB11/12 (ADC_IN15/16) analog read

Fixes #736