stm32duino/Arduino_Core_STM32 #738: (F103RET6) undefined reference to HAL_CAN_Init

StarlightNL, 2019-10-29:
**Describe the bug**
I am trying to use CAN with a STM32F103RET6 but I cannot compile my program. It always says that it cannot find the functions like HAL_CAN_Init or HAL_CAN_ConfigFilter.
I did add `#include <stm32f1xx_hal_can.h>` to the .h file and even to the .cpp file in the hope that it works

**To Reproduce**

include `#include <stm32f1xx_hal_can.h>` and call `HAL_CAN_Init` or `HAL_CAN_ConfigFilter`

**Expected behavior**
I expect it to compile.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Arduino IDE version: 1.8.5 (Using Visual Micro)
 - STM32 core version: 1.7.0 from boards menu
 - Upload method: BMP

**Board (please complete the following information):**
 - Name: STM32F103RET6 (blue button) dev board

fpistm, 2019-10-29:
Hi @StarlightNL 
This is not a bug.
By default HAL CAN module is not enabled.
You have to enable it defining `HAL_CAN_MODULE_ENABLED`.
Add `hal_conf_extra.h` to your sketch and add this:
`#define HAL_CAN_MODULE_ENABLED`
See:
https://github.com/stm32duino/wiki/wiki/HAL-configuration#core-version--150-1

StarlightNL, 2019-10-30:
After doing the steps provided it works.
However I do have one other question. 

Can it also work in a library or do I need to do some other steps for that?

fpistm, 2019-10-30:
> Can it also work in a library or do I need to do some other steps for that?

If your question is to add` hal_conf_extra.h` in a library then I don't think it will work.
The core search in the include path and IIWR the way Arduino manages library then it will not be seen by the core. You can try but I will not support that if this not work.

closed 2019-10-30 by fpistm