stm32duino/Arduino_Core_STM32 #741: entering DFU mode on nucleo_l476

andhieSetyabudi, 2019-10-30:
i use PlatformIO with stm32duino as the packages ( same as arduino for stm32)..
chip on my own board is stm32l476 same as nucleo_l476.
i wanna upload my program onto the chip through usb, so the choice is DFU.
is there library for me to set the bootloader?? or how to set MCU to entering the DFU ?

thanks

fpistm, 2019-10-30:
Hi @andhieSetyabudi 
Currently there is no automatic way to enter in DFU mode.
stm32l476  has the STM32 builtin bootloader which support DFU.
Refers to the [AN2606 STM32 microcontroller system memory boot mode](https://www.st.com/content/ccc/resource/technical/document/application_note/b9/9b/16/3a/12/1e/40/0c/CD00167594.pdf/files/CD00167594.pdf/jcr:content/translations/en.CD00167594.pdf)

For your case set BOOT0 to 1 will allow enter in DFU mode. USB cable have to be plug before powering the board if this is not USB which power it.

andhieSetyabudi, 2019-10-30:
hi @fpistm 
thanks for your reply..
i've already test to entering DFU by set BOOT0 to 1 and it success..
but that makes not simply by press BOOT0 to set it 1 every flashing it..
maybe there is other way or trick to set my MCU to entering DFU mode automaticaly ??

thanks

fpistm, 2019-10-30:
See #706

closed 2019-10-30 by fpistm