stm32duino/Arduino_Core_STM32 #743: Clock does not start on STM32F373-Board

michsens, 2019-10-30:
Hello,

I developed a pcb with a STM32F373 in a LQFP-64 package. Programming is no big deal as long as I use TrueStudio. Clock configuration works well. Same for adc, dac, gpio input and output. Other modules I did not test. My aim was to see if the board itself works. It does.

The next step was to bring this board into the arduino environment. I defined the new board in the board file and created a folder with the variant,  the ldscript, the PeripheralPins and PinNamesVAr files. I made I test with the well known Blink example. The compilation itself is successful, same with the upload. 

But the controller does not start working. The 8MHz crystal does not start to operate what leads to a completely not working device.

I compared the libraries used by TrueStudio and arduino. I could not find anything suspicious. I also tried different versions of the stm32 cores (1.6.0, 1.6.1, 1.7.0) - no success.

What else can I observe to get the board working.

Regards

Robert.

fpistm, 2019-10-30:
Hi @michsens 
this could comes from several reasons.
Could you share your variant then I could have a look and helps also for your other issue #742

michsens, 2019-10-30:
Thanks for the reply. Here is the variant.h:

```
/*
  Copyright (c) 2011 Arduino.  All right reserved.

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
  See the GNU Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
*/

#ifndef _VARIANT_ARDUINO_STM32_
#define _VARIANT_ARDUINO_STM32_

#ifdef __cplusplus
extern "C" {
#endif // __cplusplus

/*----------------------------------------------------------------------------
 *        Pins
 *----------------------------------------------------------------------------*/

#define PC0			0			// I_mirr_1mA

#define PC8			1			// Temp_AN_1
#define PC9			2			// Temp_AN_2
#define PA8			3			// Temp_AN_3
#define PA11		4			// Temp_AN_4
#define PA12		5			// Temp_intern_AN
#define PF7			6			// Temp_kal_3k_AN
#define PF6			7			// Temp_kal_30k_AN

#define PC4			8			// AI24_AN_2
#define PC5			9			// AI24_AN_1
#define PC6			10			// AI24_AN_4
#define PC7			11			// AI24_AN_3
#define PA7			12			// 0_20mA_1
#define PC1			13			// 0_20mA_2

#define PC2			14			// EN_DA_PWM_1
#define PC3			15			// EN_DA_PWM_2

#define PB1			16			// AI_1
#define PB2			17			// AI_2
#define PE8			18			// AI_3
#define PE9			19			// AI_4					20
#define PD8			20			// AI_temperatur
#define PB14		21			// AI_kal_3k
#define PB15		22			// AI_kal_30k
#define PB0			23			// AI_curr

#define PA15		24			// DI_1
#define PD2			25			// DI_2
#define PC12		26			// DI_3
#define PC11		27			// DI_4, USART3_RX
#define PB9			28			// DI_5, USART3_RX

#define PC10		29			// USART3_TX
#define PA4			30			// DA_PWM_1
#define PA5			31			// DA_PWM_2
[... 286 lines cut ...]
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_HSE;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_0) != HAL_OK)
  {
    Error_Handler();
  }
  PeriphClkInit.PeriphClockSelection = RCC_PERIPHCLK_ADC1;
  PeriphClkInit.Adc1ClockSelection = RCC_ADC1PCLK2_DIV2;

  if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInit) != HAL_OK)
  {
    Error_Handler();
  }
}

#ifdef __cplusplus
}
#endif

```

What else can be of any help?

fpistm, 2019-10-30:
Could you also precise which core version you used ? (or maybe the git repository?)
Host OS?
Thanks in advance

michsens, 2019-10-30:
Core Version 1.6.1

I also tried 1.7.0. But there was no difference.

michsens, 2019-10-30:
[Example_STM32F373R8.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3789349/Example_STM32F373R8.zip)
[boards_Example.txt](https://github.com/stm32duino/Arduino_Core_STM32/files/3789354/boards_Example.txt)

fpistm, 2019-10-30:
Thanks.
This is more easy to review.
Well they are several issue. The main one is the _estack definition in the linker script which I guess 
avoid the your board to boot.
Here all the fixes:
https://github.com/fpistm/Arduino_Core_STM32/tree/Example_STM32F373R8

michsens, 2019-10-31:
Thanks a lot for your help.

I have some question about the change in the PeripheralPins.c:
Why do you call the SDADC by HAL_SD_MODULE_ENABLED? In the stm32 libraries it is defined as HAL_SDADC_MODULE_ENABLED [https://github.com/fpistm/Arduino_Core_STM32/blob/Example_STM32F373R8/system/STM32F3xx/stm32f3xx_hal_conf_default.h](url) ?

It also confuses me, why you activate a PULLUP for each channel. I would change the lines as follows:

from: `{PB_0,  SDADC1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF_NONE)}, // SDADC1_AIN6P`

to: `{PB_0,  SDADC1,STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, SDADC1_AIN6P, 0)},`

Would you agree? Do I missunderstand somethings?

michsens, 2019-10-31:
Thanks a lot for your help.

I have some question about the change in the PeripheralPins.c:
Why do you call the SDADC by HAL_SD_MODULE_ENABLED? In the stm32 libraries it is defined as HAL_SDADC_MODULE_ENABLED ()

fpistm, 2019-10-31:
This is a mistake.
The core doesn't manage SDADC.
I have to update the script which generate the PeripheralPins.c to handle this properly.

fpistm, 2019-10-31:
@michsens 
is it ok now? Your board starts properly?

michsens, 2019-10-31:
I am home today. I check it tomorrow when I am in the lab. I give you some feedback then. Thanks until here.

michsens, 2019-11-01:
Good Morning,

the change in the ldscript was the key. The oscillator starts properly and the LED blinks now. Many thanks fpistm.

fpistm, 2019-11-01:
Welcome

closed 2019-11-01 by fpistm