stm32duino/Arduino_Core_STM32 #747: Sampling at 16KHz using ADC prescaler?

Bambofy, 2019-10-31:
Hi, I need to sample the ADC at 16KHz to create an audio file. How can I do this using stm32duino?

Many thanks,
Richard

fpistm, 2019-10-31:
Hi @Bambofy 
on which board? Nucleo-L452RE-P?

ADC is limited on this core to `analogRead()` API which made a single conversion at a time. This is really not optimized and do not use all ADC capabilities.
You can try to play with `ADC_SAMPLINGTIME` and `ADC_CLOCK_DIV` which can be redefined thanks build_opt.h.
Else use directly HAL/LL to play with ADC.

Bambofy, 2019-10-31:
ok thank you

Bambofy, 2019-10-31:
I'm trying to use the HAL api to set the prescaler of the ADC like this:

```
void setup()
{
  Serial.begin(9600);
  
  // put your setup code here, to run once:
  pinMode(CONFIG_MIC_PIN, INPUT);
  lastTick = millis();

  ADC_HandleTypeDef adcConfiguration;
  adcConfiguration.Instance = ADC1;
  adcConfiguration.Init.ClockPrescaler = ADC_CLOCKPRESCALER_PCLK_DIV1;
  adcConfiguration.Init.Resolution = ADC_RESOLUTION_8B;
  adcConfiguration.Init.DataAlign = ADC_DATAALIGN_LEFT;
  adcConfiguration.Init.ScanConvMode = DISABLE;
  adcConfiguration.Init.EOCSelection = ADC_EOC_SINGLE_CONV;
  adcConfiguration.Init.LowPowerAutoWait = ENABLE;

  Serial.println(adcConfiguration.Init.ClockPrescaler);
  
  HAL_StatusTypeDef result = HAL_ADC_Init(&adcConfiguration);

  // 0 = ok
  // 1 = error
  // 2 = busy
  // 3 = timeout
  Serial.println(result); // returns 1
}
```

But it still samples at 9KHz? The analog pin is A0 / PA0?

fpistm, 2019-10-31:
which board ?

Bambofy, 2019-10-31:
> 
> 
> which board ?

NUCLEO-L452RE-P

fpistm, 2019-10-31:
and how you read value ?

Bambofy, 2019-10-31:
> 
> 
> and how you read value ?

```
void loop()
{
  byte sample = analogRead(CONFIG_MIC_PIN); // this should be 8 bits now.
 }
```

fpistm, 2019-10-31:
Using `analogRead` call again the `HAL_ADC_Init` so your setup is simply ignore.

Bambofy, 2019-10-31:
when i poll for conversion, the function HAL_ADC_GetValue() returns nothing.

fpistm, 2019-10-31:
Here an example:
https://github.com/STMicroelectronics/STM32CubeL4/blob/f6877af03b5c8b666b3443f5295cf95ec9c0b15d/Projects/NUCLEO-L452RE/Examples/ADC/ADC_RegularConversion_Polling/Src/main.c#L52

you have to call start and poll for conversion, I guess

fpistm, 2019-10-31:
Ah! Also call first `analogRead(CONFIG_MIC_PIN)` to init the `g_current_pin` then the `HAL_ADC_MspInit`
will enable the clock and properly init the pin.

So, in `Setup()`:
```C++
analogRead(CONFIG_MIC_PIN);
<your init code>
```

Bambofy, 2019-10-31:
how do i find out what channel my analog pin is?

fpistm, 2019-10-31:
In the datasheet or in the PeripheralsPin.c

Bambofy, 2019-10-31:
Ok so both ADC init functions return OK but the code get stuck polling for a value?

```
#include "stm32l4xx_hal_adc.h"

const int CONFIG_MIC_PIN = PA0;

unsigned long lastTick;
unsigned long timeNow;
unsigned long numberOfLoopIterations = 0;

ADC_HandleTypeDef adcConfiguration;

void setup()
{
  Serial.begin(9600);
  
  // put your setup code here, to run once:
  //pinMode(CONFIG_MIC_PIN, INPUT);
  analogRead(CONFIG_MIC_PIN);
  lastTick = millis();

//  adcConfiguration.Init.ClockPrescaler = ADC_CLOCKPRESCALER_PCLK_DIV1;
//  adcConfiguration.Init.Resolution = ADC_RESOLUTION_8B;
//  adcConfiguration.Init.DataAlign = ADC_DATAALIGN_LEFT;
//  adcConfiguration.Init.ScanConvMode = DISABLE;
//  adcConfiguration.Init.EOCSelection = ADC_EOC_SINGLE_CONV;
//  adcConfiguration.Init.LowPowerAutoWait = ENABLE;
//  

  // ADC peripheral
  adcConfiguration.Instance = ADC1;

  adcConfiguration.Init.ClockPrescaler = ADC_CLOCKPRESCALER_PCLK_DIV2;
  adcConfiguration.Init.Resolution = ADC_RESOLUTION_12B;
  adcConfiguration.Init.ScanConvMode = DISABLE;
  adcConfiguration.Init.ContinuousConvMode = ENABLE;
  adcConfiguration.Init.DiscontinuousConvMode = DISABLE;
  adcConfiguration.Init.NbrOfDiscConversion = 0;
  adcConfiguration.Init.ExternalTrigConvEdge = ADC_EXTERNALTRIGCONVEDGE_NONE;
  adcConfiguration.Init.ExternalTrigConv = ADC_EXTERNALTRIG_T1_CC1;
  adcConfiguration.Init.DataAlign = ADC_DATAALIGN_RIGHT;
  adcConfiguration.Init.NbrOfConversion = 1;
  adcConfiguration.Init.DMAContinuousRequests = ENABLE;
  adcConfiguration.Init.EOCSelection = DISABLE;

  
  HAL_StatusTypeDef result = HAL_ADC_Init(&adcConfiguration);
  // 0 = ok
  // 1 = error
  // 2 = busy
  // 3 = timeout
  Serial.print("ADC Peripheral init: ");
  Serial.println(result); // returns 1

  
  
  // ADC channel
  ADC_ChannelConfTypeDef adcChannelConfiguration;
  adcChannelConfiguration.Channel = ADC_CHANNEL_1;
  adcChannelConfiguration.Rank = 1;
  adcChannelConfiguration.SamplingTime = LL_ADC_SAMPLINGTIME_92CYCLES_5;
  adcChannelConfiguration.Offset = 0;

  Serial.print("ADC Channel init: ");
  Serial.println(HAL_ADC_ConfigChannel(&adcConfiguration, &adcChannelConfiguration));

}

void loop()
{
  //byte sample = analogRead(CONFIG_MIC_PIN); // this should be 8 bits now.
  if (HAL_ADC_PollForConversion(&adcConfiguration, 1000000) == HAL_OK)
  {
      uint32_t g_ADCValue = HAL_ADC_GetValue(&adcConfiguration);
  //    //Serial.print("Value:");
      Serial.println(g_ADCValue);
      //Serial.println("");
  }
        
  timeNow = millis();

  // every 1 second.
  if ( (timeNow - lastTick) >= 1000)
  {
    Serial.println(numberOfLoopIterations);
    numberOfLoopIterations = 0;
    lastTick = timeNow;
  }
  
  numberOfLoopIterations++;
}
```

fpistm, 2019-10-31:
Could you try this, I'm at home and I've not the board.

```C++
const int CONFIG_MIC_PIN = PA0;

ADC_HandleTypeDef adcConfiguration = {};
ADC_ChannelConfTypeDef  AdcChannelConf = {};

void setup()
{
  Serial.begin(9600);

  // put your setup code here, to run once:
  analogRead(CONFIG_MIC_PIN);

  adcConfiguration.Init.ClockPrescaler = ADC_CLOCKPRESCALER_PCLK_DIV2;
  adcConfiguration.Init.Resolution = ADC_RESOLUTION_12B;
  adcConfiguration.Init.ScanConvMode = DISABLE;
  adcConfiguration.Init.ContinuousConvMode = ENABLE;
  adcConfiguration.Init.DiscontinuousConvMode = DISABLE;
  adcConfiguration.Init.NbrOfDiscConversion = 0;
  adcConfiguration.Init.ExternalTrigConv = ADC_EXTERNALTRIG_T1_CC1;
  adcConfiguration.Init.ExternalTrigConvEdge = ADC_EXTERNALTRIGCONVEDGE_NONE;
  adcConfiguration.Init.DataAlign = ADC_DATAALIGN_RIGHT;
  adcConfiguration.Init.NbrOfConversion = 1;
  adcConfiguration.Init.DMAContinuousRequests = ENABLE;
  adcConfiguration.Init.EOCSelection = DISABLE;

  if ( HAL_ADC_Init(&adcConfiguration) != HAL_OK) {
    Serial.println("HAL_ADC_Init failed");
    while (1);
  }
  AdcChannelConf.Channel      = ADC_CHANNEL_5;                          /* Specifies the channel to configure into ADC */
  AdcChannelConf.Rank         = ADC_REGULAR_RANK_1;               /* Specifies the rank in the regular group sequencer */
  /*##-2- Configure ADC regular channel ######################################*/
  if (HAL_ADC_ConfigChannel(&adcConfiguration, &AdcChannelConf) != HAL_OK) {
    /* Channel Configuration Error */
    Serial.println("HAL_ADC_ConfigChan failed");
    while (1);
  }

  if (HAL_ADCEx_Calibration_Start(&adcConfiguration, ADC_SINGLE_ENDED) !=  HAL_OK) {
    Serial.println("HAL_ADCEx_Calibration_Start failed");
    while (1);
  }
}

void loop()
{
  if (HAL_ADC_Start(&adcConfiguration) != HAL_OK) {
    /* Start Conversation Error */
    Serial.println("HAL_ADC_Start failed");
    while (1);
  }
  //byte sample = analogRead(CONFIG_MIC_PIN); // this should be 8 bits now.
  if (HAL_ADC_PollForConversion(&adcConfiguration, 10) == HAL_OK)
  {
    if ((HAL_ADC_GetState(&adcConfiguration) & HAL_ADC_STATE_REG_EOC) == HAL_ADC_STATE_REG_EOC) {
      Serial.println(HAL_ADC_GetValue(&adcConfiguration));
    }
  }
}
```

Bambofy, 2019-10-31:
> 
> 
> Could you try this, I'm at home and I've not the board.
> 
> ```c++
> const int CONFIG_MIC_PIN = PA0;
> 
> ADC_HandleTypeDef adcConfiguration = {};
> ADC_ChannelConfTypeDef  AdcChannelConf = {};
> 
> void setup()
> {
>   Serial.begin(9600);
> 
>   // put your setup code here, to run once:
>   analogRead(CONFIG_MIC_PIN);
> 
>   adcConfiguration.Init.ClockPrescaler = ADC_CLOCKPRESCALER_PCLK_DIV2;
>   adcConfiguration.Init.Resolution = ADC_RESOLUTION_12B;
>   adcConfiguration.Init.ScanConvMode = DISABLE;
>   adcConfiguration.Init.ContinuousConvMode = ENABLE;
>   adcConfiguration.Init.DiscontinuousConvMode = DISABLE;
>   adcConfiguration.Init.NbrOfDiscConversion = 0;
>   adcConfiguration.Init.ExternalTrigConv = ADC_EXTERNALTRIG_T1_CC1;
>   adcConfiguration.Init.ExternalTrigConvEdge = ADC_EXTERNALTRIGCONVEDGE_NONE;
>   adcConfiguration.Init.DataAlign = ADC_DATAALIGN_RIGHT;
>   adcConfiguration.Init.NbrOfConversion = 1;
>   adcConfiguration.Init.DMAContinuousRequests = ENABLE;
>   adcConfiguration.Init.EOCSelection = DISABLE;
> 
> 
>   if ( HAL_ADC_Init(&adcConfiguration) != HAL_OK) {
>     Serial.println("HAL_ADC_Init failed");
>     while (1);
>   }
>   AdcChannelConf.Channel      = ADC_CHANNEL_5;                          /* Specifies the channel to configure into ADC */
>   AdcChannelConf.Rank         = ADC_REGULAR_RANK_1;               /* Specifies the rank in the regular group sequencer */
>   /*##-2- Configure ADC regular channel ######################################*/
>   if (HAL_ADC_ConfigChannel(&adcConfiguration, &AdcChannelConf) != HAL_OK) {
>     /* Channel Configuration Error */
>     Serial.println("HAL_ADC_ConfigChan failed");
>     while (1);
>   }
> 
>   if (HAL_ADCEx_Calibration_Start(&adcConfiguration, ADC_SINGLE_ENDED) !=  HAL_OK) {
>     Serial.println("HAL_ADCEx_Calibration_Start failed");
>     while (1);
>   }
> }
> 
> void loop()
> {
>   if (HAL_ADC_Start(&adcConfiguration) != HAL_OK) {
>     /* Start Conversation Error */
>     Serial.println("HAL_ADC_Start failed");
>     while (1);
>   }
> 
>   if (HAL_ADC_PollForConversion(&adcConfiguration, 10) == HAL_OK)
>   {
>     if ((HAL_ADC_GetState(&adcConfiguration) & HAL_ADC_STATE_REG_EOC) == HAL_ADC_STATE_REG_EOC) {
>       Serial.println(HAL_ADC_GetValue(&adcConfiguration));
>     }
>   }
> }
> ```

It has "HAL_ADC_Init failed"

Bambofy, 2019-10-31:
This code is working but when i do HAL_ADC_GetValue i get no value

fpistm, 2019-10-31:
I use your config, so maybe something is wrong.
Please check.

fpistm, 2019-10-31:
HAL_ADC_Start(&adcConfiguration); should be call in the loop.

fpistm, 2019-10-31:
`ADC_EOC_SINGLE_CONV ` should be set to use `HAL_ADC_PollForConversion`
`  adcConfiguration.Init.EOCSelection          = ADC_EOC_SINGLE_CONV;           /* EOC flag picked-up to indicate conversion end */`

Bambofy, 2019-10-31:
ok i'm still testing, how do i define which pin i want to use?

Bambofy, 2019-10-31:
ok heres my current code:
```
#include "stm32l4xx_hal_adc.h"

const int CONFIG_MIC_PIN = PA0;

unsigned long lastTick;
unsigned long timeNow;
unsigned long numberOfLoopIterations = 0;

ADC_HandleTypeDef adcConfiguration;

void SetupADC()
{
  //  adcConfiguration.Init.ClockPrescaler = ADC_CLOCKPRESCALER_PCLK_DIV1;
//  adcConfiguration.Init.Resolution = ADC_RESOLUTION_8B;
//  adcConfiguration.Init.DataAlign = ADC_DATAALIGN_LEFT;
//  adcConfiguration.Init.ScanConvMode = DISABLE;
//  adcConfiguration.Init.EOCSelection = ADC_EOC_SINGLE_CONV;
//  adcConfiguration.Init.LowPowerAutoWait = ENABLE;
//  

  // GPIO config

  // ADC peripheral
  adcConfiguration.Instance = ADC1;

  adcConfiguration.Init.ClockPrescaler = ADC_CLOCKPRESCALER_PCLK_DIV1;
  adcConfiguration.Init.Resolution = ADC_RESOLUTION_8B;
  adcConfiguration.Init.ScanConvMode = DISABLE;
  adcConfiguration.Init.ContinuousConvMode = ENABLE;
  adcConfiguration.Init.DiscontinuousConvMode = DISABLE;
  adcConfiguration.Init.NbrOfDiscConversion = 0;
  adcConfiguration.Init.ExternalTrigConvEdge = ADC_EXTERNALTRIGCONVEDGE_NONE;
  adcConfiguration.Init.ExternalTrigConv = ADC_EXTERNALTRIG_T1_CC1;
  adcConfiguration.Init.DataAlign = ADC_DATAALIGN_RIGHT;
  adcConfiguration.Init.NbrOfConversion = 1;
  adcConfiguration.Init.DMAContinuousRequests = ENABLE;
  adcConfiguration.Init.EOCSelection = ADC_EOC_SINGLE_CONV;

  
  HAL_StatusTypeDef result = HAL_ADC_Init(&adcConfiguration);
  // 0 = ok
  // 1 = error
  // 2 = busy
  // 3 = timeout
  Serial.print("ADC Peripheral init: ");
  Serial.println(result); // returns 1

  
  
  // ADC channel
  ADC_ChannelConfTypeDef adcChannelConfiguration;
  adcChannelConfiguration.Channel = ADC_CHANNEL_2;
  adcChannelConfiguration.Rank = 1;
  adcChannelConfiguration.SamplingTime = ADC_SAMPLETIME_2CYCLES_5;
  adcChannelConfiguration.Offset = 0;
  adcChannelConfiguration.SingleDiff = ADC_SINGLE_ENDED;
  adcChannelConfiguration.OffsetNumber = ADC_OFFSET_NONE;

  Serial.print("ADC Channel init: ");
  Serial.println(HAL_ADC_ConfigChannel(&adcConfiguration, &adcChannelConfiguration));

  Serial.print("ADC Start: ");
  Serial.println(HAL_ADC_Start(&adcConfiguration));
}

void setup()
{
  Serial.begin(9600);
  
  // put your setup code here, to run once:
  pinMode(CONFIG_MIC_PIN, INPUT);
  lastTick = millis();
}

void loop()
{
  SetupADC();
  
  //byte sample = analogRead(CONFIG_MIC_PIN); // this should be 8 bits now.
  if (HAL_ADC_PollForConversion(&adcConfiguration, 100000) == HAL_OK)
  {
      uint8_t sample = HAL_ADC_GetValue(&adcConfiguration);
     Serial.print("Value:");
     Serial.println(sample);
      //Serial.println("");
  }
        
  timeNow = millis();

  // every 1 second.
  if ( (timeNow - lastTick) >= 1000)
  {
    Serial.println(numberOfLoopIterations);
    numberOfLoopIterations = 0;
    lastTick = timeNow;
  }
  
  numberOfLoopIterations++;
}
```

fpistm, 2019-10-31:
Well, I will test on monday and check why it does not work.
As I said I'm at home with no board so hard to check.

Bambofy, 2019-10-31:
it's alive!! you have to call Start() then Poll() then Stop() for the process to get a new value :) 👍  here's a working sketch (analog input on pin A4/PC0)

`#include "stm32l4xx_hal_adc.h"

const int CONFIG_MIC_PIN = PA0;

unsigned long lastTick;
unsigned long timeNow;
unsigned long numberOfLoopIterations = 0;

ADC_HandleTypeDef adcConfiguration;

void SetupADC()
{
  //  adcConfiguration.Init.ClockPrescaler = ADC_CLOCKPRESCALER_PCLK_DIV1;
//  adcConfiguration.Init.Resolution = ADC_RESOLUTION_8B;
//  adcConfiguration.Init.DataAlign = ADC_DATAALIGN_LEFT;
//  adcConfiguration.Init.ScanConvMode = DISABLE;
//  adcConfiguration.Init.EOCSelection = ADC_EOC_SINGLE_CONV;
//  adcConfiguration.Init.LowPowerAutoWait = ENABLE;
//  

  // GPIO config

  // ADC peripheral
  adcConfiguration.Instance = ADC1;

  adcConfiguration.Init.ClockPrescaler = ADC_CLOCKPRESCALER_PCLK_DIV1;
  adcConfiguration.Init.Resolution = ADC_RESOLUTION_12B;
  adcConfiguration.Init.ScanConvMode = DISABLE;
  adcConfiguration.Init.ContinuousConvMode = ENABLE;
  adcConfiguration.Init.DiscontinuousConvMode = ENABLE;
  adcConfiguration.Init.NbrOfDiscConversion = 0;
  adcConfiguration.Init.ExternalTrigConvEdge = ADC_EXTERNALTRIGCONVEDGE_NONE;
  adcConfiguration.Init.ExternalTrigConv = ADC_EXTERNALTRIG_T1_CC1;
  adcConfiguration.Init.DataAlign = ADC_DATAALIGN_RIGHT;
  adcConfiguration.Init.NbrOfConversion = 1;
  adcConfiguration.Init.DMAContinuousRequests = ENABLE;
  adcConfiguration.Init.EOCSelection = ADC_EOC_SEQ_CONV;

  
  HAL_StatusTypeDef result = HAL_ADC_Init(&adcConfiguration);
  // 0 = ok
  // 1 = error
  // 2 = busy
  // 3 = timeout
  Serial.print("ADC Peripheral init: ");
  Serial.println(result); // returns 1

  
  
  // ADC channel
  ADC_ChannelConfTypeDef adcChannelConfiguration;
  adcChannelConfiguration.Channel = ADC_CHANNEL_1;
  adcChannelConfiguration.Rank = 1;
  adcChannelConfiguration.SamplingTime = ADC_SAMPLETIME_2CYCLES_5;
  adcChannelConfiguration.Offset = 0;
  adcChannelConfiguration.SingleDiff = ADC_SINGLE_ENDED;
  adcChannelConfiguration.OffsetNumber = ADC_OFFSET_NONE;

  Serial.print("ADC Channel init: ");
  Serial.println(HAL_ADC_ConfigChannel(&adcConfiguration, &adcChannelConfiguration));
}

void setup()
{
  Serial.begin(9600);
 
  SetupADC();

  lastTick = millis();
}

void loop()
{
  // start the conversion process
  HAL_ADC_Start(&adcConfiguration);
  
  //wait for the conversion to finish
  if (HAL_ADC_PollForConversion(&adcConfiguration, 100000) == HAL_OK)
  {
    // conversion completed successfully.
    uint16_t sample = HAL_ADC_GetValue(&adcConfiguration);
    //Serial.print("Value:");
    Serial.println(sample);
  }

  HAL_ADC_Stop(&adcConfiguration);
        
  timeNow = millis();

  // every 1 second.
  if ( (timeNow - lastTick) >= 1000)
  {
    //Serial.println(numberOfLoopIterations);
    numberOfLoopIterations = 0;
    lastTick = timeNow;
  }
  
  numberOfLoopIterations++;
}`

fpistm, 2019-10-31:
Great!

Bambofy, 2019-10-31:
> 
> 
> Great!

Just gotta figure out how to configure which GPIO pin it is reading haha

Bambofy, 2019-10-31:
Figured it out :) The answer is to change which "Channel" the channel configuration is listening to. For the NUCLEO-L452RE-P it is here https://www.st.com/content/ccc/resource/technical/document/user_manual/group0/ff/5d/51/50/db/12/47/98/DM00387966/files/DM00387966.pdf/jcr:content/translations/en.DM00387966.pdf at page 36.

Looking at a row in the table:
PA0 -> ADC1_IN5 means for ADC 1 the channel 5.

fpistm, 2019-10-31:
Right, that's why in my example code i've set `ADC_CHANNEL_5`  😉

Bambofy, 2019-11-07:
Hey, sorry to bring this issue up again but i'm struggeling to change the pin that the data in being input to, see this code:

```
#include "stm32l4xx_hal_adc.h"

const int CONFIG_MIC_PIN = PC1;

unsigned long lastTick;
unsigned long timeNow;
unsigned long numberOfLoopIterations = 0;

ADC_HandleTypeDef adcConfiguration;

void SetupADC()
{
  // GPIO config
  __GPIOC_CLK_ENABLE();
  
  GPIO_InitTypeDef gpioConfiguration;
  gpioConfiguration.Pin = GPIO_PIN_1; // PC1
  gpioConfiguration.Mode = GPIO_MODE_ANALOG;
  gpioConfiguration.Pull = GPIO_NOPULL;
  HAL_GPIO_Init(GPIOC, &gpioConfiguration);

  //pinMode(CONFIG_MIC_PIN, INPUT);
    
  // ADC peripheral
  adcConfiguration.Instance = ADC1;

  adcConfiguration.Init.ClockPrescaler = ADC_CLOCKPRESCALER_PCLK_DIV1;
  adcConfiguration.Init.Resolution = ADC_RESOLUTION_8B;
  adcConfiguration.Init.ScanConvMode = DISABLE;
  adcConfiguration.Init.ContinuousConvMode = ENABLE;
  adcConfiguration.Init.DiscontinuousConvMode = ENABLE;
  adcConfiguration.Init.NbrOfDiscConversion = 0;
  adcConfiguration.Init.ExternalTrigConvEdge = ADC_EXTERNALTRIGCONVEDGE_NONE;
  adcConfiguration.Init.ExternalTrigConv = ADC_EXTERNALTRIG_T1_CC1;
  adcConfiguration.Init.DataAlign = ADC_DATAALIGN_RIGHT;
  adcConfiguration.Init.NbrOfConversion = 1;
  adcConfiguration.Init.DMAContinuousRequests = ENABLE;
  adcConfiguration.Init.EOCSelection = ADC_EOC_SEQ_CONV;

  
  HAL_StatusTypeDef result = HAL_ADC_Init(&adcConfiguration);
  // 0 = ok
  // 1 = error
  // 2 = busy
  // 3 = timeout
  Serial.print("ADC Peripheral init: ");
  Serial.println(result); // returns 1

  
  
  // ADC channel
  ADC_ChannelConfTypeDef adcChannelConfiguration;
  adcChannelConfiguration.Channel = ADC_CHANNEL_2;
  adcChannelConfiguration.Rank = 1;
  adcChannelConfiguration.SamplingTime = ADC_SAMPLETIME_2CYCLES_5;
  adcChannelConfiguration.Offset = 0;
  adcChannelConfiguration.SingleDiff = ADC_SINGLE_ENDED;
  adcChannelConfiguration.OffsetNumber = ADC_OFFSET_NONE;

  Serial.print("ADC Channel init: ");
  Serial.println(HAL_ADC_ConfigChannel(&adcConfiguration, &adcChannelConfiguration));
}

void setup()
{
  Serial.begin(9600);
 
  SetupADC();

  lastTick = millis();
}

void loop()
{
  // start the conversion process
  HAL_ADC_Start(&adcConfiguration);
  
  //wait for the conversion to finish
  if (HAL_ADC_PollForConversion(&adcConfiguration, 100000) == HAL_OK)
  {
    // conversion completed successfully.
    uint16_t sample = HAL_ADC_GetValue(&adcConfiguration);
    //Serial.print("Value:");
    Serial.println(sample);
  }

  HAL_ADC_Stop(&adcConfiguration);
        
  timeNow = millis();

  // every 1 second.
  if ( (timeNow - lastTick) >= 1000)
  {
    //Serial.println(numberOfLoopIterations);
    numberOfLoopIterations = 0;
    lastTick = timeNow;
  }
  
  numberOfLoopIterations++;
}
```

fpistm, 2019-11-07:
@Bambofy 
I could not debug all code users do.
Anyway, as I said before the call of `analogRead()` one time allow to use the `HAL_ADC_MspInit`  which init the ADC clock. I didn't see any `__HAL_RCC_ADC_CLK_ENABLE();` in your code. I guess this could be a good track.

closed 2019-11-07 by fpistm

LMESTM, 2019-11-07:
@Bambofy 
I agree with @fpistm that he can't debug everyone's program :-)
Issues in stm32duino repo should refer to problems found inside the stm32duino core implementation, not in program using them.
Moreover, you are not using stm32duino core / arduino APIs but rather accessing directly to HAL layer from stm32 CubeFW. This is a very valid approach, but you hopefully could find more adequate support around here: https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus

Bambofy, 2020-01-15:
See here for detailed ADC usage:
- https://github.com/stm32duino/Arduino_Core_STM32/blob/master/system/Drivers/STM32L4xx_HAL_Driver/Inc/stm32l4xx_hal_adc.h
- https://github.com/stm32duino/Arduino_Core_STM32/blob/master/system/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_adc.c