stm32duino/Arduino_Core_STM32 #754: SDADC implementation

michsens, 2019-11-04:
How can I use the Sigma-Delta ADC (SDADC) in the Arduino environment? The "analogRead" command belongs to the SAR ADCs from what I understand. Going the HAL way like I did in the TrueStudio seems not to work even all the nesseccary libs are available. 

My questions:
1.  How can I use the HAL SDADC functions?
2. Is there a roadmap to implement the SDADC native into Arduino?
3. How can I contribute to the implementation?

Regards.

michsens, 2019-11-04:
Here is my code for hal initialization:

`
SDADC_HandleTypeDef hsdadc;

void Error_Handler_2(uint32_t dacValue)
{
  /* USER CODE BEGIN Error_Handler_Debug */
  /* User can add his own implementation to report the HAL error return state */
  analogWrite(DA_PWM_2,dacValue);
  delay(5000);
  /* USER CODE END Error_Handler_Debug */
}

static void MX_SDADC1_Init(void)
{

  SDADC_ConfParamTypeDef confParam;

  hsdadc.Instance = SDADC1;
  hsdadc.Init.ReferenceVoltage = SDADC_VREF_VDDA;//SDADC_VREF_VDDA or SDADC_VREF_VREFINT2;
  hsdadc.Init.IdleLowPowerMode = SDADC_LOWPOWER_NONE;
  hsdadc.Init.FastConversionMode = SDADC_FAST_CONV_DISABLE;
  hsdadc.Init.SlowClockMode = SDADC_SLOW_CLOCK_DISABLE;

  /* -1- Initialise the SDADC */
  if (HAL_SDADC_Init(&hsdadc) != HAL_OK)
  {
    Error_Handler_2(2000);
  }

  /* -2- Prepare the channel configuration */
  confParam.CommonMode = SDADC_COMMON_MODE_VSSA;//_2;
  confParam.Gain = SDADC_GAIN_1_2;
  confParam.InputMode = SDADC_INPUT_MODE_SE_OFFSET; //SDADC_INPUT_MODE_SE_ZERO_REFERENCE; //SDADC_INPUT_MODE_SE_OFFSET
  confParam.Offset = 0x0000000F;

  if (HAL_SDADC_PrepareChannelConfig(&hsdadc, SDADC_CONF_INDEX_0, &confParam) != HAL_OK)
  {
    Error_Handler_2(2500);
  }

  /* Select software trigger for regular conversion */
  if (HAL_SDADC_SelectRegularTrigger(&hsdadc, SDADC_SOFTWARE_TRIGGER) != HAL_OK)
  {
    Error_Handler_2(3000);
  }
  /* Start Calibration in polling mode */
  if (HAL_SDADC_CalibrationStart(&hsdadc, SDADC_CALIBRATION_SEQ_1) != HAL_OK)
  {
    Error_Handler_2(3500);
  }

  /* Pool for the end of calibration */
  if (HAL_SDADC_PollForCalibEvent(&hsdadc, HAL_MAX_DELAY) != HAL_OK)
  {
    Error_Handler_2(4000);
  }

}

static uint32_t readSelectedSDADCChannel(SDADC_HandleTypeDef* hsdadc, uint32_t channel_number) //uint32_t Channel
{
  /* Associate the channel of PT100 sensor to the configuration 0 */
  if (HAL_SDADC_AssociateChannelConfig(hsdadc, channel_number, SDADC_CONF_INDEX_0) != HAL_OK)
  {
    Error_Handler_2(100);
  }

  /* Select the channel of PT100 for a regular conversion and not for continuous mode */
  if (HAL_SDADC_ConfigChannel(hsdadc, channel_number, SDADC_CONTINUOUS_CONV_OFF) != HAL_OK)
  {
    Error_Handler_2(500);
  }

  if (HAL_SDADC_Start(hsdadc) != HAL_OK)
  {
    Error_Handler_2(1000);
  }

  /* wait for end of regular conversion */
  if (HAL_SDADC_PollForConversion(hsdadc, HAL_MAX_DELAY) != HAL_OK)
  {
    Error_Handler_2(1500);
  }
  HAL_Delay(100);
  return HAL_SDADC_GetValue(hsdadc);
}

`

The compilation finishs with success. But after uploading the mcu runs into the error handlers. For a quick debug I throw the error hanler with a value written by the already used dac. So I can see with a multimeter which error handler is thrown. The only one that is not is the 
`
if (HAL_SDADC_Init(&hsdadc) != HAL_OK)
  {
    Error_Handler_2(2000);
  }
`

In the TrueStudio the whole code is working like a charm.

Please give me a hint.

fpistm, 2019-11-04:
> 1\. How can I use the HAL SDADC functions?

No idea, I never test this feature. And seems only available for some F3.

> 2\. Is there a roadmap to implement the SDADC native into Arduino?

No, moreover it seems very specific to some F3.

> 3\. How can I contribute to the implementation?

Maybe, by writing a built-in library to handle this. Maybe could be part of #5 

About issue with your code, I don't know probably a init part is missing. Clock? Power?
The only way to get an HAL error returned by `HAL_SDADC_Init` is to get `hsdadc`  equal to `NULL`.
Other issue is to be in an infinite wait loop.
Check in the project you generate with CubeMx the `HAL_SDADC_MspInit` function and also the `SystemClock_Config`.

michsens, 2019-11-04:
I found the issue. It really belonged to a reference of a missing lib. Within TrueStudio a file is generated that initializes special peripherials. The file's name is "stm32f3xx_hal_msp.c". It includes equal routines for the analog parts as you find in the "analog.c".
Copying the "stm32f3xx_hal_msp.c" into the variants folder and modifying it to the specific needs, directly includes it into the core since it is included per default in the "stm32f3xx_hal_sdadc.c".
Thus, it must not be implemented in the "analog.c" for now. 
I am going to modify the code to the common arduino style. Maybe later it could be included in the "analog.c".

fpistm, 2019-11-04:
> Check in the project you generate with CubeMx the `HAL_SDADC_MspInit` function and also the `SystemClock_Config`.

That's is exactly the point I've raised.

fpistm, 2019-11-05:
As you now able to use this feature on your own.
I close this as it could be part of #5

closed 2019-11-05 by fpistm

michsens, 2019-11-05:
Thanks for now. I give some response when I get the code working in Ardunio style.