stm32duino/Arduino_Core_STM32 #756: Unable to compile Half-Duplex serial example.

AndKe, 2019-11-04:
I am trying to compile Half-Duplex Example from Wiki about API

Board: Generic STMF1
Board Part : Bluepill F103C8
U(S)ART support: Enabled(generic Serial)
USB Support: CDC(generic "serial" supersede U(S)ART"
USB Speed: Low/Full Speed
Optimize: Smallest (-Os default)
C Runtime Library : newlib Nano(default)

my code: (basically wiki, search'n'replaced Serial4 with Serial2 because STM32F103 don't have Serial4)

```
HardwareSerial Serial3(PA0);
HardwareSerial Serial2(PB10);

//HardwareSerial Serial3(PA_0);
//HardwareSerial Serial2(PB_10);

//HardwareSerial Serial3(UART4, true);
//HardwareSerial Serial2(USART3, true);

//HardwareSerial Serial3(PA0, PA0);
//HardwareSerial Serial2(PB10, PB10);

//HardwareSerial Serial3(PA_0, PA_0);
//HardwareSerial Serial2(PB_10, PB_10);

//HardwareSerial Serial3(NC, PA_0);
//HardwareSerial Serial2(NC, PB_10);

//HardwareSerial Serial3(NUM_DIGITAL_PINS, PA0);
//HardwareSerial Serial2(NUM_DIGITAL_PINS, PB10);

static uint32_t nbTestOK = 0;
static uint32_t nbTestKO = 0;
void test_uart(int val)
{
  int recval = -1;
  uint32_t error = 0;

  Serial2.write(val);
  delay(10);
  while (Serial3.available()) {
    recval = Serial3.read();
  }
  /* Enable Serial2 to RX*/
  Serial2.enableHalfDuplexRx();
  if (val == recval) {
    Serial3.write(val);
    delay(10);

    while (Serial2.available()) {
      recval = Serial2.read();
    }
    /* Enable Serial3 to RX*/
    Serial3.enableHalfDuplexRx();
    if (val == recval) {
      nbTestOK++;
      Serial.print("Exchange: 0x");
      Serial.println(recval, HEX);
    } else {
      error = 2;
    }
  }
  else {
    error = 1;
  }
  if (error) {
    Serial.print("Send: 0x");
    Serial.print(val, HEX);
    Serial.print("\tReceived: 0x");
    Serial.print(recval, HEX);
    Serial.print(" --> KO <--");
    Serial.println(error);
    nbTestKO++;
  }
}

void setup() {
  Serial.begin(115200);
  Serial2.begin(9600);
  Serial3.begin(9600);
}

void loop() {

  for (uint32_t i = 0; i <= (0xFF); i++) {
    test_uart(i);
  }

  Serial.println("Serial Half-Duplex test done.\nResults:");
  Serial.print("OK: ");
  Serial.println(nbTestOK);
  Serial.print("KO: ");
  Serial.println(nbTestKO);
  while (1);
}
```

The error I get:

```
....
Multiple libraries were found for "SrcWrapper.h"
 Used: /home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/libraries/SrcWrapper
/home/andre/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_cache_284773/core/core_d2d4f59cd5ddb04efe7cc8993937479a.a(HardwareSerial.cpp.o):(.bss.Serial2+0x0): multiple definition of `Serial2'; /tmp/arduino/sketch/temp-sketch_nov04a.ino.cpp.o:(.bss.Serial2+0x0): first defined here
collect2: error: ld returned 1 exit status
Using library SrcWrapper at version 1.0.0 in folder: /home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/libraries/SrcWrapper 
exit status 1
Error compiling for board Generic STM32F1 series.
```

It does help to comment out "HardwareSerial Serial2(PB10);"  
- but then, but would one tell it to use PB10 ?

fpistm, 2019-11-04:
As it is wrote in the Wiki:
https://github.com/stm32duino/wiki/wiki/API#enable-half-duplex-mode

> Available in core version greater than 1.7.0

So currently this is only available using the git repo.

AndKe, 2019-11-04:
Sorry, forgot to mention that I downloaded master and extracted into ~/.arduino15/packages/STM32/hardware/stm32/1.7.0

AndKe, 2019-11-10:
So in the greed for ta third serial port (CDC) I discovered that this:

```
HardwareSerial Serial3(PA2);
HardwareSerial Serial2(NC, PB_10);

//HardwareSerial Serial3(NUM_DIGITAL_PINS, PA0);
//HardwareSerial Serial2(NUM_DIGITAL_PINS, PB10);

static uint32_t nbTestOK = 0;
static uint32_t nbTestKO = 0;
void test_uart(int val)
{
  int recval = -1;
  uint32_t error = 0;

  Serial2.write(val);
  delay(10);
  while (Serial3.available()) {
    recval = Serial3.read();
  }
  /* Enable Serial4 to RX*/
  Serial2.enableHalfDuplexRx();
  if (val == recval) {
    Serial3.write(val);
    delay(10);

    while (Serial2.available()) {
      recval = Serial2.read();
    }
    /* Enable Serial3 to RX*/
    Serial3.enableHalfDuplexRx();
    if (val == recval) {
      nbTestOK++;
      Serial.print("Exchange: 0x");
      Serial.println(recval, HEX);
    } else {
      error = 2;
    }
  }
  else {
    error = 1;
  }
  if (error) {
    Serial.print("Send: 0x");
    Serial.print(val, HEX);
    Serial.print("\tReceived: 0x");
    Serial.print(recval, HEX);
    Serial.print(" --> KO <--");
    Serial.println(error);
    nbTestKO++;
  }
}

void setup() {
  Serial.begin(115200);
  Serial2.begin(9600);
  Serial3.begin(9600);
}

void loop() {

  for (uint32_t i = 0; i <= (0xFF); i++) {
    test_uart(i);
  }

  Serial.println("Serial Half-Duplex test done.\nResults:");
  Serial.print("OK: ");
  Serial.println(nbTestOK);
  Serial.print("KO: ");
  Serial.println(nbTestKO);
  while (1);
}
```

Compiles fine as long as: "USB Support" is set to "None"  , if set to :
 "CDC (Generic serials supersuede U(S)ART)"

then it fails like this:

```
/home/andre/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_cache_83/core/core_d2d4f59cd5ddb04efe7cc8993937479a.a(HardwareSerial.cpp.o):(.bss.Serial3+0x0): multiple definition of `Serial3'; /tmp/arduino/sketch/temp_sketch_oct25a.ino.cpp.o:(.bss.Serial3+0x0): first defined here
/home/andre/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_cache_83/core/core_d2d4f59cd5ddb04efe7cc8993937479a.a(HardwareSerial.cpp.o):(.bss.Serial2+0x0): multiple definition of `Serial2'; /tmp/arduino/sketch/temp_sketch_oct25a.ino.cpp.o:(.bss.Serial2+0x0): first defined here
collect2: error: ld returned 1 exit status
```

AndKe, 2019-11-10:
Something strange is going on, it seems that "USB Support" does affect it, but sometimes, after a complete IDE restart, it works, then stops after playing with that setting, and IDE may even be restarted, and still fail to compile.

fpistm, 2019-11-11:
I made several build with or without USB and I do not had error.
I think the issue comes from the Arduino cache ( `/tmp/arduino_cache_83`) which is not properly refreshed according to the option you set but I can't tell why.
This is an Arduino feature.

fpistm, 2019-11-11:
I've already had some issue due to my antivirus real time analysis enabled. If you have one try to disable it. Else you can submit an issue in Arduino github.

closed 2019-11-11 by fpistm