stm32duino/Arduino_Core_STM32 #757: Could not blink with STM32 bue pill

yeckel, 2019-11-05:
Hi, I'm trying to run a blink example app on STM32F103C8T6 blue board. And it doesn't work. 
I've Board: Generic STM32f1 P/N: Blue Pill F103F8 using STMCube programmer. With mBed it works so the hardware is probably fine. 

The code is (according gdb) stuck in stm32/1.7.0/variants/PILL_F103XX/variant.cpp
`void SystemClock_Config(void)`

```
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
    while (1);
  }
```
Is it a bug or a feature? Any tips? I was trying with an older version 1.3.X too without success.

fpistm, 2019-11-05:
Hi @yeckel 
If the `HAL_RCC_OscConfig` failed on your side this is probably due to an issue with the HSE on the board.
On mbed, it tries to use HSE, then HSE bypass and finally HSI.
https://github.com/ARMmbed/mbed-os/blob/cdcef4babe74e9d4cf3ea8262b359be3eeed2970/targets/TARGET_STM/TARGET_STM32F1/TARGET_BLUEPILL_F103C8/device/system_clock.c#L134-L150

So I guess MBed fallback to HSI.

fpistm, 2019-11-05:
Note: `SystemClock_Config` is a weaked function so you can override it at sketch level to test:
```C
extern "C" void SystemClock_Config(void) {
...
}
```

yeckel, 2019-11-05:
Bad HW, that's what I've thought. You mean by replacing HSE (crystal) with HSI (local oscilator), like mBed does? 

```
    RCC_OscInitStruct.OscillatorType      = RCC_OSCILLATORTYPE_HSI | RCC_OSCILLATORTYPE_HSE;
    RCC_OscInitStruct.HSIState            = RCC_HSI_ON;
    RCC_OscInitStruct.HSEState            = RCC_HSE_OFF;
    RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
    RCC_OscInitStruct.PLL.PLLState        = RCC_PLL_ON;
    RCC_OscInitStruct.PLL.PLLSource       = RCC_PLLSOURCE_HSI_DIV2;
    RCC_OscInitStruct.PLL.PLLMUL          = RCC_PLL_MUL16; // 64 MHz (8 MHz/2 * 16)
```

yeckel, 2019-11-05:
Ok, it helped slightly. Now it's idling in WWDG_IRQHandler() which I have no idea where it is. There is just HAL_WWDG_IRQHandler. From the sketch even the setup() is not called. If there's no easy solution I'll wait for new HW. Thanks!

fpistm, 2019-11-05:
You could try this one:

```C
extern "C" void SystemClock_Config(void) {
  RCC_OscInitTypeDef RCC_OscInitStruct = {};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {};
  RCC_PeriphCLKInitTypeDef PeriphClkInit = {};

  /** Initializes the CPU, AHB and APB busses clocks
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI_DIV2;
  RCC_OscInitStruct.PLL.PLLMUL = RCC_PLL_MUL12;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }
  /** Initializes the CPU, AHB and APB busses clocks
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_SYSCLK
                                | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV2;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_1) != HAL_OK)
  {
    Error_Handler();
  }
  PeriphClkInit.PeriphClockSelection = RCC_PERIPHCLK_ADC | RCC_PERIPHCLK_USB;
  PeriphClkInit.AdcClockSelection = RCC_ADCPCLK2_DIV4;
  PeriphClkInit.UsbClockSelection = RCC_USBCLKSOURCE_PLL;
  if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInit) != HAL_OK)
  {
    Error_Handler();
  }
}
```

yeckel, 2019-11-06:
Now I'm stuck in UsageFault_handler() and sometimes in WWDG_IRQHandler(). I'll wait for new hardware. Thanks for help, I've learned a lot.

fpistm, 2019-11-06:
OK.
I've tested the SystemClock_Config I provided using HSI and it works.
I guess we are more strict in Arduino than MBed.
One other issue you can met is some BP are sold with another MCU than a STM32F103C8.
I already saw GD32 or STM32F103C6 that could lead to some issue like the one you have.

closed 2019-11-06 by fpistm

yeckel, 2019-11-13:
Today new blue pill arrive and it's blinking without any tweaks. I've checked the chip and they are the same.  I had probably a bad luck. Tnx

fpistm, 2019-11-13:
Thanks for the feedback @yeckel 
You probably get an issue around the HSE part (soldier, Xtal,...)