stm32duino/Arduino_Core_STM32 #758: Duplicate/unneeded code in premain?

matthijskooijman, 2019-11-06:
Looking around the code, I noticed some code in premain that seems duplicate:

https://github.com/stm32duino/Arduino_Core_STM32/blob/db1b90f4cc057c82d91535f82e8ffacf39eba8f0/cores/arduino/main.cpp#L34-L46

The same things seem to be done by `HAL_Init()` already, e.g. on F4:

https://github.com/stm32duino/Arduino_Core_STM32/blob/db1b90f4cc057c82d91535f82e8ffacf39eba8f0/system/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.c#L160-L173

This seems rather pointless, though perhaps when using RTOS the `init()` function might be replaced and `HAL_Init()` is not called? But then maybe the RTOS `init()` should just set up this NVIC stuff it needs by itself?

For the caches, it seems there are now two separate macros to control them, which could also be a bit confusing?

fpistm, 2019-11-06:
` #if (__CORTEX_M == 0x07U) `
F4 is not an M7.

matthijskooijman, 2019-11-06:
Ah, good point. Looking at the F7 (which does have an M7 AFAIU), it indeed does not initialize the caches in `HAL_Init()`: https://github.com/stm32duino/Arduino_Core_STM32/blob/db1b90f4cc057c82d91535f82e8ffacf39eba8f0/system/Drivers/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal.c#L138-L161. Any idea why this is different for different MCU series?

Still, it seems the NVIC enable is still duplicate?

fpistm, 2019-11-06:
> Any idea why this is different for different MCU series?

No

> Still, it seems the NVIC enable is still duplicate?

Right. Looking deeply it seems properly set for all STM32 series with `NVIC_PRIORITYGROUP_4`.
So it could probably removed without any issue while  it is set before the RTOS is started.

fpistm, 2019-11-07:
So, what did you expect on this?

matthijskooijman, 2019-11-07:
It was mostly something I noticed while reading the code that looked like it could be cleaned up. I wanted to share that observation, but it seems cleaning it up might be more involved (i.e. needs more research and testing), so perhaps it should just be left as-is for now. Feel free to close this if you agree :-)

fpistm, 2019-11-07:
I'm always ready for made some clean. 
In this case I think you spotted the right thing , it will require some study.

closed 2019-11-07 by fpistm