stm32duino/Arduino_Core_STM32 #759: Possible I2C bug

Luca-007, 2019-11-06:
**Describe the bug**
Adding the address XX to the Wire.begin(XX) command we have Wire.endTransmission() = 4 even with or without any other I2C device connected 

**To Reproduce**
use the scanner.ino example replacing 

`Wire.begin(); `
with  
`Wire.begin(70);
`
or any other I2C valid address.
 

**Desktop (please complete the following information):**

OS: win10
Arduino IDE version: 1.8.8
STM32 core version: 1.7.0 downloaded on 5th Nov 2019
Tools menu settings if not the default: default
Upload method: SWD

**Board (please complete the following information):**
Name: Bluepill f103c8
Hardware Revision: unknown 
Extra hardware used if any: none 
Additional context: none

fpistm, 2019-11-07:
Hi @Luca-007 
This is normal as when you set an @ and so you specify the board is a slave.
https://www.arduino.cc/en/Reference/WireBegin

Luca-007, 2019-11-07:
Hi Frederic, 
tnx for quick answer !

I'm not agree... it is not really normal to me 
please consider that: 

- on the examples we have the following nice information:   
`Wire.begin();        // **join i2c bus (address optional for master)**`
so we can assign an address even when the unit acts as a master 

- the sketch I'm working on works fine on UNO, MEGA2560 and Leonardo 
the unit act as a master with the command Wire.begin(XX) 
(address scan works fine too) 

- I think that  the device become a master when the `Wire.beginTransmission(address)` command is performed (not at the `Wire.begin()` point) 
Other words the I2C circuit starts to generate clk signal when the beginTrasmission(address)  is processed
This behavior allows to create multi master bus where all masters have their own address 
Am I wrong ? 

tnx
L.

fpistm, 2019-11-07:
@Luca-007 
I'm not agree, API ref said for `Wire.begin()`:

> address: the 7-bit slave address (optional); if not specified, join the bus as a master.

So the device act as a slave as there is no master defined this could not work.
The comment said only the `Wire.begin` allow to join the bus and if no @ then it join as master.
I guess this is confusing.

Luca-007, 2019-11-07:
yes, confusing.

In my mind the comment says that if there is no @ the unit acts as a master (because without an  @ there is no other way) 
but 
if there is an @ the unit can act as a master **or** as a slave 

Nowhere we have something that says   "if 7-bit address is specified it will act as a slave only"

Indeed using Wire.Begin(@) on Arduino boards i2c_scanner.ino works fine

tnx
L.

fpistm, 2019-11-07:
> Indeed using Wire.Begin(@) on Arduino boards i2c_scanner.ino works fine

What would you mean by works fine?
No error reported? In that case Check the error code value, I guess it is 2 or 3.

Luca-007, 2019-11-07:
> What would you mean by works fine?

... I mean it works fine :-) 
The sketch running on Arduino boards with Wire.begin(@) 
locates the connected addresses in the right way, returning Wire.endTransmission() = 0

It works fine on Arduino, doesn't on STM32 
this is the reason I open this issue 

> No error reported? In that case Check the error code value, I guess it is 2 or 3.
No errors reported. 
Tested with Mega2560, Uno, Leonardo 

thanks
L.

fpistm, 2019-11-07:
I've tested with a DUE and as I expected this not works. 
The DUE reports error 2 which means NACK.
With STM32 Core, the error code returned  in that case is 4 that's why you see an error because it is printed by the sketch only when equal to 4
You declare the device as a slave with no master. SCL is managed by master this could not work.

I close this issue as this works as expected.

closed 2019-11-07 by fpistm