stm32duino/Arduino_Core_STM32 #760: OpenAMP list.h name collision with freertos list.h

adammunich, 2019-11-07:
When compiling with stm32freertos, list.h in OpenAMP's name over-rides list.h with freertos. renaming OpenAMP's list.h solves this issue.

fpistm, 2019-11-07:
Hi @adammunich 
Thanks for reporting this.
Currenlty, I guess I will remove OpenAMP from include list in the platform.txt as it is not used.
/CC @kbumsik

fpistm, 2019-11-07:
@adammunich 
How you met the issue?
I have no issue. I thought OpenAmp include path was in the platform.txt but this is not.

adammunich, 2019-11-07:
Some platforms and makefiles I have seen include Arduino_Core_STM32/system/Middlewares/ST recursively, which causes the issue.

kbumsik, 2019-11-07:
@fpistm Please reconsider deleting. The virtual serial using OpenAMP is completed and it's code cleaning phase. I'm going to make a PR right after #717 is merged.

@adammunich Could I ask what platforms are? I'm personally interested in projects using stm32freertos.
> Arduino_Core_STM32/system/Middlewares/ recursively

This sounds like their responsibility to update the project build though.

fpistm, 2019-11-15:
> This sounds like their responsibility to update the project build configurations though.

I agree with @kbumsik.
I see no issue with `list.h` as OpenAmp requires to include it thanks: `#include <metal/list.h>`

closed 2019-11-15 by fpistm