stm32duino/Arduino_Core_STM32 #761: STM32L031K6 UART pins

crameur, 2019-11-07:


fpistm, 2019-11-07:
Hi @crameur  
How do you wired the 2 boards?

crameur, 2019-11-07:
Same GND, RX of the 1st one to the TX of the second one and conversely

fpistm, 2019-11-07:
You use the 1.7.0 core version ?

crameur, 2019-11-08:
@fpistm Yes, STM32 Cores version 1.7.0

fpistm, 2019-11-08:
I think this could be related to interrupt prio. 
Anyway, the best way should be to use an hardware serial. SoftwareSerial was mainly designed for mcu with one hardware serial and we add it mainly for libraries compatibility.
Like STM32 has more U(S)ART this shoukd superseded this.

crameur, 2019-11-08:
Yeah that's why I tried to implement it using an hardware serial but it seems as if there is only one UART on these stm.. And the Serial for the Arduino monitor is already using it ?
Am I mistaking, which pins should I use then ?

fpistm, 2019-11-08:
I will check when got some time. Keep in touch

fpistm, 2019-11-13:
Hi  @crameur 
Unfortunately, this is not possible to use the SoftwareSerial with the STM32L031K6. 
It spends all its time in the Timer IRQ handler, it is not enough fast for that use case.
About using an hardware Serial. This is not possible on this Nucleo as there is only 2 UART's available:
 * `USART2` used for default Serial
 * `LPUART1` but the pins are not accessible and reserved for SWD stuff. That's why it is commented in the `PeripheralPins.c`.

fpistm, 2019-11-15:
I close it as there is nothing to do.

closed 2019-11-15 by fpistm