stm32duino/Arduino_Core_STM32 #762: Multiple libraries were found

mrguen, 2019-11-08:
Error at compilation:
 "Multiple libraries were found for ..." 
for all the libraries that the sketch is using.

I checked if there was multiple libraries by searching in the explorer. The only duplicate (after failed compilation) is for example:

C:\Users\TG\AppData\Local\Temp\arduino_build_836171\libraries\Adafruit_GFX_Library
and
C:\Users\TG\Documents\Arduino\libraries\Adafruit_GFX_Library

So apparently the compiler is confused by the content of C:\Users\TG\AppData\Local\Temp\arduino_build_836171

Configuration

Windows 7 pro 64 bits
IDE 1.8.10
Arduino core STM32 1.7.0

Board: Generic STM32F1 Series 
Board part number: BluePill F103C8
USART support: enabled (generic serial)
USB support: CDC supersede USART
USB: Low/full speed
Optimize: smallest
C runtime library: Newlib nano (default)
Upload method Maple DFU bootloader 2.0

fpistm, 2019-11-08:
This is Arduino IDE issue. Try to close and reopen.

mrguen, 2019-11-08:
Tried to restart the computer: same. Also is a compilation time from 1 to 3 minutes. Something changed either with IDE 1.8.10 or core 1.7.0 or something else because it was not like that before.

fpistm, 2019-11-08:
I guess it is a malformed library or something like that but I could not help on this.

closed 2019-11-08 by fpistm

mrguen, 2019-11-08:
I also noticed other things:

I tried to clean AppData and reinstall boards:

installing STM32 Cores takes forever
it is HUGE aroung 835 Mb!

fpistm, 2019-11-08:
> * installing STM32 Cores takes forever

Mainly on Windows and it is worst if you have an antivirus.

> * it is HUGE aroung 835 Mb!

ARM GCC toolchain + CMSIS take around 600 Mb.

fpistm, 2019-11-08:
@mrguen 
I've just met this issue working on another stuff.
In fact the message seems told Adafruit_GFX_Library was the problem but in fact it misses one other.
```
Alternatives for Adafruit_miniTFTWing.h: []
ResolveLibrary(Adafruit_miniTFTWing.h)
  -> candidates: []
basic:3:10: error: Adafruit_miniTFTWing.h: No such file or directory
Multiple libraries were found for "SPI.h"
 #include "Adafruit_miniTFTWing.h"
 Used: /local/data/work/IDE/arduino/arduino-1.8.10/portable/packages/STM32/hardware/stm32/1.6.1/libraries/SPI
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Multiple libraries were found for "Adafruit_GFX.h"
 Used: /local/data/work/IDE/arduino/arduino-1.8.10/portable/sketchbook/libraries/Adafruit_GFX_Library
Multiple libraries were found for "Adafruit_ST7735.h"
 Used: /local/data/work/IDE/arduino/arduino-1.8.10/portable/sketchbook/libraries/Adafruit_ST7735_and_ST7789_Library
Using library Adafruit_GFX_Library at version 1.6.1 in folder: /local/data/work/IDE/arduino/arduino-1.8.10/portable/sketchbook/libraries/Adafruit_GFX_Library 
Using library Adafruit_ST7735_and_ST7789_Library at version 1.5.6 in folder: /local/data/work/IDE/arduino/arduino-1.8.10/portable/sketchbook/libraries/Adafruit_ST7735_and_ST7789_Library 
Using library SPI at version 1.0 in folder: /local/data/work/IDE/arduino/arduino-1.8.10/portable/packages/STM32/hardware/stm32/1.6.1/libraries/SPI 
exit status 1
Adafruit_miniTFTWing.h: No such file or directory
```

You should enable all verbose output in preferences.

mrguen, 2019-11-09:
Yes the same issue but 

* raised for all the libraries used in the project

`collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "Adafruit_GFX.h"
 Used: C:\Users\TG\Documents\Arduino\libraries\Adafruit_GFX_Library
Multiple libraries were found for "Adafruit_ST7735.h"
 Used: C:\Users\TG\Documents\Arduino\libraries\Adafruit_ST7735_and_ST7789_Library
Multiple libraries were found for "SPI.h"
 Used: C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\SPI
Multiple libraries were found for "OneButton.h"
 Used: C:\Users\TG\Documents\Arduino\libraries\OneButton-master
Multiple libraries were found for "EEPROM.h"
 Used: C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\EEPROM
Multiple libraries were found for "Si5351A.h"
 Used: C:\Users\TG\Documents\Arduino\libraries\Pandauino-Si5351A
Multiple libraries were found for "Wire.h"
 Used: C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\Wire
Multiple libraries were found for "RotaryEncoder.h"
 Used: C:\Users\TG\Documents\Arduino\libraries\RotaryEncoder-master
exit status 1` 

* happens only when using the maple DFU 2.0 upload method / not when using  STMCubeProgrammer for example

And I am sorry I don't get if you have an answer to the problem?

fpistm, 2019-11-09:
Copy the full log with All verbose output enable in the Arduino preference

fpistm, 2019-11-09:
And **"Multiple libraries were found for ..."** is not an error.
Arduino simply says that it found several library and which one it used.

fpistm, 2019-11-09:
And what is exactly the libraries you use? And where they come from?

mrguen, 2019-11-09:
Ok, but it ends up in error. Only when using the maple DFU uplaod method.

`c:/users/tg/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\TG\AppData\Local\Temp\arduino_build_412032/clock_dev_STM32_v0.2.ino.elf section `.rodata' will not fit in region `FLASH'

c:/users/tg/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 800 bytes

collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "Adafruit_ST7735.h"
 Used: C:\Users\TG\Documents\Arduino\libraries\Adafruit_ST7735_and_ST7789_Library
Multiple libraries were found for "SPI.h"
 Used: C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\SPI
Multiple libraries were found for "OneButton.h"
 Used: C:\Users\TG\Documents\Arduino\libraries\OneButton-master
Multiple libraries were found for "EEPROM.h"
 Used: C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\EEPROM
Multiple libraries were found for "Si5351A.h"
 Used: C:\Users\TG\Documents\Arduino\libraries\Pandauino-Si5351A
Multiple libraries were found for "Wire.h"
 Used: C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\Wire
Multiple libraries were found for "RotaryEncoder.h"
 Used: C:\Users\TG\Documents\Arduino\libraries\RotaryEncoder-master
Multiple libraries were found for "Adafruit_GFX.h"
 Used: C:\Users\TG\Documents\Arduino\libraries\Adafruit_GFX_Library
exit status 1
Error compiling for board Generic STM32F1 series.`

mrguen, 2019-11-09:
Ok, I get it. The assembly is too big when using Madpel DFU method.

fpistm, 2019-11-09:
yes. 
Seriously, please before submit an issue read carefully the log. Or at least provide a full log...
It really a time consuming issue for nothing... I always pleased to help but when it is really needed.

mrguen, 2019-11-09:
Sorry. I am also quite tired with waiting 3 minutes for compilation and so many parameters and upload method that I can't make work that I quite "loose my latin" as we say in french.

mrguen, 2019-11-09:
It would be nice at one point to have a summary of working upload methods witht their configuration and appropriate bootloaders depending on board version etc...

fpistm, 2019-11-09:
> waiting 3 minutes for compilation

On windows this is long, I know but linked to Arduino IDE. On linux for example It took only few seconds to build. You can try to disable your antivirus to speed.

> It would be nice at one point to have a summary of working upload methods witht their configuration and appropriate bootloaders depending on board version etc...

The proposed upload method are linked to the target so if it is  proposed it works. Anyway for those using bootloader, it is mainly for legacy and it's mainly up to end user to properly install the bootloader.
This is documented here:
https://github.com/stm32duino/wiki/wiki/Upload-methods#maple-dfu-bootloader

Unfortunately the legacy wiki is offline. Probably it requires to be documented more.
Feel free to complete, it is a community project. All contribution are welcome.

I'm also tired and providing lot of support which avoid to target some new dev or tasks ( documentations improvements, ACI tests,...). Time is always the missing ingredients.

J'en perd mon latin aussi des fois :wink:

mrguen, 2019-11-10:
As explained here https://forum.arduino.cc/index.php?topic=384321.0 

The bottleneck is Comodo firewall... Now I don't know what Arduino is searching over the Net each time it compiles.

mrguen, 2019-11-11:
Another reason is that the core is completely recompiled. The sketch copilation takes seconds but the core compilation takes forever.

avandalen, 2020-01-06:
I have the same problem 
Always when there is a compiler error in the sketch, I get the "Multiple libraries were found"  error too. 
After the sketch is ok, than the " Multiple libraries"  error disappears.
So it is an Arduino IDE issue. I use Arduino version 1.8.10, older versions don't have this issue.

fpistm, 2020-01-06:
> Multiple libraries were found

is not an error, IDE warns you that there are multiple libraries having the same include entry.
```
Ex for `Wire.h`
Multiple libraries were found for "Wire.h"
 Used: packages/STM32/hardware/stm32/1.8.0/libraries/Wire
...
Using library Wire at version 1.0 in folder: packages/STM32/hardware/stm32/1.6.1/libraries/Wire 
```

avandalen, 2020-01-06:
It is still a difficult problem, I dont know how to get rid of the " multiple libraries" message.
I have a sketch thats running fine, without any error messages after compiling, so without " multiple libraries". I dont see any multiple libraries on my PC.
If I delete just 1 line in the sketch:
// # include "LTC2493.h"
then I get a very long list of errors, 
I would like it to be reported only: LTC2493.h not found
This is the errror list:

test-LTC2493:111:29: error: 'LTC2499_SPEED_1X' was not declared in this scope

 static int16_t two_x_mode = LTC2499_SPEED_1X;       //!< The LTC2499 2X Mode settings

                             ^~~~~~~~~~~~~~~~

test-LTC2493:113:33: error: 'LTC2499_R50_R60' was not declared in this scope

 static uint8_t rejection_mode = LTC2499_R50_R60;    //!< The LTC2499 rejection mode

                                 ^~~~~~~~~~~~~~~

D:\DATA BACKUPPED\Documenten Albert\0 Spotwelder-controller\Vacuum part\Code\Vacuum-regulator-v1\Vacuum-regulator-v1\test-LTC2493.ino:113:33: note: suggested alternative: 'LTC2499_vref'

 static uint8_t rejection_mode = LTC2499_R50_R60;    //!< The LTC2499 rejection mode

                                 ^~~~~~~~~~~~~~~

                                 LTC2499_vref

test-LTC2493:114:30: error: 'LTC2497_I2C_ADDRESS' was not declared in this scope

 static uint8_t i2c_address = LTC2497_I2C_ADDRESS;   //!< I2C address in 7 bit format for part

                              ^~~~~~~~~~~~~~~~~~~

D:\DATA BACKUPPED\Documenten Albert\0 Spotwelder-controller\Vacuum part\Code\Vacuum-regulator-v1\Vacuum-regulator-v1\test-LTC2493.ino:114:30: note: suggested alternative: 'EEPROM_I2C_ADDRESS'

 static uint8_t i2c_address = LTC2497_I2C_ADDRESS;   //!< I2C address in 7 bit format for part

                              ^~~~~~~~~~~~~~~~~~~

                              EEPROM_I2C_ADDRESS

test-LTC2493:120:50: error: 'LTC2499_CH0' was not declared in this scope

 const uint16_t BUILD_COMMAND_SINGLE_ENDED[16] = {LTC2499_CH0, LTC2499_CH1, LTC2499_CH2, LTC2499_CH3,

                                                  ^~~~~~~~~~~

D:\DATA BACKUPPED\Documenten Albert\0 Spotwelder-controller\Vacuum part\Code\Vacuum-regulator-v1\Vacuum-regulator-v1\test-LTC2493.ino:120:50: note: suggested alternative: 'LTC2499_vref'

 const uint16_t BUILD_COMMAND_SINGLE_ENDED[16] = {LTC2499_CH0, LTC2499_CH1, LTC2499_CH2, LTC2499_CH3,

                                                  ^~~~~~~~~~~

                                                  LTC2499_vref

test-LTC2493:120:63: error: 'LTC2499_CH1' was not declared in this scope

 const uint16_t BUILD_COMMAND_SINGLE_ENDED[16] = {LTC2499_CH0, LTC2499_CH1, LTC2499_CH2, LTC2499_CH3,

                                                               ^~~~~~~~~~~

D:\DATA BACKUPPED\Documenten Albert\0 Spotwelder-controller\Vacuum part\Code\Vacuum-regulator-v1\Vacuum-regulator-v1\test-LTC2493.ino:120:63: note: suggested alternative: 'LTC2499_vref'

 const uint16_t BUILD_COMMAND_SINGLE_ENDED[16] = {LTC2499_CH0, LTC2499_CH1, LTC2499_CH2, LTC2499_CH3,

                                                               ^~~~~~~~~~~

                                                               LTC2499_vref

[... 643 lines cut ...]
 Used: D:\DATA
Multiple libraries were found for "Linduino.h"
 Used: D:\DATA
Multiple libraries were found for "LTC24XX_general.h"
 Used: D:\DATA
Multiple libraries were found for "Streaming.h"
 Used: D:\DATA
Multiple libraries were found for "Adafruit_MCP4725.h"
 Used: D:\DATA
Multiple libraries were found for "SPI.h"
 Used: C:\Program
Multiple libraries were found for "USE_WIRE.h"
 Used: D:\DATA
Multiple libraries were found for "avdweb_Switch.h"
 Used: D:\DATA
Multiple libraries were found for "Metro.h"
 Used: D:\DATA
Multiple libraries were found for "Adafruit_ILI9341.h"
 Used: D:\DATA
Multiple libraries were found for "UserInterface.h"
 Used: D:\DATA
Multiple libraries were found for "LT_I2C.h"
 Used: D:\DATA
exit status 1
'LTC2499_SPEED_1X' was not declared in this scope

fpistm, 2020-01-06:
> I would like it to be reported only: LTC2493.h not found

How you want it to know which library is missing?
That is normal you get all those errors like you removed the library inclusion.

It simply told you all unknown type, function,....

Anyway, this should be better you discuss that point on Arduino Forum or GitHub issue.

avandalen, 2020-01-06:
I have found the cause of the false error messages " Multiple libraries were found " .
The preferences.txt was corrupted. After deleting preferences.txt, I no longer received any false error messages "Multiple libraries were found" .

Here is the corrupted preferences.txt file:
`board=uno
boardsmanager.additional.urls=
build.verbose=false
build.warn_data_percentage=75
builtin_platform_is_newer=10809
cache.enable=true
compiler.cache_core=true
compiler.warning_level=none
console=true
console.auto_clear=true
console.error.file=stderr.txt
console.length=500
console.lines=4
console.output.file=stdout.txt
custom_bodmodeactive=atxy7_disabled
custom_bodmodesleep=atxy7_disabled
custom_bodvoltage=atxy7_1v8
custom_chip=atxy7_3217
custom_clock=atxy7_20
custom_cpu=nano_atmega328
custom_dacvref=atxy7_0v55
custom_eesave=atxy7_enable
custom_millis=atxy7_enabled
custom_neopixelport=atxy7_porta
custom_resetpin=atxy7o_reset
custom_spipins=atxy7_standard
custom_twipins=atxy7_standard
custom_uartpins=atxy7_standard
custom_version=a-star328PB_8mhz
editor.antialias=true
editor.auto_close_braces=true
editor.caret.blink=true
editor.code_folding=false
editor.divider.size=2
editor.external=false
editor.font=Monospaced,plain,16
editor.indent=true
editor.invalid=false
editor.keys.alternative_cut_copy_paste=true
editor.keys.home_and_end_beginning_end_of_doc=false
editor.keys.shift_backspace_is_delete=true
editor.languages.current=en
editor.linenumbers=false
editor.save_on_verify=true
editor.tabs.expand=true
editor.tabs.size=2
editor.update_extension=true
editor.window.height.default=600
editor.window.height.min=290
editor.window.width.default=500
editor.window.width.min=400
export.applet.separate_jar_files=false
export.application.fullscreen=false
export.application.platform=true
export.application.stop=true
export.delete_target_folder=true
gui.scale=auto
ide.accessible=false
last.ide.1.8.10.daterun=1578347217
last.ide.1.8.10.hardwarepath=C:\Program Files (x86)\Arduino\hardware
last.ide.1.8.5.daterun=1567543035
last.ide.1.8.5.hardwarepath=C:\Program Files (x86)\Arduino\hardware
last.ide.1.8.9.daterun=1563204653
last.ide.1.8.9.hardwarepath=C:\Program Files (x86)\Arduino\hardware
last.screen.height=1200
last.screen.width=1920
[... 31 lines cut ...]
run.options.memory.maximum=256
run.present.bgcolor=#666666
run.present.exclusive=false
run.present.stop.color=#cccccc
serial.databits=8
serial.debug_rate=115200
serial.line_ending=1
serial.parity=N
serial.port=COM3
serial.port.file=COM3
serial.port.iserial=null
serial.show_timestamp=false
serial.stopbits=1
sketchbook.path=D:\DATA BACKUPPED\Documenten Albert\Arduino\Sketchbook\Arduino
software=ARDUINO
target_package=arduino
target_platform=avr
theme.file=
update.check=false
update.id=5025037511160597429
update.last=1570269646623
upload.using=bootloader
upload.verbose=false
upload.verify=false
`