stm32duino/Arduino_Core_STM32 #765: Serial.println stuck in the middle of a printing.

amirc2013, 2019-11-08:
The bug :
The STM got stuck in the middle of printing using Serial.println.
why ?

I just cant find a good reason it will happen

it literally stated to print a sentence and stop in the middle of it.

My board : Nucleo L4R5ZI
OS: Windows
Arduino IDE version: 1.8.10
STM32 core version: How can I check that ?
Tools menu settings if not the default. Newlib Nano (default)
Upload method: Mass storage

pkourany, 2019-11-08:
@amirc2013 can you give a little more context or show your code before and after the Serial.println() statement?

amirc2013, 2019-11-08:
`  fwu->partitions = 0;`
`  for (offset = 0; offset < V7_PARTITION_SUPPORT_BYTES; offset++) {`
`    for (ii = 0; ii < 8; ii++) {`
`      if (query_1_7.data[index + offset] & (1 << ii))`
`        fwu->partitions++;`
`    }`

`    Serial.print(__func__);`
`    Serial.print(": Supported partitions: 0x");`
`    Serial.println(query_1_7.data[index + offset]);`
`  }`

thie is the code ... but im sure its not about it ... If I add a printing somewhere it will stuck in a different position.

this is the message i receive :

[image]

fpistm, 2019-11-08:
@amirc2013, as @pkourany said, please provide more information.
Is the UART Serial or the USB VCP?
Which speed you used?
Provide the full sketch as it seems you made several stuff.

amirc2013, 2019-11-08:
I cant send the sketch .. its too long and wont give much information.
about the other thing u asked me to - I use the arduino serial monitor to see the printing VIA the USB.
i tried different speeds .. none of them helped.
now im on 9600

amirc2013, 2019-11-08:
watch this :

[image] 

if i change the strings, it will stuck in different place. there is a problem with the implementation ..

[image]

[image]

help ?

fpistm, 2019-11-08:
This is not my question. You use the Serial over STlink or the VCP thanks the User USB?
I've made a quick test (see sketch hereafter) and I've no issue (using Serial through STLink or user USB).
My guess is your sketch have an issue and it raises an exception.

```C++
char *lorem[] = {
  "What is Lorem Ipsum?",
  "Lorem Ipsum is simply dummy text of the printing and typesetting industry. Lorem Ipsum has been the industry's standard dummy text ever since the 1500s, when an unknown printer took a galley of type and scrambled it to make a type specimen book. It has survived not only five centuries, but also the leap into electronic typesetting, remaining essentially unchanged. It was popularised in the 1960s with the release of Letraset sheets containing Lorem Ipsum passages, and more recently with desktop publishing software like Aldus PageMaker including versions of Lorem Ipsum.",
  "Where does it come from?",
  "Contrary to popular belief, Lorem Ipsum is not simply random text. It has roots in a piece of classical Latin literature from 45 BC, making it over 2000 years old. Richard McClintock, a Latin professor at Hampden-Sydney College in Virginia, looked up one of the more obscure Latin words, consectetur, from a Lorem Ipsum passage, and going through the cites of the word in classical literature, discovered the undoubtable source. Lorem Ipsum comes from sections 1.10.32 and 1.10.33 of \"de Finibus Bonorum et Malorum\" (The Extremes of Good and Evil) by Cicero, written in 45 BC. This book is a treatise on the theory of ethics, very popular during the Renaissance. The first line of Lorem Ipsum, \"Lorem ipsum dolor sit amet..\", comes from a line in section 1.10.32.",
  "The standard chunk of Lorem Ipsum used since the 1500s is reproduced below for those interested. Sections 1.10.32 and 1.10.33 from \"de Finibus Bonorum et Malorum\" by Cicero are also reproduced in their exact original form, accompanied by English versions from the 1914 translation by H. Rackham.",
  "Why do we use it?",
  "It is a long established fact that a reader will be distracted by the readable content of a page when looking at its layout. The point of using Lorem Ipsum is that it has a more-or-less normal distribution of letters, as opposed to using 'Content here, content here', making it look like readable English. Many desktop publishing packages and web page editors now use Lorem Ipsum as their default model text, and a search for 'lorem ipsum' will uncover many web sites still in their infancy. Various versions have evolved over the years, sometimes by accident, sometimes on purpose (injected humour and the like).",
};

// the setup routine runs once when you press reset:
void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
}

// the loop routine runs over and over again forever:
void loop() {
  static uint32_t i = 0;
  Serial.print(__func__); Serial.print(" called "); Serial.println(i++);
  for (uint8_t l = 0; l < 7; l++) {
    Serial.println(lorem[l]);
  }
  Serial.println("####################################");
  delay(1);        // delay in between reads for stability
}
```

amirc2013, 2019-11-08:
Ok, which exception can be raised because of me ?
What can I do wrong in the sketch that can make the Serial.print fails in the middle of his own run ?
I dont use threads or something parallel to that code ...
Please help me with that because I cant find any answer to this wired bug

amirc2013, 2019-11-08:
Oh and by the way, a really important fact - this bug can happen and disappear without any code changes ...
It can be also happen or disappear after changing the C runtime library. from the new one to the default one...

fpistm, 2019-11-08:
Buffer overflow, wrong pointer, index,...
I could not know what you do exactly nor debug your code.

fpistm, 2019-11-08:
The example I provided to test the `Serial.println` runs more than 2000 iterations without any hangs.
So I close this issue, you should try to debug your sketch (ex using PIO)

closed 2019-11-08 by fpistm

uzi18, 2019-11-08:
can't help if you don't want to provide code to reproduce problem

amirc2013, 2019-11-09:
I can give the sketch, no problem. Just wonder if it will help you...
[Sketch.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3827444/Sketch.zip)

fpistm, 2019-11-09:
At a first quick review, this is not correct:
```
  Wire.begin();
  Wire.setSCL(2);           // Setting the SCL to be D14 (pin 2 according to the datasheet) on CN7 pin-group
  Wire.setSDA(4);           // Setting the SCL to be D15 (pin 4 according to the datasheet) on CN7 pin-group
```

The `setSCL/SDA` have to be called before `begin()`

amirc2013, 2019-11-09:
I'll fix it right away. thanks

amirc2013, 2019-11-09:
btw, it seems that those line messing things up ... idk why so i removed them for now :S

amirc2013, 2019-11-12:
@fpistm 
I debbuged the code. it seems that malloc is getting hard_fault ?
I jump to WWDG in the middle of it. why >?

fpistm, 2019-11-12:
> @fpistm
> I debbuged the code. it seems that malloc is getting hard_fault ?
> I jump to WWDG in the middle of it. why >?

Probably not enough space.

amirc2013, 2019-11-12:
@fpistm 
no its not the problem. I have a huge amount of memory ... 
It crashes at this line everytime 
[image]