stm32duino/Arduino_Core_STM32 #769: Compile error with boolean constants when porting a library to stm32duino

cdrose, 2019-11-11:
I'm not sure if this counts as a bug in stm32duino as such, but I am not sure where else to raise this question. I am attempting to adapt the [kissStepper](https://github.com/risitt/kissStepper) library to run on stm32duino, however I have run into an issue. 

The code in kissStepper which won't compile seems pretty uninteresting, theres a bool class member and its being initialised to 'false' in an initialisation list, however it fails with the error:
`In constructor 'kissStepperNoAccel::kissStepperNoAccel(uint8_t, uint8_t, uint8_t)':
kissStepper.cpp:56:17: error: cannot convert 'volatile uint32_t*' {aka 'volatile long unsigned int*'} to 'volatile uint8_t* const' {aka 'volatile unsigned char* const'} in initialization
     m_init(false)`

From my reading of that error message the type of 'false' is volatile uint32_t* which is probably sensible, but the class member is bool which apparently is of type volatile uint8_t* const.

I have tried to replicate this in a minimal example but my simple test compiles fine, and I cant work out what the difference is in the library code. Googling the error brings up references to other libraries being ported from AVR to 32bit platforms like ESP32 but I have spotted any hints on how to actually fix the problem.

Any suggestions on how to debug this (or alternatively where I should post this question if this is the wrong place).

fpistm, 2019-11-11:
Hi @cdrose 
This is a third party issue. You should ask to the library maintainer.
Anyway, this kind of error is really confusing. In fact it does not rely to the bool.
Error comes from this line:
https://github.com/risitt/kissStepper/blob/62136cd1cca3a092ecdfd83c420b58bbd14226e8/src/kissStepper.cpp#L50
this is a volatile uint8_t while the `portOutputRegister` return an `uint32_t `one.
https://github.com/risitt/kissStepper/blob/62136cd1cca3a092ecdfd83c420b58bbd14226e8/src/kissStepper.h#L170
so change the classe member to `uint32_t `

closed 2019-11-11 by fpistm

cdrose, 2019-11-11:
That was a very misleading compiler message, your explanation makes much more sense. Thanks so much for the help Frederic, I really appreciate it!