stm32duino/Arduino_Core_STM32 #770: Not using previously compiled files and slow compilation.

mrguen, 2019-11-11:
1) When re-compiling a sketch without changing the configuration or anything, the compiler does recompile many files instead of using previously compiled files. 

Tried for "BluePill F103C8 (128k)"

~195 files were re-used
~193 files were re-compiled

Files that where supposed to be re-used are anyway re-compiled

Example

Using previously compiled file: C:\Users\TG\AppData\Local\Temp\arduino_build_531845\core\stm32\HAL\stm32yyxx_hal_i2s_ex.c.o

Then
C:\\Users\\TG\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\arm-none-eabi-gcc\\8.2.1-1.7/bin/arm-none-eabi-gcc-ar rcs "C:\\Users\\TG\\AppData\\Local\\Temp\\arduino_build_531845\\core\\core.a" "C:\\Users\\TG\\AppData\\Local\\Temp\\arduino_build_531845\\core\\stm32\\HAL\\stm32yyxx_hal_i2s_ex.c.o"

The log is attached
[Blink_verbose_compilation.txt](https://github.com/stm32duino/Arduino_Core_STM32/files/3831585/Blink_verbose_compilation.txt)

2) Each file compilation is extremely slow from a couple of seconds to tens of seconds. I have excluded Arduino files from AVG antivirus and authorized Arduino through the Comodo firewall. The CPU works at 1-30% load

fpistm, 2019-11-11:
Well it seems you do not understand how Arduino handles the build.
Firstly the object (.o) is not recompiled and reused but the second line is the step to create an archive this is not a compilation.
If you have any issue with how Arduino handles the build then open an issue to Arduino GitHub.
As I said in an other issue opened by you the slow compilation is not linked to the core but to Arduino under Windows.

closed 2019-11-11 by fpistm

mrguen, 2019-11-11:
This was not my commentary, but the logs.
The files are reused but also re-compiled.

mrguen, 2019-11-11:
Ok, thanks for the input. Well, at the difference of a compilation for let say a Nano, your process does archive files even when they were already archived. This writing over is the most time consuming.

fpistm, 2019-11-11:
> your process

Not mine process... Arduino process

mrguen, 2019-11-11:
Well, I don't know. If I compile for Aduino it does not archive amready archived files (and so re-used)

mrguen, 2019-11-11:
Posted https://forum.arduino.cc/index.php?topic=82527.new#new

mrguen, 2019-11-11:
The issue happens only with the third party boards I could test: STM32 and MightyCore. The behaviour is regular for 5 official boards I tested including cortex m0 and m3.

mrguen, 2019-11-25:
In Arduino 1.8.9: the option "Aggressively cache compiled core" in the Arduino IDE/settings had to be set. 

Now it is a default behaviour in Arduino 1.8.10 and working as expected after a clean install.

Bambofy, 2020-01-13:
Hi, this problem still persists for me using Arduino version 1.8.10. It recompiles the STM32 src at every build. See this build log:
[stm32l452re successful build.txt](https://github.com/stm32duino/Arduino_Core_STM32/files/4054413/stm32l452re.successful.build.txt)

fpistm, 2020-01-24:
As I said several time seems to be confirmed, latest Arduino IDE bring some regression:
https://github.com/arduino-libraries/ArduinoBLE/issues/54#issuecomment-578033856

BCsabaEngine, 2021-06-23:
I have same feeling. My stm32 project compiles for a minute, same proj is 12sec for arduino nano. I have wrote a fast compile batch file to compile it to nano, i check syntax (need because cpp inheritance :)) check warnings and file structure.
After compile (once a minute approx) in every halh hour i upload it to stm32 normally.

Here is my batch: https://raw.githubusercontent.com/BCsabaEngine/SegmentController/develop/Firmware/_build.Master.fastdev.bat

Avamander, 2025-05-24:
I find that this is still an issue. For example if the previous build failed then even already cached core/HAL objects are not used, I've not seen this behaviour with other build systems. Building these objects also seems entirely sequential. Which is just incredibly slow on any remotely decent hardware, glacially slow if on good hardware. Not only that, a build without any changes takes ~15 seconds. This seems excessively long?