stm32duino/Arduino_Core_STM32 #772: Mixing analog output and PWM output at DAC-Pins

michsens, 2019-11-13:
Hello,

Enable the DAC unit disables the PWM output for all DAC pins. How can I use some pins as analog output and some as PWM output? 

In wiring_analog.c I found that analogWrite strictly overrides the timer (`HAL_DAC_MODULE_ENABLED` prior `HAL_TIM_MODULE_ENABLED`).

I tried a workaround, that does not touch the board definitions. 

I redifined the PinMap_DAC in the ardunio sketch, means disabling of the unnecessary pins:
```
#ifdef HAL_DAC_MODULE_ENABLED
extern const PinMap PinMap_DAC[] = {
  {PA_4,  DAC1, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 1, 0)}, // DAC1_OUT1
  //{PA_5,  DAC1, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 2, 0)}, // DAC1_OUT2
  //{PA_6,  DAC2, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 1, 0)}, // DAC2_OUT1
  {NC,    NP,    0}
};
#endif
```
I also tried to add a  function `analogWritePWM` in the wiring_analog.c with exact the same content as `analogWrite` except the DAC part:
```
void analogWritePWM(uint32_t ulPin, uint32_t ulValue) {
#if defined(HAL_DAC_MODULE_ENABLED) || defined(HAL_TIM_MODULE_ENABLED)
  uint8_t do_init = 0;
#endif
  PinName p = digitalPinToPinName(ulPin);
  if (p != NC) {
#ifdef HAL_TIM_MODULE_ENABLED
    if (pin_in_pinmap(p, PinMap_PWM)) {
      if (is_pin_configured(p, g_anOutputPinConfigured) == false) {
        do_init = 1;
        set_pin_configured(p, g_anOutputPinConfigured);
      }
      ulValue = mapResolution(ulValue, _writeResolution, PWM_RESOLUTION);
      pwm_start(p, _writeFreq * PWM_MAX_DUTY_CYCLE,
                PWM_MAX_DUTY_CYCLE,
                ulValue, do_init);
    } else
#endif /* HAL_TIM_MODULE_ENABLED */
    {
      //DIGITAL PIN ONLY
      // Defaults to digital write
      pinMode(ulPin, OUTPUT);
      ulValue = mapResolution(ulValue, _writeResolution, 8);
      if (ulValue < 128) {
        digitalWrite(ulPin, LOW);
      } else {
        digitalWrite(ulPin, HIGH);
      }
    }
  }
}
```
The ladder one did not end in success.

Do you see a more flexible solution then just disabling the DAC pin?

Thanks.

fpistm, 2019-11-13:
Hi @michsens 
Redefined the PinMap_DAC at the Ardunio sketch works.
Could you tell me more about how you test?
First, which board, full sketch you used?

Other way is to use the HardwareTimer to generate a PWM.
https://github.com/stm32duino/STM32Examples/blob/master/examples/Peripherals/HardwareTimer/All-in-one_setPWM/All-in-one_setPWM.ino

fpistm, 2019-11-13:
It seems you leave `extern` in your sketch before PinMap array which I guess  is ignored.

Here the full Fading sketch tested on a Nucleo G474RE which works fine.
```C++
const PinMap PinMap_DAC[] = {
  // {PA_4,  DAC1, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 1, 0)}, // DAC1_OUT1
  {PA_5,  DAC1, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 2, 0)}, // DAC1_OUT2
  {PA_6,  DAC2, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 1, 0)}, // DAC2_OUT1
  {NC,    NP,    0}
};

int ledPin = PA4;    // LED connected to digital pin 9

void setup() {
  // nothing happens in setup
}

void loop() {
  // fade in from min to max in increments of 5 points:
  for (int fadeValue = 0 ; fadeValue <= 255; fadeValue += 5) {
    // sets the value (range from 0 to 255):
    analogWrite(ledPin, fadeValue);
    // wait for 30 milliseconds to see the dimming effect
    delay(30);
  }

  // fade out from max to min in increments of 5 points:
  for (int fadeValue = 255 ; fadeValue >= 0; fadeValue -= 5) {
    // sets the value (range from 0 to 255):
    analogWrite(ledPin, fadeValue);
    // wait for 30 milliseconds to see the dimming effect
    delay(30);
  }
}
```

michsens, 2019-11-13:
The board is the one we discussed earlier. 
You modified the board definitions in a way I finally could run the board:
[https://github.com/fpistm/Arduino_Core_STM32/tree/Example_STM32F373R8](url)

I did following:
In the 'variant.h' I enabled `HAL_DAC_MODULE_ENABLED` and `HAL_TIM_MODULE_ENABLED`. With this DAC1 with channel 1 and 2 as well as DAC2 with channel 1 are enabled.
I then wrote a short sketch to test the DAC funcionallity with:
```
#define DA_PWM_1 PA4
#define DA_PWM_2 PA5

// the setup function runs once when you press reset or power the board
void setup() {
  pinMode(DA_PWM_1, OUTPUT);
  pinMode(DA_PWM_2, OUTPUT);
  analogWriteResolution(12);
  analogWrite(DA_PWM_1, 0);
  analogWrite(DA_PWM_2, 0);
}

// the loop function runs over and over again forever
void loop() {
  analogWrite(DA_PWM_1, 1000);
  analogWrite(DA_PWM_2, 2000);
  delay(1000);
}
```

With an oscilloscope I see a perfect strait line according to the wirtten values.

Disabling `HAL_DAC_MODULE_ENABLED` assures that the above code generates a PWM signal at the outputs - also verified with the oscilloscope.

To only enable the  first channel I used the above redefinition of the PinMap_DAC.
With the oscilloscope I than see a real analog signal at `DA_PWM_1` and a PWM at channel `DA_PWM_2`.

michsens, 2019-11-13:
I found another oportunity by including the lines
```
      dac_stop(p);
      reset_pin_configured(p, g_anOutputPinConfigured);
```
into the `analogWritePWM` mentioned above.

The full function looks like:
```
void analogWritePWM(uint32_t ulPin, uint32_t ulValue) {
#if defined(HAL_TIM_MODULE_ENABLED)
  uint8_t do_init = 0;
#endif
  PinName p = digitalPinToPinName(ulPin);
  if (p != NC) {
#ifdef HAL_TIM_MODULE_ENABLED
    if (pin_in_pinmap(p, PinMap_PWM)) {
      dac_stop(p);
      reset_pin_configured(p, g_anOutputPinConfigured);
      if (is_pin_configured(p, g_anOutputPinConfigured) == false) {
        do_init = 1;
        set_pin_configured(p, g_anOutputPinConfigured);
      }
      ulValue = mapResolution(ulValue, _writeResolution, PWM_RESOLUTION);
      pwm_start(p, _writeFreq * PWM_MAX_DUTY_CYCLE,
                PWM_MAX_DUTY_CYCLE,
                ulValue, do_init);
    } else
#endif /* HAL_TIM_MODULE_ENABLED */
    {
      //DIGITAL PIN ONLY
      // Defaults to digital write
      pinMode(ulPin, OUTPUT);
      ulValue = mapResolution(ulValue, _writeResolution, 8);
      if (ulValue < 128) {
        digitalWrite(ulPin, LOW);
      } else {
        digitalWrite(ulPin, HIGH);
      }
    }
  }
}
```

I placed it in the 'wiring_analog.c' since there are all dependencies fullfilled.

This approach also works.

michsens, 2019-11-13:
Thanks for your answers fpistm.

> It seems you leave extern in your sketch before PinMap array which I guess is ignored.

In my case It did not change anything when I did not use `extern`. Only by `extern` it made a difference.

michsens, 2019-11-13:
If I am correct `pwm_stop()` works with the real timer module. Is this correct?

fpistm, 2019-11-13:
I really don't understand your point and I will not debug your code.
If you read carefully the original code this is not possible to use the DAC if it is not in the PinMap array.
`    if (pin_in_pinmap(p, PinMap_DAC)) {`
That's all.

michsens, 2019-11-13:
This, I am fully aware of. The only thing is, that I can not change this pinmap in the runtime or I just did not find a good way.

michsens, 2019-11-13:
Let's asume following.
A digital input decides wether I need a pwm or an analog output.
How can I change between the both output signals depending on this digital input?

michsens, 2019-11-13:
The original code also shows that it is not possible to use the pwm when `HAL_DAC_MODULE_ENABLED` is set, other then disable the corresponding pin in the `PinMap_DAC`.

fpistm, 2019-11-13:
This in not supported at runtime.
If you want sometime made DAC use the `analogWrite()` else to made PWM use HardwareTimer as I said before.

michsens, 2019-11-13:
What about a little modification of 'wiring_analog.h' and 'wiring_analog.c'?

Be free to use it.

[wiring_analog.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3841820/wiring_analog.zip)

fpistm, 2019-11-13:
Thanks.
Anyway, I would not want change it.
You can do your own `analogWritePWM` using the HardwareTimer.

closed 2019-11-13 by fpistm

michsens, 2019-11-14:
It's fine. Thanks anyway.