stm32duino/Arduino_Core_STM32 #776: STM32 mentioned in build error of ATMEGA2560 - (how is it possible ?)

AndKe, 2019-11-15:
### Description

I have been working on a STM32Core project, then switched to Marlin (a project) for Atmega2560  , on Arduino 1.8.10 - I get a build error, but it somehow references STM32 ?

I did shutdown Arduino IDE, delete /tmp/arduino*  and restarted Arduino IDE - it still remembers the ATMEGA2560 selection, but still fails to compile, Marlin 2.0, mentioning HAL_STM32 - which should not be involved, at all.

```

/home/andre/Arduino/arduino-1.8.10/arduino-builder -dump-prefs -logger=machine -hardware /home/andre/Arduino/arduino-1.8.10/hardware -hardware /home/andre/.arduino15/packages -hardware /home/andre/Arduino/projects/hardware -tools /home/andre/Arduino/arduino-1.8.10/tools-builder -tools /home/andre/Arduino/arduino-1.8.10/hardware/tools/avr -tools /home/andre/.arduino15/packages -built-in-libraries /home/andre/Arduino/arduino-1.8.10/libraries -libraries /home/andre/Arduino/projects/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10810 -build-path /tmp/arduino -warnings=default -build-cache /tmp/arduino_cache_177968 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/andre/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/andre/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=/home/andre/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/home/andre/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.arduinoOTA.path=/home/andre/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/andre/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /home/andre/Arduino/projects/Marlin-bugfix-2.0.x-new/Marlin/Marlin.ino
......................
I/home/andre/Arduino/projects/libraries/U8glib/src /tmp/arduino/sketch/src/HAL/HAL_STM32/SoftwareSerial.cpp -o /dev/null
Alternatives for timer.h: []
ResolveLibrary(timer.h)
  -> candidates: []
/tmp/arduino/sketch/src/HAL/HAL_STM32/SoftwareSerial.cpp:38:10: fatal error: timer.h: No such file or directory
Multiple libraries were found for "U8glib.h"
 #include <timer.h>
 Used: /home/andre/Arduino/projects/libraries/U8glib
          ^~~~~~~~~
compilation terminated.
Using library U8glib at version 1.19.1 in folder: /home/andre/Arduino/projects/libraries/U8glib 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

```

fpistm, 2019-11-16:
HAL_STM32 directory is linked to Marlin not this core.
You should  ask support to Marlin.

closed 2019-11-16 by fpistm

AndKe, 2019-11-16:
true, found out, sorry.