stm32duino/Arduino_Core_STM32 #778: Please clearify UART buffer size options.

AndKe, 2019-11-18:
https://github.com/stm32duino/wiki/wiki/Custom-definitions#serial-rxtx-buffer-size  describes the option to change serial buffer size.
The device is a STM32F103C8

**Describe the solution you'd like**
I would like to know whatever it is some fixed buffer size options within the uart, or a DMA assigned buffer that can be any value, or just numbers like 1024/2048/4096 bytes ?
is the upper limit the RAM ?
-can it be asymmetric ? (what about 64byte TX buffer, and a 3KB RX buffer ?)

My application requires recieving a 56kbit serial stream which is fine, but then I need to send a 19.2Kbit serial stream with special spacing 1ms spacing between bytes.
sending the 1ms spaced packets is slow, and unless I have a significant buffer, the less critical 56k stream will not be processed. 

**Describe alternatives you've considered**
I did consider RTOS/ChibiOS, but then again, simpler is better, the sending of packets is short enough, that a 2-3kB RX buffer would easily solve the issue.

fpistm, 2019-11-18:
> I would like to know whatever it is some fixed buffer size options within the uart, or a DMA assigned buffer that can be any value, or just numbers like 1024/2048/4096 bytes ?

It is described in the wiki. The default size is 64 for both.
https://github.com/stm32duino/Arduino_Core_STM32/blob/42efbeb48728748965ce50df1a71e0db259d37da/cores/arduino/HardwareSerial.h#L32-L58

> is the upper limit the RAM ?
Yes but I do not advise to use all the RAM. 

> -can it be asymmetric ? (what about 64byte TX buffer, and a 3KB RX buffer ?)
Yes it should be but try to "power 2" aligned.

AndKe, 2019-11-18:
Thank you.
one last thing: what is the proper name to set buffer for "Serial2"  (or any other UART)  - will it be SERIAL2_TX_BUFFER_SIZE  ?

fpistm, 2019-11-18:
This is the same name for all Serial instance

AndKe, 2019-11-18:
yes, that's what I feared.
so , should I specify 1kB buffer, it will be 1kB for each of the three UART's I defined?  
 (I am using SerilaUSB , Serial1,Serial2) ? 
- that would be an expensive buffer.

fpistm, 2019-11-18:
SerialUSB  is not linked to HardwareSerial so does not depends on this definition.
So if you set 1KB for Rx and TX definition and instantiate Serial1 and Serial2 then you will consume 4KB.
This is you customization.

closed 2019-11-18 by fpistm

AndKe, 2019-11-18:
Are there technical reasons that prohibit buffer size per UART ? (Like increase only one serial ports RX buffer?)

fpistm, 2019-11-18:
No. This is inherit from Arduino.
Even in the new ArduinoCoreAPI:
https://github.com/arduino/ArduinoCore-API/blob/398e70f188e2b861c10d9ffe5e2bfcb6a4a4f489/api/RingBuffer.h#L30

As you can see we already extend definition to be able to define a size per transmission type and also redefine them.