stm32duino/Arduino_Core_STM32 #779: STM32 serial ports

MCUdude, 2019-11-18:
Hi!
I want to design a custom board around the STM32F401, and on this particular board I need all three serial ports the F401 has to offer. At the moment I'm using a Nucleo F401RE to test my code. However, I've ran into some issues.

Serial and Serial2 is exactly the same and is the serial port that's connected to the on-board stlink programmer. Serial1 does not exist, and how about the last serial port?

How can I get complete access to each serial port?
What physical pins can they be routed to?

fpistm, 2019-11-18:
Hi @MCUdude 

> Serial and Serial2 is exactly the same and is the serial port that's connected to the on-board stlink programmer. Serial1 does not exist, and how about the last serial port?

This is normal that `Serial` is the same than `Serial2`. `Serial` is the generic name for Arduino and then it is mapped to the default `Serialx` instance. In case of Nucleo F401 this is the `USART2` connected to STLink.
For other Serial up to you to define them.

https://github.com/stm32duino/wiki/wiki/API#hardwareserial

Ex:
```
HardwareSerial Serial1(PA10, PA9);
HardwareSerial Serial6(PC7, PC6);
```

Note there is other pins possibilities:
https://github.com/stm32duino/Arduino_Core_STM32/blob/42efbeb48728748965ce50df1a71e0db259d37da/variants/NUCLEO_F401RE/PeripheralPins.c#L139-L159

MCUdude, 2019-11-18:
Great, that's all I need to know! I'll re-open if I stumble over some issues.

closed 2019-11-18 by MCUdude