stm32duino/Arduino_Core_STM32 #789: Frequency measurement is not accurate

jmchiappa, 2019-11-23:
Hello,

I had  a try of the frequency measurement using the provided HardwareTimer example.
I tested it on NUCLEO L476 with 1.7.0.

**Procedure**

1. open InputCapture.ino sketch

2. add just after the `Serial.begin(115200);` a frequency generator:

```
void setup()
{
  Serial.begin(115200);
  tone(D15,38000);
   // Automatically retrieve TIM instance and channel associated to pin
  // This is used to be compatible with all STM32 series automatically.
...
```
This will generate an cute accurate 38kHz, confirmed on my logic analyzer:
nb of periods: 76555
f average: 38.17KHz
duty cycle: 53%

So I assume that the frequency is quite stable (event if I can't measure the std-dev, frequency is quite homogeneous).

3. wire pin D15 to D2. I should then measure frequency close to 38kHz.

**Results**

4. Here is the (part of) measured frequency:
```
Frequency = 38022
Frequency = 36546
Frequency = 36297
Frequency = 36297
Frequency = 36297
Frequency = 38022
Frequency = 38022
Frequency = 36781
Frequency = 36297
Frequency = 36297
Frequency = 36663
Frequency = 38022
Frequency = 37611
Frequency = 36297
Frequency = 36297
Frequency = 36297
...
```
Note that the frequency is not accurate. Following picture is the distribution (over 720 samples):
[image]

**Questions**

How do you explain this huge difference ? The error measurement is about 10% (2KHz @ 38KHz).
Could the generated frequency impact the frequency measurement ?

Regards
Jm

fpistm, 2019-11-23:
@jmchiappa 
I will let @ABOSTM comments on this.

ABOSTM, 2019-11-25:
Hi @jmchiappa ,
I don't see obvious reason for this 2kHz difference,
I don't think generated frequency impact the frequency measured,
but any interruption can impact tone generation and thus your signal generator. For example your serial monitor which is configured at 115200, or hardwaretimer configured for frequency measurement.
When you verified frequency generation with your logic analyzer, does your sketch already include serial monitor and frequency measurement ?

jmchiappa, 2019-11-25:
Hello Alexandre, Yes it does.
But you're right. So I will separate the tone generator from the frequency measurement by using another board.
BTW, I hope that using the serial monitor doesn't affect the frequency measurement, otherwise this feature is  just unusuable.
Please did you reproduce this deviation ?

ABOSTM, 2019-11-25:
@jmchiappa,
Yes I reproduce your issue.
But I also try with an external signal generator 
(keep the same binary, just remove wire between D15 and D2 and apply ext generator), 
and the measurement is quite stable:
```
Frequency = 37700
Frequency = 37682
Frequency = 37682
Frequency = 37682
Frequency = 37700
Frequency = 37682
Frequency = 37700
Frequency = 37682
Frequency = 37682
Frequency = 37700
Frequency = 37664
Frequency = 37700
Frequency = 37700
Frequency = 37682
Frequency = 37664
Frequency = 37682
Frequency = 37700
Frequency = 37700
Frequency = 37700
Frequency = 37718
Frequency = 37700
```

I also replace tone generation by HardwareTimer signal generation (see example All-in-one_setPWM.ino):
replace tone (...); by
```
  #define pin_generator  D15
  TIM_TypeDef *Instance_generator = (TIM_TypeDef *)pinmap_peripheral(digitalPinToPinName(pin_generator), PinMap_PWM);
  uint32_t channel_generator = STM_PIN_CHANNEL(pinmap_function(digitalPinToPinName(pin_generator), PinMap_PWM));
  HardwareTimer *MyTim_generator = new HardwareTimer(Instance_generator);
  MyTim_generator->setPWM(channel_generator, pin_generator, 38000, 50); // 38000 Hertz, 50% dutycycle
```
Result: measurement is very stable.
My conclusion is that tone generation is not very stable has it is done by software and it is sensible to interrupts.

```
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
Frequency = 38004
```

jmchiappa, 2019-11-25:
@ABOSTM ,

I tested by writing D15 with analogWrite(D15,128); PWM @ 50% and wired to D2.
It sounds working well. Frequency is quite stable **BUT** be aware that printed traces don't proved anything as, once you remove the wire, you're still displaying the same frequency (so the latest one) ! It's due to there's no more call to the callback and MeasuredFrequency variable is never updated. The consequence is that you display a wrong frequency. So you think that everything is ok, while there's no signal anymore.
Anyway, you convince me and this example is a good base for a wrap up class.

Thank you.

closed 2019-11-25 by jmchiappa

ABOSTM, 2019-11-27:
@jmchiappa ,
I fixed the issue you highlighted in PR 
https://github.com/stm32duino/STM32Examples/pull/13