stm32duino/Arduino_Core_STM32 #790: `HardwareTimer` PWM resolution stuck at 12-bit?

jeffstaley, 2019-11-23:
If I run a one-line Arduino program on a STM32F7 of 

```
analogWrite(FAN1_PIN, 255)
```

I get a PWM signal with a very low duty cycle instead of the expect 100%. 

Digging into the code I see `pwm_start()` setting up a HardwareTimer with a __hard coded__ 12-bit resoultion:

```
  HT->setCaptureCompare(channel, value, RESOLUTION_12B_COMPARE_FORMAT);

```

fpistm, 2019-11-24:
FAN1_PIN is equal to?

jeffstaley, 2019-11-24:
Any of the PinMap_PWM pins.  But in this case 30.

fpistm, 2019-11-24:
The default PWM resolution is 12 bits. 
So you should do:
`analogWrite(FAN1_PIN, 1023);`
To get a 100% duty cycle

jeffstaley, 2019-11-24:
But if

```
#define PWM_RESOLUTION              12
```

is redefined in variant.h to

```
#define PWM_RESOLUTION              8
```

`analogWrite()` is still in 12-bit mode.

fpistm, 2019-11-24:
The 8 bits value is converted to 12 bits thanks the map function.
Which board you used?

jeffstaley, 2019-11-24:
I'm not seeing that.  If `PWM_RESOLUTION` is set to 8 the map function does nothing (since we're in 8-bit mode).  But the HWTimer is still _hard coded_ to 12-bit mode.

It appears that the map function is used when you dynamically change the resoultion via `analogWriteResolution()`

Board is `RemRam v1`

fpistm, 2019-11-25:
@jeffstaley 
You're right there is an issue.
There is a misalignment when the `PWM_RESOLUTION` is redefined.
This was introduced when HardwareTimer was released. We will fix this.
As a workaround simply comment the `PWM_RESOLUTION` in the variant.h.

jeffstaley, 2019-11-25:
Commenting out `PWM_RESOLUTION` in variant.h doesn't work for me since all the code I'm using (Marlin) assumes 8-bit resolution.  Presumably I could do that and then dynamically change the resolution to 8-bit with `analogWriteResolution()` but I haven't tested that.  Instead I pass in `RESOLUTION_8B_COMPARE_FORMAT` to the HardwareTimer module when compiled in 8-bit PWM mode.  That my not be a general enough solution for you if accept additional resolutions.

fpistm, 2019-11-25:
By default `analogWriteResolution()` is 8 bits. So the range you can used is 0-255. At a lower layer it is converted as 12 bits value to match Timer config at 12 bits. So you should have the proper duty cycle if you comment the `PWM_RESOLUTION` in `variant.h`.

jeffstaley, 2019-11-25:
OK I see that now and that should be cleaner.  I'll give that a try. Thanks.

PR #814 (2019-12-09): Rework PWM_RESOLUTION
**Summary**

Rework `PWM_RESOLUTION`

There is no more distinction between pwm resolution
at Arduino API level and at HardwareTimer level.
`PWM_RESOLUTION` is default resolution (Arduinio API and HardwareTimer)
It is set to 8 bits like default Arduino pwm resolution.
HardwareTimer resolution can now take any value from 1bit to 16bit
(like Arduino API).
Remove variant `PWM_RESOLUTION` configuration when it is set to 8,
as it is already the default value.

Fixes #790
commit e3b3aba7 (2019-12-09): Rework PWM_RESOLUTION

There is no more distinction between pwm resolution
at arduino API level and at HardwareTimer level.
PWM_RESOLUTION is default resolution (Arduinio API and HardwareTimer)
It is set to 8bit like default arduino pwm resolution.
HardwareTimer resolution can now take any value from 1bit to 16bit
(like Arduino API).
Remove variant PWM_RESOLUTION configuration when it is set to 8,
as it is already the default value.
Fixes #790
closed 2019-12-09 by fpistm