stm32duino/Arduino_Core_STM32 #791: CDC ACM not working with 1.7.0

lpechacek, 2019-11-24:
USB stack is perhaps not initialized on Maple Mini (STM32F103CB). 

** Test code **
```
void setup() {
}

void loop() {
  SerialUSB.write("+U");
  delay(1000);
}
```

** Symtoms **
Linux kernel log after uploading the above sketch.
```
[37476.100137] usb 1-2: new full-speed USB device number 90 using xhci_hcd
[37476.249181] usb 1-2: New USB device found, idVendor=1eaf, idProduct=0003, bcdDevice= 2.01
[37476.249190] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[37476.249196] usb 1-2: Product: Maple 003
[37476.249200] usb 1-2: Manufacturer: LeafLabs
[37476.249205] usb 1-2: SerialNumber: LLM 003
[37478.739198] usb 1-2: USB disconnect, device number 90
[37478.920186] debugfs: Directory '18' with parent 'devices' already present!
[37479.048127] usb 1-2: new full-speed USB device number 91 using xhci_hcd
[37479.196534] usb 1-2: device descriptor read/all, error -32
[37479.196667] debugfs: Directory '19' with parent 'devices' already present!
[37479.324115] usb 1-2: new full-speed USB device number 92 using xhci_hcd
[37479.476513] usb 1-2: device descriptor read/all, error -32
[37479.476580] usb usb1-port2: attempt power cycle
[37479.792185] debugfs: Directory '20' with parent 'devices' already present!
[37480.128107] usb 1-2: new full-speed USB device number 93 using xhci_hcd
[37480.148510] usb 1-2: device descriptor read/8, error -32
```

**Expected behavior**
With 1.6.1 the above code works as expected - CDC ACM device appears.
```
[38085.592190] usb 1-2: new full-speed USB device number 98 using xhci_hcd
[38085.613237] usb 1-2: New USB device found, idVendor=1eaf, idProduct=0003, bcdDevice= 2.01
[38085.613239] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[38085.613240] usb 1-2: Product: Maple 003
[38085.613241] usb 1-2: Manufacturer: LeafLabs
[38085.613242] usb 1-2: SerialNumber: LLM 003
[38099.856349] usb 1-2: reset full-speed USB device number 98 using xhci_hcd
[38102.379110] usb 1-2: USB disconnect, device number 98
[38102.688114] usb 1-2: new full-speed USB device number 99 using xhci_hcd
[38102.837406] usb 1-2: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[38102.837414] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[38102.837419] usb 1-2: Product: MAPLEMINI_F103CB CDC in FS Mode
[38102.837423] usb 1-2: Manufacturer: STMicroelectronics
[38102.837427] usb 1-2: SerialNumber: 8D7C315C4949
[38102.840303] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
```

**Desktop (please complete the following information):**
 - OS: openSUSE Tumbleweed 20191119
 - Arduino IDE version: 1.8.10
 - STM32 core version: 1.7.0
 - Tools menu settings if not the default: 
 -- U(S)ART support: Enabled (generic Serial)
 -- USB support: CDC (generic 'Serial' supersede U(S)ART)
 -- USB speed: Low/Full speed
 - Upload method: DFU

**Board (please complete the following information):**
 - Name: Generic STM32F1 series / Maple Mini F103CB
 - Hardware Revision: ?? - some random clone
 - Extra hardware used if any: n/a

**Additional context**
There is far more trouble with UART support in 1.7.0. More in additional reports.

fpistm, 2019-11-24:
Hi @lpechacek 
This is a know issue and this is fixed. See #670.
Duplicate of #664

lpechacek, 2019-11-24:
@fpistm, thanks for the hint! I saw the issue but it didn't occur to me that this might be linked. Let me try the fix.

lpechacek, 2019-11-24:
The fix for issue 664 indeed solved the trouble. Many thanks, @fpistm, for fast response and help!

closed 2019-11-24 by lpechacek